Koenkk/zigbee2mqtt #11479: Unable to bind TS0001's ?

mzealey, 2022-02-18:
### What happened?

I have 3 TS0001 which I would like to be in sync with each other ie when one button gets pressed all of them toggle. I believe this is device binding, I guess for the `OnOff` cluster. But when I go into the binding tab of the device, the only clusters are `Ota` and `genTime` even though `OnOff` is shown on the controller bind
[image]

### What did you expect to happen?

I was expecting that I could bind (either to each other, or to a group which I have created) the state of these devices together

### Zigbee2MQTT version

1.23.0 (commit #6ea61ba)

### Adapter firmware version

20211217

### Adapter

Sonoff SNZB-1 zStack3x0

olili, 2022-02-27:
I'm interessted in this feature as well.
 
I think TUYA calls this feature "muti-control association" and according sales statement this feature is also available at least with some TS0001.

vk496, 2022-03-07:
Me too. Not sure if is Tuya devices problem or is limited to only few devices to bind

closed 2022-04-15 by github-actions[bot]

toxuin, 2022-08-29:
Why is this closed as completed? This is still happening and is still indeed a problem.

setfire2, 2024-01-21:
I would be interested in a solution as well. I think it's the intended way you do changeover switching with these kind of switches. A quite common use case.

olili, 2024-01-21:
feature request was closed due to missing interesst...
[https://github.com/Koenkk/zigbee2mqtt/issues/13323](url)