Koenkk/zigbee2mqtt #11482: Hello, you've found a bug

VladNord, 2022-02-18:
### What happened?

My ISP was having problems, so no internet connection.
until the Internet connection was restored, it was impossible to use z2m.
the following error was displayed. dump attached
[image]

### Zigbee2MQTT version

1.23.0-1

### Adapter firmware version

0x26720700

### Adapter

deconz

### Debug log

[initialState.json.zip](https://github.com/Koenkk/zigbee2mqtt/files/8100837/initialState.json.zip)

Koenkk, 2022-02-19:
Please create issues related to the frontend here: https://github.com/nurikk/zigbee2mqtt-frontend/issues

closed 2022-02-19 by Koenkk

iluminat23, 2022-06-19:
https://github.com/nurikk/zigbee2mqtt-frontend/issues/1364