Koenkk/zigbee2mqtt #11483: [New device support]: Device - model 'TS011F' and manufacturer name '_TZ3000_3zofvcaa'

Gera1974, 2022-02-18:
### Link

https://pt.aliexpress.com/item/1005003822349807.html?gatewayAdapt=glo2bra&spm=a2g0o.order_list.0.0.21efcaa4zLrmXD

### Database entry

{"id":22,"type":"Router","ieeeAddr":"0xa4c138255c04d0df","nwkAddr":4238,"manufId":4417,"manufName":"_TZ3000_3zofvcaa","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1}},"manuSpecificTuya_3":{"attributes":{"53264":1,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"53264":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"53264":1}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"53264":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1645221345670,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,

I've got a TS011F _TZ3000_3zofvcaa zemismat wall outlet TB242-ZB that is not supported.  According the description, it is possible to control the 2 power plugs and the 2 usb plugs but there is no energy measuring feature.

I've tried to use instruction how to support new devices but I failed (seems too complicated for me). I've also tried dev branch.

Is it possible to add it ?
Thank you.

Gera1974, 2022-03-25:
Any chance to have this device officially supported on next version of Zigbee2mqtt ?

Koenkk, 2022-03-26:
Could you check if the controlling of endpoints is fixed by changing the `configure()` like this?

Change

```js
const firstEndpoint = device.getEndpoint(1);
await reporting.bind(firstEndpoint, coordinatorEndpoint, ['genOnOff']);
```

to
```js
const firstEndpoint = device.getEndpoint(1);
await firstEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(firstEndpoint, coordinatorEndpoint, ['genOnOff']);
```

Gera1974, 2022-03-26:
Dear Koen,

I just tested it and now it works correctly ! Now it is possible to control
the four plugs independently !!!!

One question: In the original application there is an additional switch
that turns on and turns off all the four plugs simultaneously. Is it
possible to also add this feature ? Do you know how to do it ?

Thank you very much,
Geraldo

Koenkk, 2022-03-26:
Added it (so it will be supported in the next z2m release)

> One question: In the original application there is an additional switch
that turns on and turns off all the four plugs simultaneously. Is it
possible to also add this feature ? Do you know how to do it ?

For this you would have to sniff the command when doing this via the TuYa bridge (https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531). Alternatively you could put all the endpoints in 1 group and control the group.

closed 2022-04-03 by github-actions[bot]

rodrigogbs, 2023-06-23:
@Koenkk, there is a missing configuration on this device: childLock

I tried with this external_converter and it works like a charm:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_3zofvcaa'}],
    model: 'TS011F_2_gang_2_usb_wall',
    vendor: 'TuYa',
    description: '2 gang 2 usb wall outlet',
    extend: tuya.extend.switch({power_on_behavior: true, backlightModeLowMediumHigh: true, childLock: true, endpoints: ['l1', 'l2', 'l3', 'l4']}),
    whiteLabel: [{vendor: 'Zemismart', model: 'ZTB242'}],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2, 'l3': 3, 'l4': 4};
    },
    meta: {multiEndpoint: true},
    configure: tuya.configureMagicPacket,
};

module.exports = definition;
```

Can you please add this to the branch?

Thanks!

rodrigogbs, 2023-06-23:
You're amazing @Koenkk , thank you!