Koenkk/zigbee2mqtt #11485: Device in multiple groups gives 'Error: Cannot read properties of undefined'

cinadr, 2022-02-19:
### What happened?

I have 3 [Immax 07088L](https://www.zigbee2mqtt.io/devices/07088L.html#immax-07088l) lights in front of my house. 
I want to get lit 3 of them when my garage door opens and only 2 of them if my front door unlocked/opened. I defined two groups one is containing all of them and the other is only two of them. If I turn on/off the lights sending commands to these groups or individually I get the following error messages:
```
error 2022-02-19 08:57:20: Error: Cannot read properties of undefined (reading 'definition')
debug 2022-02-19 08:57:20: TypeError: Cannot read properties of undefined (reading 'definition')
    at Receive.shouldProcess (C:\Program Files\Zigbee2MQTT\lib\extension\receive.ts:72:26)
    at Receive.onDeviceMessage (C:\Program Files\Zigbee2MQTT\lib\extension\receive.ts:109:19)
    at EventEmitter.emit (node:events:402:35)
    at EventBus.emitDeviceMessage (C:\Program Files\Zigbee2MQTT\lib\eventBus.ts:88:22)
    at Controller.<anonymous> (C:\Program Files\Zigbee2MQTT\lib\zigbee.ts:102:27)
    at Controller.emit (node:events:390:28)
    at Controller.onZclOrRawData (C:\Program Files\Zigbee2MQTT\node_modules\zigbee-herdsman\src\controller\controller.ts:685:18)
    at ZStackAdapter.<anonymous> (C:\Program Files\Zigbee2MQTT\node_modules\zigbee-herdsman\src\controller\controller.ts:135:70)
    at ZStackAdapter.emit (node:events:390:28)
    at ZStackAdapter.onZnpRecieved (C:\Program Files\Zigbee2MQTT\node_modules\zigbee-herdsman\src\adapter\z-stack\adapter\zStackAdapter.ts:772:30)
```
Without the second group all is working properly. If I only add one light to the second group error message appears again. And it is appearing for all the lights in the second group.

Any more info needed to fix this?

### What did you expect to happen?

Error free operation with both groups defined simultaneously.

### How to reproduce it (minimal and precise)

Probably add same devices to multiple groups and operate them.

### Zigbee2MQTT version

1.22.0

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB stick

### Debug log

[zigbee2mqtt.zip](https://github.com/Koenkk/zigbee2mqtt/files/8101679/zigbee2mqtt.zip)

Koenkk, 2022-02-19:
Make sure to reproduce with the latest z2m first (1.23.0) and provide the debug log of that.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

cinadr, 2022-02-19:
> Make sure to reproduce with the latest z2m first (1.23.0) and provide the debug log of that.
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

I've updated to 1.23.0 and flashed newer firmware to Slaesh's stick and it solved this issue. Thank you.

closed 2022-02-19 by cinadr