Koenkk/zigbee2mqtt #11491: [New device support]: TS0012 double press / hold

mzealey, 2022-02-19:
### Link

https://www.aliexpress.com/item/1005002233721784.html (2-gang, No neutral)

### Database entry

{"id":11,"type":"EndDevice","ieeeAddr":"0x60a423fffeb046e2","nwkAddr":23859,"manufId":4098,"manufName":"_TZ3000_fvh3pjaz","powerSource":"Mains (single phase)","modelId":"TS0012","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}},"genBasic":{"attributes":{"modelId":"TS0012","manufacturerName":"_TZ3000_fvh3pjaz","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc1a45","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc1a45","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1645274422481,"defaultSendRequestWhen":"immediate"}

### Comments

Pairs and works fine, but only lists single presses as options. The TS0002 etc products which are the same but with neutral, as well as the TS0042 products which are battery-operated all support double tap and long press. Is there a way I can figure out if this no-neutral devices support these events too?

MattWestb, 2022-02-20:
I think this is one ZCL light switch and not one multi click remote as the TS0042 is that is sending no ZCL commands for double and long press.

How is the light behaving then double clicking one of the gangs ?

mzealey, 2022-02-20:
Double seems to just double trigger the lights, but a few second hold does nothing which makes me think it may be a separate action?

mzealey, 2022-03-23:
not stale but i'm unable to remove the label...

closed 2022-04-30 by github-actions[bot]