Koenkk/zigbee2mqtt #11501: [New device support]: GIDEALED Smart ZigBee 3.0 Controller and RGBWW LED Strip

ancestor, 2022-02-20:
### Link

https://www.amazon.com/dp/B09P7Z2STP?psc=1&ref=ppx_yo2_dt_b_product_details

### Database entry

{   "id": 94,   "type": "Router",   "ieeeAddr": "0x94deb8fffe137263",   "nwkAddr": 64636,   "manufId": 4107,   "manufName": "Light",   "powerSource": "Mains (single phase)",   "modelId": "A11",   "epList": [     1,     242   ],   "endpoints": {     "1": {       "profId": 260,       "epId": 1,       "devId": 269,       "inClusterList": [         64513,         0,         3,         4,         5,         6,         8,         768,         4096,         65518       ],       "outClusterList": [         25,         4096,         65518       ],       "clusters": {         "genBasic": {           "attributes": {             "modelId": "A11",             "manufacturerName": "Light",             "powerSource": 0,             "zclVersion": 8,             "appVersion": 1,             "stackVersion": 3,             "hwVersion": 2,             "dateCode": "20210302",             "swBuildId": "3.1.0"           }         },         "lightingColorCtrl": {           "attributes": {             "colorCapabilities": 25,             "colorTempPhysicalMin": 153,             "colorTempPhysicalMax": 370           }         }       },       "binds": [],       "configuredReportings": [],       "meta": {}     },     "242": {       "profId": 41440,       "epId": 242,       "devId": 102,       "inClusterList": [         33       ],       "outClusterList": [         33       ],       "clusters": {},       "binds": [],       "configuredReportings": [],       "meta": {}     }   },   "appVersion": 1,   "stackVersion": 3,   "hwVersion": 2,   "dateCode": "20210302",   "swBuildId": "3.1.0",   "zclVersion": 8,   "interviewCompleted": true,   "meta": {     "configured": 88764544   },   "lastSeen": 1645322718119,   "defaultSendRequestWhen": "immediate" }

### Comments

These LED strips seem to be similar to already supported GIDERWEL ones, however they have a different model name (A11 vs A10) and they are RGBWW opposed to RGB. So I created the custom converter and they seem to work fine. The only problem I ran into is that every time I change the color of these strips the controller only utilizes the RGB diodes, both white ones remain off unless I start changing the color temperature. In that case the RGB diodes go off and only two white ones work.
Perhaps this is a controller issue/feature or an expected behavior, however my Philips Hue strips connected to the same zigbee2mqtt instance work properly (i.e. the white diodes are always in use enhancing the colors).

Is there a way to make these strips work similar to Philips Hue strips?
Thanks!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['A11'], 
    model: 'ZC05M', 
    vendor: 'GIDEALED', 
    description: 'Smart Zigbee RGB LED strip controller (by Antoshka)',
    extend: extend.light_onoff_brightness_colortemp_color({supportsHS: true, colorTempRange: [153, 370]}), 
};

module.exports = definition;
```

### Supported color modes

color temperature, color (rgb)

### Color temperature range

{"colorTempPhysicalMin": 153, "colorTempPhysicalMax": 370}

Koenkk, 2022-02-20:
> Perhaps this is a controller issue/feature or an expected behavior, however my Philips Hue strips connected to the same zigbee2mqtt instance work properly (i.e. the white diodes are always in use enhancing the colors).

This depends on logic of the device itself, what z2m does is to just request the device to show a certain color. Could you make a pr to add support for this device?

ancestor, 2022-02-21:
> This depends on logic of the device itself, what z2m does is to just request the device to show a certain color. Could you make a pr to add support for this device?

Got it, thanks for confirming! Absolutely, will submit a PR.

Koenkk, 2022-02-21:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3904

closed 2022-02-21 by Koenkk

termeaunicolas, 2022-05-12:
Hi Guy,i try to add this device,but i am stuck when i have to add the converter on the config.yaml,it's loo weird mine.
Someone can help me please
[configuration.zip](https://github.com/Koenkk/zigbee2mqtt/files/8682515/configuration.zip)
e

sreeg, 2022-07-10:
Im using [Zigbee2MQTT] usb stick and when paired it is showing the device as unsupported.
Below is the device info from "About" tab:
Last seen
N/A
Availability
Online
Device type
Router
Zigbee Model
A11
Zigbee Manufacturer
Light
Support status
Not supported

IEEE Address
0x94deb8fffe169fa1
Network address
0xFD6F
Firmware build date
20210302
Firmware version
3.1.0
Power
Interview completed
[image]

True

sreeg, 2022-07-10:
Zigbee hub version
[image]

sreeg, 2022-07-10:
Please help!

pie-rs, 2022-09-28:
@ancestor did you have these working ok?  I am finding that the Red is not working as expected. Green and Blue are fine but Red (purple through orange) - while faintly more red than white is basically white.

Same issue as you with the white and yellow - I'm pretty sure thats a controller issue but curious if you are seeing the same issues with the red channel?