Koenkk/zigbee2mqtt #11505: [New device support]: Tuya Plug model 'TS011F' with manufacture name '_TZ3000_8a833yls'

radaiming, 2022-02-20:
### Link

https://m.tb.cn/h.flAoq8i

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c138464d7ff012","nwkAddr":59833,"manufId":4417,"manufName":"_TZ3000_8a833yls","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_8a833yls","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"53264":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c24115","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-353372667},"lastSeen":1645353362957,"defaultSendRequestWhen":"immediate"}

### Comments

This plug does not provide electrical measurement(according to product page and my testing external converter), works well with `fz.on_off` and `fz.tuya_switch_power_outage_memory`

It looks similar to model 'TS011F_plug_2', but if I add any one of `fz.ts011f_plug_indicator_mode` and `fz.ts011f_plug_child_mode` to `fromZigbee` and `toZigbee`, zigbee2mqtt will crash with trace:
```(node:396) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'options' of undefined
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:83:58)
    at start (/app/index.js:100:18)
```

By the way, the product page may require login to view on desktop browser, mobile browser may be better

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS011F'],
    model: '_TZ3000_8a833yls',
    description: 'Smart plug (without power monitoring)',
    vendor: 'TuYa',
    supports: 'on/off',
    fromZigbee: [fz.on_off, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
    exposes: [
        e.switch(),
        exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore']).withDescription('Recover state after power outage'),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;
```

Koenkk, 2022-02-20:
Added, could you check it works correctly?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

radaiming, 2022-02-22:
Yes it's working, thanks!

closed 2022-02-22 by radaiming