Koenkk/zigbee2mqtt #11507: Xiaomi (aquara) WS-EUK03 not updating state in z2m

graphicsxp, 2022-02-20:
### What happened?

I have two of this devices and for one of them, the state is not updating in z2m when pressing the physial switch. (the light is turned on just fine though)

The state is updated if I toggle the switch in z2m or in home assistant. 

[image]

[image]

### What did you expect to happen?

The state should be updated in z2m when pressing the physical switch

### How to reproduce it (minimal and precise)

because the other switch updates the state, this might be a configuration issue but I don't know what.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

0x26660700

### Adapter

conbee2 

### Debug log

nothing shows up in the log when pressing the physical switch

McGiverGim, 2022-02-20:
Maybe the decoupled mode is not working? Can you try to change it to decoupled to see if it works?

graphicsxp, 2022-02-20:
I tried decoupled mode and then pushing the switch buttton has no effect. It no longer turns on the light.

Actually I don't understand what the operation_mode is. What is the difference between control_relay and decoupled ?

McGiverGim, 2022-02-20:
If I'm not wrong, control relay makes the switch control a relay (usually a light). In decoupled it does not control nothing, and it's you who must control the light "manually" when you receive the message from the switch.

z-x-f, 2022-02-21:
Try this external converter, then remove the switch restart z2m and pair the switch again and see if that fixes it.
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['lumi.switch.n1aeu1'],
        model: 'WS-EUK03',
        vendor: 'Xiaomi',
        description: 'Aqara smart wall switch H1 EU (with neutral, single rocker)',
        fromZigbee: [fz.on_off, fz.xiaomi_power, fz.xiaomi_multistate_action, fz.aqara_opple],
        toZigbee: [tz.on_off, tz.xiaomi_power, tz.xiaomi_switch_operation_mode_opple, tz.xiaomi_switch_power_outage_memory,
            tz.xiaomi_flip_indicator_light],
        exposes: [e.switch(), e.action(['single', 'double']), e.power().withAccess(ea.STATE_GET), e.energy(), e.flip_indicator_light(),
            e.power_outage_memory(), e.temperature().withAccess(ea.STATE),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled']).withDescription('Decoupled mode')],
        onEvent: preventReset,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            // set "event" mode
            await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
        },
    };
module.exports = definition;
```

tjxism, 2022-02-21:
Are you using the switch with smart bulbs? If yes, you should set the switch to 'Decoupled' mode, so the smart light connected to the switch is always powered. Then the wall switch are just smart triggers in Z2M and you can then set them to control the smart light (either physically connected to the switch or not). 'Control Relay' should only be used to control a dumb light hard-wired to the switch. It will also update its state to Z2M so you can use this state to control other smart devices over Z2M. 

Aqara stated (in it's Chinese materials) you shouldn't wire the switch to a smart bulb, or it will cause the switch to malfunction. I don't know why this should happen, but there's no point to do so anyway.

graphicsxp, 2022-02-21:
@tjxism . That's an interesting point. 

Actually I have 3 Aquara switches (two 1-gang and one 2-gang). 

First 1-gang : connnected to dumb light and reports state to z2m.  Configured as Control Relay.
Second 1-gang: connected to dumb light but does not report state to z2m. Configured as Control Relay.
2-gang :  connected to smart ikea bulbs. Does not report state to z2m. Configured as Control Relay.

So perharps I should configure the 2-gang as Decouped as you mentionned. Although if I do that the bulbs won't light up when pushing the buttons (until I do some kind of automation in node-red or ha).

However there's no explanation for the 1-gang that does not report state. I will try @z-x-f  suggestion with the external converter.

graphicsxp, 2022-02-21:
@z-x-f that converter gives me an error when restarting z2m and i can no longer access the web ui : 

` UnhandledPromiseRejectionWarning: ReferenceError: preventReset is not defined`

it's related to line 20.  Any idea ?

z-x-f, 2022-02-21:
Ah.. delete <code>onEvent: preventReset</code>

graphicsxp, 2022-02-21:
ah this converter fixed the 1-gang switch connected to dumb light. Now when I press the switch the correct state is reported to z2m 👍 

would you mind heping with the 2-gang switch as well ? The model is **WS-EUK04**

closed 2022-02-21 by Koenkk

graphicsxp, 2022-02-21:
any chance this could be done for the WS-EUK04 too  ?  There is the exact same issue.

z-x-f, 2022-02-21:
Yep

Crithane, 2022-04-27:
I also seem to be running into this issue, even after the fix

I'm running zigbee2mqtt using the home assistant addon

I've tried both version 1.25.0-1 stable and edge

I'm running 2 WS-EUK03 with dumb bulbs and Sonoff Zigbee 3.0 USB Adapter Plus

pr0xyfl00d3r, 2023-07-27:
I'm currently having this issue