Koenkk/zigbee2mqtt #11510: [New device support]: Aqara T1 smart LED bulb with model 'lumi.light.acn014'

radaiming, 2022-02-20:
### Link

https://item.jd.com/100002262502.html

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0x54ef44100035f032","nwkAddr":42656,"manufId":4447,"manufName":"LUMI\u0000","powerSource":"DC Source","modelId":"lumi.light.acn014","epList":[1,21,31,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,2,3,4,5,6,8,9,12,15,18,768,1794,2820,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.light.acn014","manufacturerName":"LUMI\u0000","powerSource":4,"zclVersion":3,"appVersion":30,"stackVersion":2,"hwVersion":1,"dateCode":"Jul 29 2021\u0000"}},"aqaraOpple":{"attributes":{"308":{"type":"Buffer","data":[4,205,21,212,213,254]}}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":268,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"31":{"profId":260,"epId":31,"devId":268,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":30,"stackVersion":2,"hwVersion":1,"dateCode":"Jul 29 2021\u0000","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1645350596512,"defaultSendRequestWhen":"immediate"}

### Comments

The product page claims its model as ZNLDP13LM, but actually I see 'lumi.light.acn014', I copied ZNLDP13LM's code as external converter and everything works fine except color_temp_startup, seems value is set but device ignore it or doesn't support this function at all.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const extend = require('zigbee-herdsman-converters/lib/extend');

const xiaomiExtend = {
    light_onoff_brightness_colortemp: (options={disableColorTempStartup: true}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        fromZigbee: extend.light_onoff_brightness_colortemp(options).fromZigbee.concat([
            fz.xiaomi_bulb_interval, fz.ignore_occupancy_report, fz.ignore_humidity_report,
            fz.ignore_pressure_report, fz.ignore_temperature_report,
        ]),
    }),
};

const definition = {
    zigbeeModel: ['lumi.light.acn014'],
    model: 'ZNLDP13LM',
    vendor: 'Xiaomi',
    description: 'Aqara T1 smart LED bulb',
    extend: xiaomiExtend.light_onoff_brightness_colortemp({disableEffect: true, colorTempRange: [153, 370]}),
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

Koenkk, 2022-02-20:
Added it to the existing ZNLDP12LM definition.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-20 by Koenkk

gorbunovav, 2022-05-27:
@Koenkk Hi!
I have both the `lumi.light.cwac02` and the `lumi.light.acn014`. Labels on bulbs themselves are identical (except the quality maybe).
The package is a bit different, but both states that it is `ZNLDP13LM` and `T1` model. 
`lumi.light.acn014` shows newer firmware build date ("Jul 29 2021" vs "Jan 16 2021" for the `lumi.light.cwac02`).

`lumi.light.acn014` reports "UNSUPPORTED_ATTRIBUTE" for the attempt to set the `power_outage_memory` value.

[image]

Koenkk, 2022-05-27:
@gorbunovav thanks, moved it to the ZNLDP13LM (which indeed doesn't support `power_outage_memory`). 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)