Koenkk/zigbee2mqtt #11511: [New device support]:

DonRonito, 2022-02-20:
### Link

https://www.philips-hue.com/en-us/p/hue-white-and-color-ambiance-play-gradient-lightstrip-55-inch/046677560409

### Database entry

{ "id": 17, 	"type": "Router", 	"ieeeAddr": "0x00178801095ebccb", 	"nwkAddr": 60628, 	"manufId": 4107, 	"manufName": "Signify Netherlands B.V.", 	"powerSource": "Mains (single phase)", 	"modelId": "LCX001", 	"epList": [11, 242], 	"endpoints": { 		"11": { 			"profId": 260, 			"epId": 11, 			"devId": 269, 			"inClusterList": [0, 3, 4, 5, 6, 8, 4096, 64515, 768, 64513], 			"outClusterList": [25], 			"clusters": { 				"genBasic": { 					"attributes": { 						"modelId": "LCX001", 						"manufacturerName": "Signify Netherlands B.V.", 						"powerSource": 1, 						"zclVersion": 2, 						"appVersion": 2, 						"stackVersion": 1, 						"hwVersion": 1, 						"dateCode": "20211215", 						"swBuildId": "1.94.2" 					} 				} 			}, 			"binds": [], 			"configuredReportings": [], 			"meta": {} 		}, 		"242": { 			"profId": 41440, 			"epId": 242, 			"devId": 97, 			"inClusterList": [], 			"outClusterList": [33], 			"clusters": {}, 			"binds": [], 			"configuredReportings": [], 			"meta": {} 		} 	}, 	"appVersion": 2, 	"stackVersion": 1, 	"hwVersion": 1, 	"dateCode": "20211215", 	"swBuildId": "1.94.2", 	"zclVersion": 2, 	"interviewCompleted": true, 	"meta": {}, 	"lastSeen": 1645364856639, 	"defaultSendRequestWhen": "immediate" }

### Comments

..

Koenkk, 2022-02-20:
Added.

Changes will be available in the dev branch tomorrow. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-20 by Koenkk

DonRonito, 2022-02-20:
Thanks a lot. Much appreciated :)