Koenkk/zigbee2mqtt #11517: No endpoint names other than l1 and l2 work

zgdump, 2022-02-20:
### What happened?

I am developing my own zigbee thermostat. In the converter, in exposes, I wrote `climate()` with some field. Like this:

```js
exposes.climate()
    .withSystemMode(['off', 'auto'])
    .withLocalTemperature()
    .withLocalTemperatureCalibration(0.0, 2.5, 0.5)
    .withSetpoint('occupied_heating_setpoint', 15.0, 50.0, 0.5)
    .withEndpoint(epName),
```

I have two thermostat endpoints in my device, I wrote this in the converter: 

```js
meta: {
    multiEndpoint: true
},
endpoint: (device) => {
    return {
        'l1': 1,
        'l2': 2
    };
},
```

And everything works. 
But should I change `l1` to `thermostat_1` or `th1` and always get this:

[image]

Which names are allowed and which are not? Is it possible to make readable names for the user, and leave l1 and l2 under the hood? 

### What did you expect to happen?

Any name for the endpoint is valid and does not cause strange errors 

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20210120

### Adapter

SONOFF ZigBEE 3.0 USB Dongle Plus

Koenkk, 2022-02-20:
only endpoint names in https://github.com/Koenkk/zigbee2mqtt/blob/6ea61bae2cf441e15612dc68c257797fba46312b/lib/util/utils.ts#L30 can be used

zgdump, 2022-02-20:
> only endpoint names in
> 
> https://github.com/Koenkk/zigbee2mqtt/blob/6ea61bae2cf441e15612dc68c257797fba46312b/lib/util/utils.ts#L30
> can be used

Thanks, now I understand! Will arbitrary names be supported? Is it possible to create a user-friendly name for the endpoint, an alias?

Koenkk, 2022-02-20:
Only if they are added to endpoint names

zgdump, 2022-02-20:
Sad, thank you

closed 2022-02-20 by zgdump