Koenkk/zigbee2mqtt #11524: Schneider Electric Odace S520530W: Failure and error on pairing

adevress, 2022-02-20:
### What happened?

- On pairing of  Schneider Electric Odace S520530W, Zigbee2MQTT returned an error in the logs ( debug log joined )
- After that, S520530W does not communicate anymore with Zigbee2MQTT

- Removing the device manually ( https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request ) and factory reset the device does not change the problem: The device stay invisible and refuses to pair again

- Version of Zigbee2MQTT is 1.23.0

- Do not hesitate if I can help you to get more information on this ( debug, logs, etc )

### What did you expect to happen?

Schneider Electric Odace S520530W should join properly and operate as specified https://www.zigbee2mqtt.io/devices/S520530W.html

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

uknown

### Adapter

CC2652RB

### Debug log

Zigbee2MQTT:info  2022-02-20 17:20:40: Successfully interviewed '0x8cf681fffee0352a', device has successfully been paired
Zigbee2MQTT:info  2022-02-20 17:20:40: Device '0x8cf681fffee0352a' is supported, identified as: Schneider Electric Odace connectable relay switch 10A (S520530W)
Zigbee2MQTT:info  2022-02-20 17:20:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Odace connectable relay switch 10A","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"S520530W","options":[],"supports_ota":false,"vendor":"Schneider Electric"},"friendly_name":"0x8cf681fffee0352a","ieee_address":"0x8cf681fffee0352a","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2022-02-20 17:20:40: Configuring '0x8cf681fffee0352a'
Zigbee2MQTT:info  2022-02-20 17:20:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Odace connectable relay switch 10A","friendly_name":"0x8cf681fffee0352a","model":"S520530W","supported":true,"vendor":"Schneider Electric"},"type":"pairing"}'
Zigbee2MQTT:error 2022-02-20 17:20:41: Failed to configure '0x8cf681fffee0352a', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:24)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/schneider_electric.js:256:29)
    at Configure.configure (/app/lib/extension/configure.ts:115:37)
    at EventEmitter.<anonymous> (/app/lib/extension/configure.ts:79:62)
    at EventEmitter.emit (events.js:412:35)
    at EventBus.emitDeviceInterview (/app/lib/eventBus.ts:67:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:86:27)
    at Controller.emit (events.js:400:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:481:14)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:535:22))
Zigbee2MQTT:info  2022-02-20 17:20:43: Configuring '0x8cf681fffee0352a'
Zigbee2MQTT:error 2022-02-20 17:20:43: Failed to configure '0x8cf681fffee0352a', attempt 2 (TypeError: Cannot read property 'bind' of undefined
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:24)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/schneider_electric.js:256:29)
    at Configure.configure (/app/lib/extension/configure.ts:115:37)
    at EventEmitter.<anonymous> (/app/lib/extension/configure.ts:80:62)
    at EventEmitter.emit (events.js:412:35)
    at EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:45:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:66:27)
    at Controller.emit (events.js:400:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:481:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:695:18))
Zigbee2MQTT:info  2022-02-20 17:20:43: Configuring '0x8cf681fffee0352a'
Zigbee2MQTT:error 2022-02-20 17:20:43: Failed to configure '0x8cf681fffee0352a', attempt 3 (TypeError: Cannot read property 'bind' of undefined
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:24)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/schneider_electric.js:256:29)
    at Configure.configure (/app/lib/extension/configure.ts:115:37)
    at EventEmitter.<anonymous> (/app/lib/extension/configure.ts:80:62)
    at EventEmitter.emit (events.js:412:35)
    at EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:45:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:66:27)
    at Controller.emit (events.js:400:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:481:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:695:18))

vspek, 2022-05-02:
I removed the device and then tried to join it again with debug on, here is my log below.
To @[adevress](https://github.com/adevress) I had much difficulty to get it joined again. I Reset it (3 * fast, then 10) then 3* fast 6 seconds, suddenly it was pairing again, but not after just the reset.

Info 2022-05-02 13:37:16Device '0x680ae2fffe117998' joined
Info 2022-05-02 13:37:16MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe117998","ieee_address":"0x680ae2fffe117998"},"type":"device_joined"}'
Info 2022-05-02 13:37:16Starting interview of '0x680ae2fffe117998'
Info 2022-05-02 13:37:16MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe117998","ieee_address":"0x680ae2fffe117998","status":"started"},"type":"device_interview"}'
Debug 2022-05-02 13:37:25Device '0x680ae2fffe117998' announced itself
Info 2022-05-02 13:37:25MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe117998","ieee_address":"0x680ae2fffe117998"},"type":"device_announce"}'
Debug 2022-05-02 13:37:50Received Zigbee message from '0x680ae2fffe117998', type 'readResponse', cluster 'genBasic', data '{"modelId":"NHPB/SWITCH/1"}' from endpoint 1 with groupID 0
Debug 2022-05-02 13:37:50Received Zigbee message from '0x680ae2fffe117998', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Schneider Electric"}' from endpoint 1 with groupID 0
Debug 2022-05-02 13:37:51Received Zigbee message from '0x680ae2fffe117998', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
Debug 2022-05-02 13:37:51Received Zigbee message from '0x680ae2fffe117998', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Debug 2022-05-02 13:37:51Received Zigbee message from '0x680ae2fffe117998', type 'readResponse', cluster 'genBasic', data '{"appVersion":2}' from endpoint 1 with groupID 0
Debug 2022-05-02 13:37:51Received Zigbee message from '0x680ae2fffe117998', type 'readResponse', cluster 'genBasic', data '{"stackVersion":6}' from endpoint 1 with groupID 0
Debug 2022-05-02 13:37:51Received Zigbee message from '0x680ae2fffe117998', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Debug 2022-05-02 13:37:51Received Zigbee message from '0x680ae2fffe117998', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Debug 2022-05-02 13:37:51Received Zigbee message from '0x680ae2fffe117998', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"002.004.015 R"}' from endpoint 1 with groupID 0
Debug 2022-05-02 13:37:51Received Zigbee message from '0x680ae2fffe117998', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 21 with groupID 0
Info 2022-05-02 13:37:51Successfully interviewed '0x680ae2fffe117998', device has successfully been paired
Info 2022-05-02 13:37:51Device '0x680ae2fffe117998' is supported, identified as: Schneider Electric Odace connectable relay switch 10A (S520530W)
Info 2022-05-02 13:37:51MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Odace connectable relay switch 10A","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"S520530W","options":[],"supports_ota":false,"vendor":"Schneider Electric"},"friendly_name":"0x680ae2fffe117998","ieee_address":"0x680ae2fffe117998","status":"successful","supported":true},"type":"device_interview"}'
Info 2022-05-02 13:37:51Configuring '0x680ae2fffe117998'
Info 2022-05-02 13:37:51MQTT publish: topic 'homeassistant/switch/0x680ae2fffe117998/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x680ae2fffe117998/set","device":{"identifiers":["zigbee2mqtt_0x680ae2fffe117998"],"manufacturer":"Schneider Electric","model":"Odace connectable relay switch 10A (S520530W)","name":"0x680ae2fffe117998","sw_version":"002.004.015 R"},"name":"0x680ae2fffe117998","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x680ae2fffe117998","unique_id":"0x680ae2fffe117998_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Info 2022-05-02 13:37:51MQTT publish: topic 'homeassistant/sensor/0x680ae2fffe117998/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x680ae2fffe117998"],"manufacturer":"Schneider Electric","model":"Odace connectable relay switch 10A (S520530W)","name":"0x680ae2fffe117998","sw_version":"002.004.015 R"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x680ae2fffe117998 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x680ae2fffe117998","unique_id":"0x680ae2fffe117998_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
[... cut ...]
Debug 2022-05-02 13:41:04Publishing 'set' 'state' to '0x680ae2fffe117998'
Info 2022-05-02 13:41:04MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe117998', payload '{"linkquality":60,"state":"ON"}'
Debug 2022-05-02 13:41:06Received MQTT message on 'zigbee2mqtt/0x680ae2fffe117998/set' with data '{"state":"OFF"}'
Debug 2022-05-02 13:41:06Publishing 'set' 'state' to '0x680ae2fffe117998'
Info 2022-05-02 13:41:06MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe117998', payload '{"linkquality":63,"state":"OFF"}'`

closed 2022-06-10 by github-actions[bot]

atxcris1, 2024-10-17:
same problem here, did anyone found a solution for this?

Failed to configure 'OdaceBttn', attempt 2 (Error: ConfigureReporting 0x70ac08fffe2dccea/21 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:762:22) at Object.onOff (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:76:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/schneider_electric.js:512:13) at Configure.configure (/app/lib/extension/configure.ts:117:13))

vspek, 2025-07-23:
> @atxcris1 <https://github.com/atxcris1> @vspek <https://github.com/vspek> @adevress <https://github.com/adevress> may i ask an update regarding S520530W support on Z2M? i was planning to buy this model for my install but your issue make me re thinking about it :)

I bought several and I am quite happy with them.

eflye, 2025-07-23:
thanks for your reply @vspek! 
quick other question, what are the values captured? only on / off ? or also press, hold, release ?