Koenkk/zigbee2mqtt #11527: [Feature request]: allow setting destination endpoint when binding to a group

rpatel3001, 2022-02-20:
### Is your feature request related to a problem? Please describe

I would like to bind my Jasco dimmer switch to a number of Sengled bulbs so that the wall switch controls all the lights in a room. I'm able to bind the switch to each bulb individually, with the source endpoint = 2, destination endpoint = 3, and clusters = LevelCtrl and OnOff. Binding the switch to the group does not work because I can't select the destination endpoint and I assume it defaults to the source endpoint.

### Describe the solution you'd like

Allow choosing the destination endpoint when binding to groups.

### Describe alternatives you've considered

Binding each bulb individually.

### Additional context

I'm not very familiar with the details of how zigbee works so this may not even be possible.

malaakso, 2022-02-21:
Groups don't have their own endpoints. You select the used destination endpoint when adding member to the group.

rpatel3001, 2022-02-21:
Oh that makes sense, I see the /1 endpoints in the group definition in the config file. Is it possible for a device to support binding but not binding to a group then?

malaakso, 2022-02-21:
> Oh that makes sense, I see the /1 endpoints in the group definition in the config file. Is it possible for a device to support binding but not binding to a group then?

Yes. For example IKEA remotes used to support only group binding with old firmware (and not binding to individual devices), but with new FW vice versa, supporting only individual devices.

rpatel3001, 2022-02-21:
Ah OK. I'll open a PR on the enbrighten 43082 page to make a note that it doesn't support group binding. Thanks for your help.

closed 2022-02-21 by rpatel3001