Koenkk/zigbee2mqtt #11528: [Feature request]: Add `last_motion` for motion sensors

0ip, 2022-02-20:
### Is your feature request related to a problem? Please describe

On my home automation dashboard I would like to display the date and time of the last movements my motion sensors have registered. For that, I need an additional layer between Z2M and the dashboard that keeps looking for `occupancy: true` messages and injects a `last_motiont` field based on the previous changes. A similar approach could be taken with contact sensors.

### Describe the solution you'd like

Ideally, Z2M would offer a `last_motion` or `last_occupancy` field for all motion sensors. This feature could be extended to contact sensors as well.

### Describe alternatives you've considered

An alternative would be an additional software that watches for state changes and updates/injects said fields. 

### Additional context

‎

sjorge, 2022-03-24:
This seems like something that should be easy to do with a [user-extensions](https://www.zigbee2mqtt.io/advanced/more/user_extensions.html#user-extensions).

You can hook into onDeviceMessage and if it has the motion or occupancy attribute inject the timestamp before publishing.

0ip, 2022-04-25:
Perhaps someone more knowledgable than me w/ user extensions can pull this off. Maybe one can also think about a fixed integration. @Koenkk @sjorge

0ip, 2022-05-26:
still relevant

0ip, 2022-08-06:
Perhaps someone more knowledgable than me w/ user extensions can pull this off. Maybe one can also think about a fixed integration considering it's a basic and quite useful feature. @Koenkk @sjorge

0ip, 2022-09-10:
Still relevant

closed 2022-10-20 by github-actions[bot]