Koenkk/zigbee2mqtt #11531: Group current_heating_setpoint with TS0601 _TZE200_ckud7u2l not working

copystring, 2022-02-20:
### What happened?

I read here (https://github.com/Koenkk/zigbee2mqtt/issues/2505) it is possible to have two thermostats grouped together and then set current_heating_setpoint to that group. Nice! That really makes things a lot easier to work with a frontend.

So I made a group of two thermostats called HeizungWohnzimmer.
Successfully added both HeizungWohnzimmer1 and HeizungWohnzimmer2.
Published zigbee2mqtt/HeizungWohnzimmer/set/current_heating_setpoint with payload 17 but nothing happens to the thermostats.

Sniffing traffic. Something happens which I fail to understand. Here's a sniff me trying to set zigbee2mqtt/HeizungWohnzimmer/set/current_heating_setpoint with payload 17 and also setting zigbee2mqtt/HeizungWohnzimmer1/set/current_heating_setpoint with payload 17

The devices in question have address 5c:02:72:ff:fe:91:4c:16 (HeizungWohnzimmer1) and 84:2e:14:ff:fe:e0:f8:82 (HeizungWohnzimmer2)

[zigbee_group_thermostat_sniff.zip](https://github.com/Koenkk/zigbee2mqtt/files/8105018/zigbee_group_thermostat_sniff.zip)

let me know if you need anything else.

### Zigbee2MQTT version

1.23.0-dev commit: 7f4f4830

### Adapter firmware version

20220219

### Adapter

Egony V4

copystring, 2022-03-23:
Unstale

copystring, 2022-04-23:
unstale

copystring, 2022-05-05:
@Koenkk do you think you could have a look at this? Do you need anything else?

copystring, 2022-05-06:
Oh. Hm. Alright. How would I know which thermostats support this? Is it only TuYa which don't work?

Koenkk, 2022-05-06:
@copystring also the ones using standard converters might not work. This is because of the nature of groups commands, these commands only stay in the network for X seconds; if the thermostat (which are mostly battery powered) don't wakeup in that time window they will miss the command.

So I recommend not to use group commands for battery powered devices (end devices).

closed 2022-06-14 by github-actions[bot]