Koenkk/zigbee2mqtt #11539: QS-Zigbee-CP03

John1322, 2022-02-21:
### Link

https://www.aliexpress.com/item/1005003703776508.html

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0xa4c138ba3be15fb2","nwkAddr":53296,"manufId":4417,"manufName":"_TZ3000_zirycpws","powerSource":"Mains (single phase)","modelId":"TS130F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[4,5,6,258,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS130F","manufacturerName":"_TZ3000_zirycpws","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"closuresWindowCovering":{"attributes":{"tuyaMovingState":1,"currentPositionLiftPercentage":99,"tuyaMotorReversal":0,"moesCalibrationTime":100,"tuyaCalibration":1}},"genOnOff":{"attributes":{"tuyaBacklightMode":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1645372582826,"defaultSendRequestWhen":"immediate"}

### Comments

QS-Zigbee-CP03 is a successor of almost the same to QS-Zigbee-CP01 (by Lonsonho)

Currently it is identified as TS130F (by Tuya) but this QS-Zigbee-CP03 has no backlight

I had also problems with calibration. I might have fixed it by updataing usb dongle/editing converter (by removing backlight)/reading again instructions

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: [{modelID: 'TS130F', manufacturerName: '_TZ3000_zirycpws'}],
        model: 'QS-Zigbee-C03',
        vendor: 'Tuya',
        description: 'Curtain/blind motor controller',
        fromZigbee: [fz.cover_position_tilt, fz.tuya_cover_options],
        toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.tuya_cover_calibration, tz.tuya_cover_reversal],
        meta: {coverInverted: true},
        exposes: [e.cover_position(), exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']),
            exposes.binary('calibration', ea.ALL, 'ON', 'OFF'), exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF'),
            exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time')],
    };

module.exports = definition;
```

Koenkk, 2022-02-21:
If this converter works, please create a pull request.

MiguelRenato, 2022-03-09:
Is ok
[image]

Koenkk, 2022-03-10:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-10 by Koenkk

galambert75, 2022-03-29:
I tested this device on Zigbee2MQTT Edge (HA addon).

Two hints:
1. The "Model" points to this [link](https://www.zigbee2mqtt.io/devices/QS-Zigbee-C03.html#lonsonho-qs-zigbee-c03), which is a broken link. Note that the model name has a P in there CP03 instead of C03. Also linked in [Blakadder DB](https://zigbee.blakadder.com/Tuya_QS-Zigbee-CP03.html) 
2. The icon shows the standard ZigBee logo instead of the image uploaded by @MiguelRenato

vk496, 2022-08-30:
Could someone confirm me that binding is working properly or not?

I'm interested in buying this one if it have support for binding (I would like to put a entire floor of them in the same group and control them with a remote from IKEA or something like this).

If it doesn't, I would be open to buy something that is not supported yet and make it work with Z2M instead of spending money in a buggy one

MiguelRenato, 2022-08-31:
Many thanks @vk496, I tested the power-reset and it worked as you said. As soon as I have the opportunity I will install it on the wall.

vk496, 2022-09-14:
@MiguelRenato have you been able to verify the binding of those devices? Did it worked?

taduo, 2022-11-22:
When will this be available at normal brach?

`Info 2022-11-22 15:42:35Starting interview of '0xa4c138ceae2fed1b'
Info 2022-11-22 15:42:36MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138ceae2fed1b","ieee_address":"0xa4c138ceae2fed1b","status":"started"},"type":"device_interview"}'
Info 2022-11-22 15:42:36MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138ceae2fed1b"},"type":"pairing"}'
Error 2022-11-22 15:43:05Failed to interview '0xa4c138ceae2fed1b', device has not successfully been paired
Info 2022-11-22 15:43:05MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138ceae2fed1b","ieee_address":"0xa4c138ceae2fed1b","status":"failed"},"type":"device_interview"}'`

taduo, 2022-11-23:
Forget it. It was an issue with my dongle.
It now works on normal brunch.
QS-Zigbee-CP03 is being identified as QS-Zigbee-C03.

bepece1, 2023-05-05:
Is possible setting QS-Zigbee-CP03 devices like “end of device” in Zigbee network??

I have issues with this device and I think the problem is the routing 

thanks in advance

attila-lendvai, 2023-09-26:
> Note that the model name has a P in there CP03 instead of C03.

note that this is still not fixed:

this link works: https://www.zigbee2mqtt.io/devices/QS-Zigbee-C03.html
while this is a broken link: https://www.zigbee2mqtt.io/devices/QS-Zigbee-CP03.html

Animizio, 2023-12-11:
> > Note that the model name has a P in there CP03 instead of C03.
> 
> note that this is still not fixed:
> 
> this link works: https://www.zigbee2mqtt.io/devices/QS-Zigbee-C03.html while this is a broken link: https://www.zigbee2mqtt.io/devices/QS-Zigbee-CP03.html

Why this issue is closed, when the name bug is still not fixed? Just bought some of them and I hope they will work fine in Home-Assistant Zigbee2mqtt