Koenkk/zigbee2mqtt #11540: [New device support]: Zigbee Green Power long press support

gulli1986, 2022-02-21:
### What happened?

Long press is not supported for green power devices like this [one](https://www.zigbee2mqtt.io/devices/PTM_215Z.html), could it be implemented the same way it was for [Hue Dimmer Switch](https://github.com/Koenkk/zigbee2mqtt/issues/585)?

### Zigbee2MQTT version

1.23.0-1

### Adapter firmware version

2022.2.9

### Adapter

Deconz

Koenkk, 2022-02-21:
No, the device doesn't send these events.

gulli1986, 2022-02-21:
OK thanks for the clarification, I guess I will have to make my own automation and try to simulate a long press.

closed 2022-02-21 by gulli1986