Koenkk/zigbee2mqtt #11544: Controlling multiple lights at the same time is unreliable

ikrEErki, 2022-02-21:
### What happened?

Tried to control multiple lights at the same time results in some of them not responding to the request. However, if I immediately after control them manually from the webinterface, they respond as expected. 

I'm using a group in Home Assistant when controlling them simultaneously. And only trying to control around 10 lights at the same time. All with LQI of 255 and connections to each other as well as the coordinator. Mesh has been up around a week and works in both deconz and ZHA, so should be stable by now.

Running in Home Assistant. Everything upgraded to latest version. Have previously run both Deconz and ZHA in same environment, with the same hardware, without issue.

Any suggestions of what might be the issue is highly appreciated.

Default log only gives generic "device did not respond" error:

_Zigbee2MQTT:error 2022-02-21 16:14:26: Publish 'set' 'state' to 'Light1' failed: 'Error: Command 0x680ae2fffe12040f/1 genLevelCtrl.moveToLevelWithOnOff({"level":150,"transtime":5}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2022-02-21 16:14:26: Publish 'set' 'color_temp' to 'Light2' failed: 'Error: Command 0xd0cf5efffe156363/1 lightingColorCtrl.moveToColorTemp({"colortemp":389,"transtime":5}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'_

### What did you expect to happen?

All lights to respond. If network is flooded retry or delay the action for some lights just enough to not flood the network. 

### How to reproduce it (minimal and precise)

- Create a group in HA with around 10 lights.
- Toggle the group
- Watch how only some lights react to the command sent
- Try and toggle the individual light. Voila - works every time!

### Zigbee2MQTT version

1.23.0 commit: unknown

### Adapter firmware version

0x26720700

### Adapter

ConBee2

Koenkk, 2022-02-21:
Try specifying an `adapter_delay`: https://github.com/Koenkk/zigbee2mqtt/issues/11110#issuecomment-1028100246

ikrEErki, 2022-02-21:
wow, that was quick reply and sure sounds like it could work. Will try for a few days and get back to you if it solved the issue. Thanks!

ikrEErki, 2022-02-23:
@Koenkk so 200 in delay did not help. Did increase to 1000 and it looks to have solved most of the issues. Still the occasional error in the logs, but lights does respond a lot more often now!

Did enable debug logging and catched a couple of errors during the 200 delay. Don't know if you have any use for them:

_2022-02-23T07:20:29.833Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0x7 seqNr: 8
2022-02-23T07:20:29.833Z zigbee-herdsman:deconz:driver too many timeouts - restart serial connecion
2022-02-23T07:20:29.834Z zigbee-herdsman:deconz:driver Opening with /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2480326-if00
2022-02-23T07:20:29.841Z zigbee-herdsman:deconz:driver Serialport opened
2022-02-23T07:20:29.894Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 9
2022-02-23T07:20:29.898Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-02-23T07:20:29.898Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1_

_Zigbee2MQTT:debug 2022-02-23 08:25:41: Error: Command 0xbc33acfffe624368/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:641:23)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:20)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:727:28)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:283:17)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:861:40)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)_

Koenkk, 2022-02-23:
It would be better to put the lights into a zigbee group, in this way only 1 command is send to control all lights in the group (instead of 1 per light). https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html

ikrEErki, 2022-02-23:
Yeah, I know. But I would prefer to do all my grouping in HA.

And to be honest, shouldn't we be able to control as few as 10 lights at the same time? I know of much, much, larger installations. Even if ZigBee has a low bandwidth, it isn't that much information that is being passed. Do you think this is an issue with the ConBee only, or a generic ZigBee issue? 

Anyway, the added delay has helped a lot and I'm considering it a good enough workaround for now. Thanks for the quick replies and help! And awesome work with this, it is a real step-up from both Deconz and ZHA.

closed 2022-02-23 by ikrEErki

silkyclouds, 2023-02-04:
hey here, thanks god I found people that have the same issue as me ! 

When I add lights to zigbee groups, create scenes (on/off), and recall these scenes, only a few lights are tuning on or off. 

but I noticed something else that is even more problematic:

I have several rooms (groups) created. But I'd like to create two extra groups : 

Ground floor (contains 3 groups) and all lights (contains all lights of the house) 

What I did notice is that as soon I do add my lights to a "bigger group" I do encounter the issue described above PLUS all the scenes I created for "smaller" groups are disappearing. 

Do anyone here have an idea of why this is happening ? Why can I not simply have bulbs in, lets say, my kitchen group, and also in my "ground floor" group, while keeping an on and off scene for these specific groups ?