Koenkk/zigbee2mqtt #11555: [New device support]: Tradfri GU10

chocolatedrop, 2022-02-21:
### Link

https://www.ikea.com/gb/en/p/tradfri-led-bulb-gu10-345-lumen-wireless-dimmable-white-spectrum-40517647/

### Database entry

Device type Router Zigbee model TRADFRIbulbGU10WS345lm Support status Unsupported  IEEE address 0x50325ffffe272215 Network address 0x6C4F Firmware build date 20201111 Firmware version 1.0.012

### Comments

IKEA Tradfri GU10 345 lumen white spectrum - unsupported device, have had the issue since last year. 

Zigbee2mqtt Edge version worked with the device, but the latest Edge build doesn't work on my HA. Latest standard Z2M does not support it which is a surprise since the Edge one has.

I appreciate the effort that goes into this, thanks. 

### Supported color modes

White spectrum

Koenkk, 2022-02-22:
This device is supported in z2m 1.23.0

chocolatedrop, 2022-02-22:
Hi Koenkk - apologies, I had been using the outdated .io repository. Confirmed working now I'm on 1.23.0.

Thanks very much.

closed 2022-02-22 by chocolatedrop