Koenkk/zigbee2mqtt #11558: Philips Econic

CaliLuke, 2022-02-22:
### Link

https://www.philips-hue.com/en-us/p/hue-white-and-color-ambiance-econic-outdoor-wall-light/1743830V7?origin=p69279316316&gclid=Cj0KCQiAjc2QBhDgARIsAMc3SqSG-WBDcMZUJh3sxCFiacGWlNUtD2wJgQehggPB8vMl8o8XevYfRaQaAnaCEALw_wcB&gclsrc=aw.ds

### Database entry

            "friendly_name": "0x0017880109c2165a",             "ieee_address": "0x0017880109c2165a",             "interview_completed": true,             "interviewing": false,             "manufacturer": "Philips",             "model_id": "1743830V7",             "network_address": 31100,             "power_source": "Mains (single phase)",             "software_build_id": "1.93.7",             "supported": false,             "type": "Router"

### Comments

Looks like there's a new variant of the Econic, the database contains [Philips 1743830P7](https://www.zigbee2mqtt.io/devices/1743830P7.html) but doesn't contain Philips 1743830V7 yet.

Koenkk, 2022-02-23:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

CaliLuke, 2022-02-27:
I get this:

'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

CaliLuke, 2022-02-27:
also here goes the 150x150 image
[image]

Koenkk, 2022-02-28:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-28 by Koenkk