Koenkk/zigbee2mqtt #11561: Switch back to configured channel after touchlink reset

mweinelt, 2022-02-22:
### What happened?

After touchlink resetting a bulb it was paired on the channel it was reset from, which is different from the channel my zigbee network resides on. That  broke connectivity with the rest of my network until after the next Z2M restart.

### What did you expect to happen?

- After touchlink resetting is done the radio should switch back to the configured channel
- Permit join should only work on the configured channel

### How to reproduce it (minimal and precise)

- Permit join
- Touchlink reset on remote channel
- Wait for interview to complete
- Notice that the rest new device works, but the rest of the network does not

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20220219

### Adapter

Slaesh's CC2652RB

### Debug log

https://gist.github.com/mweinelt/e1b6926dda4bfc31d217acbd8f07585a

Koenkk, 2022-02-22:
Does the following work?

- disable joining (if not already)
- touchlink reset device
- permit joining

If yes, the simple solution would be to always disable joining before doing a touchlink operation.

mweinelt, 2022-02-22:
I also tried it in this order and it led to the same problem. Are you confident that the coordinator gets returned to the original channel after touchlink finishes?

And now the problem with retesting it is that all devices that I could touchlink reset are already on channel 20, so it would mean to switch the channel again and repair everything. :tired_face:

Koenkk, 2022-02-23:
> Are you confident that the coordinator gets returned to the original channel after touchlink finishes?

Yes, otherwise you wouldn't be able to control any devices after this.

closed 2022-04-03 by github-actions[bot]