Koenkk/zigbee2mqtt #11567: [New device support]: 915005996801 Hue white ambiance ceiling light Enrave L with Bluetooth

mikkomultanen, 2022-02-22:
### Link

https://www.philips-hue.com/en-gb/p/hue-white-ambiance-enrave-extra-large-ceiling-lamp/4116131P6

### Database entry

I could find this in the UI

### Comments

There are no instructions how to add support for a new device to hassio.

Device type Router
Zigbee Model 915005996801
Zigbee Manufacturer Philips
Support status Not supported
IEEE Address 0x001788010b3f0976
Network address 0xB9E1
Firmware build date 20210305
Firmware version 1.82.10
Interview completed True

Koenkk, 2022-02-22:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

newmy-de, 2022-03-06:
Instructions are on: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

closed 2022-04-15 by github-actions[bot]