Koenkk/zigbee2mqtt #11570: [New device support]: LoraTap garage door opener

barbierp35, 2022-02-22:
### Link

https://fr.aliexpress.com/item/1005003532414239.html?gatewayAdapt=glo2fra&spm=a2g0o.9042311.0.0.44556c374lWe1J

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c138f12c3108ac","nwkAddr":9797,"manufId":4417,"manufName":"_TZE200_wfxuhoea","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId" :81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"b`�)ib`�)ir`�)is`�)i�`�)i�`�)i","65506":54,"65508":0,"stackVersion":0,"dateCode":"","appVersion":70}}},"binds":[{"c luster":0,"type":"endpoint","deviceIeeeAddress":"0x84fd27fffe8238e8","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"con figuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1645353032767,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the supporting new device page but I can't do it.

I find this device which look like mine : https://github.com/Koenkk/zigbee-herdsman-converters/pull/3714
I test to use this PR to operate my device. With this, I can open/close the door (it works aproximatively, one time in two) and the status of the door doesn't go up to Zigbee2MQTT.
In genBasic.attributes, I see special characters. I hope it won't be a problem.

Thank you for your help

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_wfxuhoea'
        },
    ],
    model: 'GDC311ZBQ1',
    vendor: 'Tuya',
    description: 'Garage door opener',
    fromZigbee: [fz.matsee_garage_door_opener, fz.ignore_basic_report, fz.tuya_data_point_dump],
    toZigbee: [tz.matsee_garage_door_opener, tz.tuya_data_point_test],
    // onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    whiteLabel: [{vendor: 'Loratap', model: 'GDC311ZBQ1'}],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        exposes.binary('trigger', ea.ALL, true, false).withDescription('Trigger the door movement'),
        e.action(['trigger']),
        exposes.binary('garage_door_contact', ea.STATE_GET, true, false)
    ],
};

module.exports = definition;
```

Uzdrayne, 2022-03-02:
hello,
I have same device and same problem...
Do you find something ?

barbierp35, 2022-03-02:
> hello, I have same device and same problem... Do you find something ?

No I don't find any solution ...

fsevilla3, 2022-04-01:
I finally received my LoraTap garage door opener with wireless sensor. So far it's working for me using the external converter below. This is very similar to the one posted by @barbierp35 but a bit closer to the merged converter for PJ-ZGD01. Can you test it and let me know if it works for you or if you still have the erratic behaviour (works 50% of the time)?

Note that I don't rely on the `trigger` entity to determine whether it's open or closed. I rely exclusively on `garage_door_contact` for that. For `trigger`, I just toggle it as the relay appears to behave similarly whether you switch that on or off.

Please let me know if this works for you before I go through the process of submitting a pull request.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_wfxuhoea'}],
    model: 'GDC311ZBQ1',
    vendor: 'TuYa',
    description: 'LoraTap garage door opener with wireless sensor',
    fromZigbee: [fz.matsee_garage_door_opener, fz.ignore_basic_report],
    toZigbee: [tz.matsee_garage_door_opener, tz.tuya_data_point_test],
    whiteLabel: [{vendor: 'LoraTap', model: 'GDC311ZBQ1'}],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [exposes.binary('trigger', ea.STATE_SET, true, false).withDescription('Trigger the door movement'),
        exposes.binary('garage_door_contact', ea.STATE_GET, true, false)],
};

module.exports = definition;
```

Uzdrayne, 2022-04-03:
HI, 

Thanks for your post .

 

garage_door_contact : OK everytime ! perfect !! 😊

*	possible to rename true/false to open/close?

 

 

trigger :  True/false it would be nice to make a button based on return state

fsevilla3, 2022-04-03:
Hi @Uzdrayne,

Below is the latest converter that I'm using/testing and will be submitting soon. You should also be using the development branch for this to work with Home Assistant. I'm using the `latest-dev` Docker image, which currently means I'm running Zigbee2Mqtt 1.25.0-dev commit: [b84ad45](https://github.com/Koenkk/zigbee2mqtt/commit/b84ad45).

In Zigbee2Mqtt, it still exposes `True` and `False` but this now has a description of what those mean. In Home Assistant, this shows as `Open` and `Closed` with the garage door icon based on the `garage_door` device class.

### Latest version of `GDC311ZBQ1.js`
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_wfxuhoea'}],
    model: 'GDC311ZBQ1',
    vendor: 'TuYa',
    description: 'LoraTap garage door opener with wireless sensor',
    fromZigbee: [fz.matsee_garage_door_opener, fz.ignore_basic_report],
    toZigbee: [tz.matsee_garage_door_opener, tz.tuya_data_point_test],
    whiteLabel: [{vendor: 'LoraTap', model: 'GDC311ZBQ1'}],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [exposes.binary('trigger', ea.STATE_SET, true, false).withDescription('Trigger the door movement'),
       exposes.binary('garage_door_contact', ea.STATE, false, true).withDescription('Indicates if the garage door contact is closed (= true) or open (= false)')],
};

module.exports = definition;
```

barbierp35, 2022-04-03:
@fsevilla3 Thank you very much for your help !

I tested your latest version of your converter, and now, it works 100% of the time for the button trigger.
But I have an other problem. I see the `trigger` entity but not the `garage_door_contact`. Do you know what is my problem ?
[image]

I'm on the `1.25.0-1 Zigbee2mqtt` version. I'm not on the `Zigbee2mqtt Edge` version, it's for this reason ?

I'm sorry, I'm new to home assistant, so I'm not sure to understand.
If I use the Edge version, I will have your converter directly when Koenkk will merge your requests ?
And if I stay on the "stable" version, I will have your converter in Zigbee2MQTT only when Koenkk will merge the dev branch on master branch ? So it may take a few days or weeks before I can delete the manualy converter `GDC311ZBQ1.js` to use your new converter in the master branch ?
And if I switch to the "Edge" version, I will loose all my devices ?

Thank you

Koenkk, 2022-04-03:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4076

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-03 by Koenkk

fsevilla3, 2022-04-03:
Hi @barbierp35. Your issue with `garage_door_contact` being `N/A` in Zigbee2MQTT looks different to the problem I was trying to  solve with the changes I made to `exposes.binary(...)`.

May I suggest you try the latest development branch by following the steps in https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html. If you're happy to do this, you should also delete `GDC311ZBQ1.js` from the `external_converters` section in `configuration.yaml` (or delete the section entirely) because the dev branch supports this device now. Let us know how you go with that.

You shouldn't have the blue `reload` button at all and you should be getting `True` or `False`, not `N/A`. Here's a screenshot of my Zigbee2MQTT:
[image]

barbierp35, 2022-04-04:
> Hi @barbierp35. Your issue with `garage_door_contact` being `N/A` in Zigbee2MQTT looks different to the problem I was trying to solve with the changes I made to `exposes.binary(...)`.
> 
> May I suggest you try the latest development branch by following the steps in https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html. If you're happy to do this, you should also delete `GDC311ZBQ1.js` from the `external_converters` section in `configuration.yaml` (or delete the `external_converters` section entirely) because the dev branch supports this device now. Testing the latest development branch may of course break some things, but in general you shouldn't lose any of your devices or settings. I'm testing Zigbee2MQTT version 1.25.0-dev commit: [e251fc0](https://github.com/Koenkk/zigbee2mqtt/commit/e251fc0) at the moment and it works fine for me.
> 
> Let me know how you go with that and I'll see if I can assist you if you still have this problem. You shouldn't have the blue `reload` button at all and you should be getting `True` or `False`, not `N/A`. Here's a screenshot of my Zigbee2MQTT: [image]

Thank you for your reply.
I don't use the linux or docker version of zigbee2mqtt, I use the Home Assistant addon.
So there are 2 addons ("Zigbee2mqtt" and "Zigbee2mqtt Edge"). The dev branch is with "Zigbee2mqtt Edge". The problem is, if I want to use the "Zigbee2mqtt Edge" addon to use the dev branch, I must uninstall the "Zigbee2mqtt" addon. So if I uninstall the "Zigbee2mqtt" addon, I will loose all my devices, because it's not like the linux or docker installation.
So I think I will stay on the "Zigbee2mqtt" addon and I will wait the May, the 1st, when Koenkk will merge the dev branch in master. After that, I can tell you if my problem with the "garage_door_contact" is still there.

Uzdrayne, 2022-04-08:
Hi,

Feedback after 1 week:
First version you give is ok for me :

```
    exposes: [exposes.binary('trigger', ea.STATE_SET, true, false).withDescription('Trigger the door movement'),
       exposes.binary('garage_door_contact', ea.STATE, false, true).withDescription('Indicates if the garage door contact is closed (= true) or open (= false)')],

```

For UI I use Custom Button card : https://github.com/custom-cards/button-card

This is my code : 

```
type: custom:button-card
entity: binary_sensor.porte_garage_garage_door_contact
name: >
  [[[ if (entity.state == 'on')  return "Porte garage fermée"; else return
  "Porte garage ouverte" ]]]
show_state: false
state:
  - value: 'on'
    color: blue
    icon: mdi:garage
  - value: 'off'
    color: orange
    icon: mdi:garage-open
tap_action:
  action: call-service
  service: switch.toggle
  service_data:
    entity_id: switch.porte_garage_trigger

```

Hope this can help :)

[image]

[image]

barbierp35, 2022-05-11:
Hi ! I Come back because I update the version of zigbee2Mqtt to get the new change for this device.

The trigger button work perfectly. But, as before with the "manual converter", I don't have the door contact status.
[image]

Do you have any idea to solve this issue please ?

Thanks

barbierp35, 2022-05-23:
Up for my previous request about my issue with the contact door. Someone can help me ? Otherwise I will have to buy an other contact door ...
Thank you !

fsevilla3, 2022-05-23:
Hi @barbierp35,

Apologies for missing your last message. It looks like we should now be on similar versions of Zigbee2MQTT as I'm also back to using the stable release this month, not the development branch. Can you double check that you're on a similar version to mine? I'm on v1.25.1 commit [3f6a137](https://github.com/Koenkk/zigbee2mqtt/commit/3f6a137) and have removed any custom integrations from when I was debugging this.

If you're on a similar version but still get `garage_door_contact = N/A` then I suspect this is an issue with either your garage door opener unit or garage door contact sensor unit. The reason I say this is the integration with Zigbee2MQTT doesn't control the wireless connection between the garage door contact and the garage door opener. That wireless connection is on a different frequency and protocol to Zigbee.

I'm sorry to not have any better news or help for you. Good luck!

Kind regards,
Federico

barbierp35, 2022-05-23:
No problem, thank you for your answer.

I'm also on the last version of master branch, commit `3f6a137` :
[image]

I was thinking maybe I deleted the device wrong before resyncing it. I had deleted the device in Zigbee2MQTT, then all entities in HA before restarting. Did I forget something?

What I also find weird is that in the "Dashboard" tab of Zigbee2MQTT I have all my devices except the garage door one. Do you think it might be related?

Thank you for your help, if ever there is no solution, I will buy a new opening sensor. At least the trigger works well and that's the most important.

fsevilla3, 2022-05-23:
Hi @barbierp35,

It does sound weird that you don't see the device in your dashboard, so you may want to try removing (and if needed force removing) the device from Zigbee2MQTT then re-pairing it. Alternatively have a look at your logs to see if there's anything about this device.

Kind regards,
Federico

barbierp35, 2022-05-23:
Thank you @fsevilla3, but I have already forced the deletion, then delete all entity, then pair the device. And everytime, it's the same problem. I see my device with a good trigger and linkquality. But the garage_door_contact value is `unknow` and I don't see the device on the Zigbee2MQTT `dashboard`.
Could there be a MQTT cache problem ?

barbierp35, 2022-05-23:
@fsevilla3 I will look logs one more time.
Thank you for your help !

grest, 2023-02-13:
> @fsevilla3 I will look logs one more time. Thank you for your help !

Hi @barbierp35! Have you managed to fix the issue? I've the same problem :(. Device reports status perfectly with Tuya Gateway, but unfortunately it doesn't work in Zigbee2MQTT: [https://github.com/Koenkk/zigbee2mqtt/issues/16656](url)

barbierp35, 2023-02-14:
> Hi @barbierp35! Have you managed to fix the issue? I've the same problem :(. Device reports status perfectly with Tuya Gateway, but unfortunately it doesn't work in Zigbee2MQTT: [...]

Hi @grest ! I didn't resolve my issue. I think I will buy an other door detector but it will not link to the door interruptor

grest, 2023-02-14:
@barbierp35, do you think it is related to strange characters following 65503?

barbierp35, 2023-02-14:
> @barbierp35, do you think it is related to strange characters following 65503?

Sorry but I don't know how we can fix our issue. If you want, you can read the doc and test your solution : https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

grest, 2023-02-14:
This is exactly what I did :) I've added additional logging and it's either empty or returns 0 value. By the way - I was wrong, it doesn't work with Tuya Zigbee Gateway and App as well. I've asked LoraTap for support, perhaps device is faulty.

barbierp35, 2023-02-15:
> This is exactly what I did :) I've added additional logging and it's either empty or returns 0 value. By the way - I was wrong, it doesn't work with Tuya Zigbee Gateway and App as well. I've asked LoraTap for support, perhaps device is faulty.

@grest Can you let me know if you have any news to make this device work please ?

grest, 2023-02-15:
Yes. I've asked LoraTap support for a help and actually they've helped me! You need to keep pressing the reset button a bit longer to make a led button lighting with blue color for a moment and in the meantime open close the wireless sensor - this forces repairing procedure. In my case it caused the device disappear to automatically from the devices list in zigbee2mgtt and after a couple of seconds it paired again, this time with the sensor reporting correctly its status. I can send you the movie.

barbierp35, 2023-02-15:
> Yes @barbierp35 . I've asked LoraTap support for a help and actually they've helped me! You need to keep pressing the reset button a bit longer to make a led button lighting with blue color for a moment and in the meantime open close the wireless sensor - this forces repairing procedure. In my case it caused the device disappear to automatically from the devices list in zigbee2mgtt and after a couple of seconds it paired again, this time with the sensor reporting correctly its status. I can send you the movie.

Good news if you found it! 
I tried to push the reset button on the red case.  After a while it flashed blue and red.  I stop to push the button, open the garage door, close it and thats all ?
If it's that, it doesn't work for me ...

grest, 2023-02-17:
In the theory it should work. Please unmount the wireless sensor and play with it manually - it's better to generate a lot of signals during the pairing procedure. Please give me your email. I'll send you the manual provided by LoraTap