Koenkk/zigbee2mqtt #11576: TuYa [TV02] Wrong current_heating_setpoint after switching to manual preset mode

Udeson, 2022-02-22:
### What happened?

Using Zigbee Thermostat radiator valve `Model TS0601 `Zigbee `Manufacturer _TZE200_e9ba97vf` recognized as TuYa [TV02-Zigbee](https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee).

When switching from preset **auto** to **manual** with a payload like `{"preset":"manual","current_heating_setpoint":"25.0"}` the current_heating_setpoint changes to the last known current_heating_setpoint set in manual preset mode before instead of the value of the payload.

### What did you expect to happen?

Manual preset mode should adjust current_heating_setpoint corresponding to the payload value instead of last known current_heating_setpoint  value from manual preset mode.

### How to reproduce it (minimal and precise)

Via frontend or MQTT cli:
- In preset mode auto switch to preset manual mode without setting current_heating_setpoint value and look for changed current_heating_setpoint value
or
- In preset mode auto switch to preset manual mode with set current_heating_setpoint value and look for changed current_heating_setpoint value

### Zigbee2MQTT version

 version 1.23.0 commit: 6ea61bae

### Adapter firmware version

20211217

### Adapter

Electrorama zzh

jypsie, 2022-02-24:
I have a related? problem. My TV02 stays offline.

[image]

Udeson, 2022-02-24:
> I have a related? problem. My TV02 stays offline.

What is your understanding of "stays offline"?

What is also crappy on the device behaviour is that you cannot change `current_heating_setpoint in preset: auto mode`... You can adjust temperature on the device itself (without changing preset mode) but not via mqtt command (setting current_heating_setpoint triggers change to preset: manual)...

Accidentally I saw in the OEM manual for my device that the device's model name is `TV01` not `TV02`, so maybe all this is an implementation issue in zigbee2mqtt....

jypsie, 2022-02-24:
Sorry, I mean disabled see picture, Is this normal?

Udeson, 2022-02-24:
Mine is showing disabled as well...

Udeson, 2022-02-24:
As a first ugly workaround I switched order of lines (for tvMode and tvHeatingSetpoint in /converters/toZigbee.js to:

```
case 'current_heating_setpoint':
await tuya.sendDataPointEnum(entity, tuya.dataPoints.tvMode, 1 /* manual */);
await utils.sleep(500);
await tuya.sendDataPointValue(entity, tuya.dataPoints.tvHeatingSetpoint, value * 10);
break;
```

Udeson, 2022-02-28:
Moreover I have issues with heating at all. Seems like the internal temperature sensor of the device needs to be minimum >4-5 °C bigger than the desired temperature of the device before valve opens.
So in all the device is almost useless to me... shit happens ;-)

closed 2022-04-08 by github-actions[bot]