Koenkk/zigbee2mqtt #11577: [Problem report]: Awox Eglo connect.2 tunable white bulb mis-identified

chedda61, 2022-02-23:
### What happened?

Paired this bulb https://www.eglo.com/en/bulb-12577.html
which should have color temperature control, but does not in zigbee2mqtt (it's there in zha).  It's identified as 33951/33948 instead of 33957. The fingerprint matches that of 33957 except for different key names:
profId <> profileID, devId <> deviceID, inClusterList <> inputClusters, etc.
The db entry is:
{
    "id": 12,
    "type": "Router",
    "ieeeAddr": "0xa4c138db1d17b039",
    "nwkAddr": 2008,
    "manufId": 4417,
    "manufName": "AwoX",
    "powerSource": "Mains (single phase)",
    "modelId": "TLSR82xx",
    "epList":
    [
        1,
        242,
        3
    ],
    "endpoints":
    {
        "1":
        {
            "profId": 260,
            "epId": 1,
            "devId": 268,
            "inClusterList":
            [
                0,
                3,
                4,
                5,
                6,
                8,
                768,
                4096,
                64599
            ],
            "outClusterList":
            [
                6
            ],
            "clusters":
            {
                "genBasic":
                {
                    "attributes":
                    {
                        "modelId": "TLSR82xx",
                        "manufacturerName": "AwoX",
                        "powerSource": 1,
                        "zclVersion": 3,
                        "appVersion": 0,
                        "stackVersion": 2,
                        "hwVersion": 0,
                        "swBuildId": "0122052017"
                    }
                },
                "lightingColorCtrl":
                {
                    "attributes":
                    {
                        "colorCapabilities": 16,
                        "colorTempPhysicalMin": 153,
                        "colorTempPhysicalMax": 454
                    }
                },
                "genOnOff":
                {
                    "attributes":
                    {
                        "onOff": 1
                    }
                },
                "genLevelCtrl":
                {
                    "attributes":
                    {
                        "currentLevel": 156
                    }
                }
            },
            "binds":
            [],
            "configuredReportings":
            [],
            "meta":
            {}
        },
        "3":
        {
            "profId": 4751,
            "epId": 3,
            "devId": 268,
            "inClusterList":
            [
                65360,
                65361
            ],
            "outClusterList":
            [
                65360,
                65361
            ],
            "clusters":
            {},
            "binds":
            [],
            "configuredReportings":
            [],
            "meta":
            {}
        },
        "242":
        {
            "profId": 41440,
            "epId": 242,
            "devId": 97,
            "inClusterList":
            [],
            "outClusterList":
            [
                33
            ],
            "clusters":
            {},
            "binds":
            [],
            "configuredReportings":
            [],
            "meta":
            {}
        }
    },
    "appVersion": 0,
    "stackVersion": 2,
    "hwVersion": 0,
    "swBuildId": "0122052017",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta":
    {
        "configured": 88764544
    },
    "lastSeen": 1645567772135,
    "useImplicitCheckin": true
}

### What did you expect to happen?

Being able to control color temperature

### How to reproduce it (minimal and precise)

just pair the device

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

SMARTLIGHT SLZB-02

chedda61, 2022-02-23:
I updated earlier today, still same behaviour.

Koenkk, 2022-02-23:
Pushed a fix, can you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

chedda61, 2022-02-23:
pulled the latest-dev docker image, but the container keeps restarting.
Switching back to 1.23.0, the container will start without problems.
This is one of the logs of the dev container:
debug 2022-02-23 19:29:57: Loaded state from file /app/data/state.json
info  2022-02-23 19:29:57: Logging to console and directory: '/app/data/log/2022-02-23.19-29-55' filename: log.txt
debug 2022-02-23 19:29:57: Removing old log directory '/app/data/log/2022-02-12.16-18-23'
info  2022-02-23 19:29:57: Starting Zigbee2MQTT version 1.23.0-dev (commit #9c33c90)
info  2022-02-23 19:29:57: Starting zigbee-herdsman (0.14.18)
debug 2022-02-23 19:29:57: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB1"}}'

Koenkk, 2022-02-23:
Please try again in two hours, the build is still running

chedda61, 2022-02-23:
Just added profile manually into the container, it works, many thanks.

closed 2022-02-23 by chedda61