Koenkk/zigbee2mqtt #11578: [New device support]: Sylvania BR30 TW (advertising as LEDVANCE)

lukeallister, 2022-02-23:
### Link

https://www.amazon.com/gp/product/B015KQ27SG/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0xf0d1b8000002b49f","nwkAddr":34591,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Mains (single phase)","modelId":"BR30 TW","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,8,768,2821,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"BR30 TW","manufacturerName":"LEDVANCE","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":0,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":0,"dateCode":"","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1645580214189}

### Comments

This was supposed to be the same as the[ already supported sylvania 73740](https://www.zigbee2mqtt.io/devices/73740.html) but it is showing up as not recognized:

```
warn  2022-02-22 20:30:45: Received message from unsupported device with Zigbee model 'BR30 TW' and manufacturer name 'LEDVANCE'
warn  2022-02-22 20:30:45: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2022-02-22 20:30:48: Received Zigbee message from 'sylvania_bulb_0', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":1057025,"imageType":26,"manufacturerCode":4489}' from endpoint 1 with groupID 0
```

I tried to adapt the "LIGHTIFY BR Tunable White" device description from [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/sylvania.js) but it crashed. 

I tried both using `extend: extend.ledvance.light_onoff_brightness_colortemp(),' and then also 'extend: extend.light_onoff_brightness_colortemp(),` but neither works.

Thanks for all your work on this project!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['BR30 TW'], 
    model: '73740', 
    vendor: 'LEDVANCE', 
    description: 'Sylvania BR30 TW', 
    extend: extend.light_onoff_brightness_colortemp(),
};

module.exports = definition;
```

### Supported color modes

color temperature

Koenkk, 2022-02-23:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-23 by Koenkk

lukeallister, 2022-02-23:
Initially I wasn't able to use the dev branch (in docker, on a raspberry pi3) but this fix, plus the changes pushed from #11545 made it work.

Thanks!