Koenkk/zigbee2mqtt #11579: Pairing fails with customer <device_name>.js file

boriken72, 2022-02-23:
### What happened?

Constantly getting:

zigbee2mqtt    | Zigbee2MQTT:debug 2022-02-22 22:50:19: No converter available for 'WF4CRDTUWHMWM6' with cluster 'lightingColorCtrl' and type 'readResponse' and data '{"colorCapabilities":24}'
zigbee2mqtt    | Zigbee2MQTT:debug 2022-02-22 22:50:19: Received Zigbee message from '0x88571dfffe5dc80e', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorTempPhysicalMax":370,"colorTempPhysicalMin":200}' from endpoint 8 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2022-02-22 22:50:19: No converter available for 'WF4CRDTUWHMWM6' with cluster 'lightingColorCtrl' and type 'readResponse' and data '{"colorTempPhysicalMax":370,"colorTempPhysicalMin":200}'

Even though it's defined in a customer .js file:

{
    zigbeeModel: ['ABL-LIGHT-Z-001'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'WF4CRDTUWHMWM6', // Vendor model number, look on the device for a model number
    vendor: 'Samsung Electronics', // Vendor of the device (only used for documentation and startup logging)
    description: 'Juno recessed light by Samsung', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [370, 200]}), // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
},

### What did you expect to happen?

That since the device is manually defined the converter would be used.

### How to reproduce it (minimal and precise)

Remove devices that constantly fail to get interviewed and re-add them.

### Zigbee2MQTT version

 1.23.0 commit: 6ea61ba

### Adapter firmware version

20220219

### Adapter

Electrolama zig-a-zig-ah! CC2652RB stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8121827/log.txt)

sjorge, 2022-02-23:
```
fromZigbee: [fz.on_off], // We will add this later
toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
extend: extend.light_onoff_brightness_colortemp({colorTempRange: [370, 200]}), // Defines what this device exposes, 
```

Your problem is here, you only set an `on_off` converter, yet you extend from `light_onoff_brightness_colortemp`, IIRC you can just drop the toZibgee and fromZigbee and it should work. Unless the light only supports on/off, then you need to extend from the correct light_* extend or not at all.

boriken72, 2022-02-23:
This is configuration file now – still fails to interview:

 

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');

const tz = require('zigbee-herdsman-converters/converters/toZigbee');

const exposes = require('zigbee-herdsman-converters/lib/exposes');

const constants = require('zigbee-herdsman-converters/lib/constants');

const reporting = require('zigbee-herdsman-converters/lib/reporting');

const extend = require('zigbee-herdsman-converters/lib/extend');

const e = exposes.presets;

const ea = exposes.access;

 

const definition = [

{

    zigbeeModel: ['ABL-LIGHT-Z-001'], // The model ID from: Device with modelID 'lumi.sens' is not supported.

    model: 'WF4CRDTUWHMWM6', // Vendor model number, look on the device for a model number

    vendor: 'Samsung Electronics', // Vendor of the device (only used for documentation and startup logging)

    description: 'Juno recessed light by Samsung', // Description of the device, copy from vendor site. (only used for documentation and startup logging)

    //fromZigbee: [fz.on_off], // We will add this later

    //toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).

    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [370, 200]}), // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend

},

];

module.exports = definition;

boriken72, 2022-02-23:
This is one of the lights:

 

zigbee2mqtt    | Zigbee2MQTT:warn  2022-02-23 08:27:39: Device '0x88571dfffe5d7c5e' with Zigbee model 'undefined' and manufacturer name 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

sjorge, 2022-02-23:
I don't think it's even finding the converter now? because all the info of the device are undefined:
`Zigbee model 'undefined' and manufacturer name 'undefined'`

boriken72, 2022-02-23:
Can you confirm this is correct format for multiple entries (I had removed it earlier since the singled I added to get battery percentage is working)

 

 

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');

const tz = require('zigbee-herdsman-converters/converters/toZigbee');

const exposes = require('zigbee-herdsman-converters/lib/exposes');

const constants = require('zigbee-herdsman-converters/lib/constants');

const reporting = require('zigbee-herdsman-converters/lib/reporting');

const extend = require('zigbee-herdsman-converters/lib/extend');

const e = exposes.presets;

const ea = exposes.access;

 

const definition = [

{

    zigbeeModel: ['ABL-LIGHT-Z-001'], // The model ID from: Device with modelID 'lumi.sens' is not supported.

    model: 'WF4CRDTUWHMWM6', // Vendor model number, look on the device for a model number

    vendor: 'Samsung Electronics', // Vendor of the device (only used for documentation and startup logging)

    description: 'Juno recessed light by Samsung', // Description of the device, copy from vendor site. (only used for documentation and startup logging)

    //fromZigbee: [fz.on_off], // We will add this later

    //toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).

    extend: extend.light_onoff_brightness_colortemp(), // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend

},

{

    zigbeeModel: ['E1D-G73'],

    model: 'E1D-G73WNA',

    vendor: 'Sengled',

    description: 'Smart window and door sensor',

    fromZigbee: [fz.ias_contact_alarm_1, fz.battery],

    toZigbee: [],

    exposes: [e.contact(), e.battery_low(), e.tamper(), e.battery()],

    configure: async (device, coordinatorEndpoint, logger) => {

        const endpoint = device.getEndpoint(1);

        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);

        await reporting.batteryPercentageRemaining(endpoint);

    },

    meta: {battery: {voltagePercentage: '3V_2500'}},

},

];

module.exports = definition;

sjorge, 2022-02-23:
The syntax looks correct to me.

boriken72, 2022-02-23:
Also interesting this is that in logs it says it fails to pair:

 

error 2022-02-23 08:23:38: Failed to interview '0x88571dfffe5d7bf8', device has not successfully been paired

info  2022-02-23 08:23:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x88571dfffe5d7bf8","ieee_address":"0x88571dfffe5d7bf8","status":"failed"},"type":"device_interview"}'

info  2022-02-23 08:23:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x88571dfffe5d7bf8"},"type":"pairing"}'

debug 2022-02-23 08:29:16: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '

 

but on dashboard it shows like it pairs successfully, but they are not controllable:

 

 

I have another set of the exact same lights which paired without issue and are controllable:

boriken72, 2022-02-23:
> So you have 2 of these devices and 1 paired and works the 2nd one doesn't?

No, sorry I should have been clearer. I have 4 of these devices – 2 are already paired (bottom named set) – I have another pair (top two) which show to be paired but are not controllable – as in I shut them on/off but the light doesn’t actually go on or off.

 

--Juan

boriken72, 2022-03-21:
any update on this?

davbay1, 2022-04-25:
> any update on this?

Hi,

I just bought a couple of these lights and am trying to get them setup. Do you by chance still have the configuration file for them and would you possibly mind sharing it with me?

I really appreciate it.

boriken72, 2022-06-03:
Curious why this was closed it was never addressed.

ludeatude, 2022-08-16:
Can this be reopened?  I have 2 of these lights and would like to pair it with HA.

CattDamon, 2024-02-01:
These are still broke and dont work in Z2M for anyone looking to buy them.

bapesupreme, 2025-02-25:
> These are still broke and dont work in Z2M for anyone looking to buy them.

are you referring to ABL-LIGHT-Z-001 / WF4C_WF6C? I'm having issues pairing them in Z2M. Some pair easily then don't work. Repairing may fix it, then others break.