Koenkk/zigbee2mqtt #11586: RTCGQ13LM not working

kusok18, 2022-02-23:
### What happened?

Sorry for my English, it's not my native language.
RTCGQ13LM does not work.
The "occupancy" entity is always in the "unknown" state.
I reconfigured the device many times, reinstalled the add-on.
Nothing helps, help revive these sensors.
Before the latest update they worked great.

### Zigbee2MQTT version

1.23.0-1

### Adapter firmware version

20200211

### Adapter

Texas_Instruments_CC2538_USB

Koenkk, 2022-02-23:
You need a CC2538 firmware with a Xiaomi fix in it. @0x3ec is this already available?

kusok18, 2022-02-23:
[image]
[image]
[image]

kusok18, 2022-02-23:
> Вам нужна прошивка CC2538 с исправлением Xiaomi. @0x3EC это уже доступно?

Tell me, can I flash this whistle without additional equipment?

Koenkk, 2022-02-24:
I think you need a JTag flasher

Otnow, 2022-03-13:
As a workaround, you can use the [previous version of the converter](https://github.com/Koenkk/zigbee2mqtt/issues/5203#issuecomment-1006955549).

phobosdigg, 2022-03-28:
I had the same problem as kusok18 and the Otnow method solved my problem. Thanks!

closed 2022-05-06 by github-actions[bot]