Koenkk/zigbee2mqtt #11594: [New device support]: TS0601_cover - M515EGBZTN (new version)

tenkaibr, 2022-02-24:
### Link

https://pl.aliexpress.com/item/4001343586623.html

### Database entry

{"id":18,"type":"EndDevice","ieeeAddr":"0xb4e3f9fffeaf50cd","nwkAddr":47610,"manufId":4098,"manufName":"_TZE200_gubdgai2","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_gubdgai2","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1645708551258,"defaultSendRequestWhen":"immediate"}

### Comments

Zenismart has created a new version that forces you to buy the HUB to configure the limits. They removed the option to configure with the physical buttons. So, would it be possible to upgrade to configure by z2m?

With the external configuration it is possible to open, close and stop. But it is not possible to reverse the motor, it does not display the battery or status information. And cant configure limits.

[image]

Log:

> Zigbee2MQTT:info  2022-02-24 10:15:49: Device '0xb4e3f9fffeaf50cd' joined
Zigbee2MQTT:info  2022-02-24 10:15:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeaf50cd","ieee_address":"0xb4e3f9fffeaf50cd"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-02-24 10:15:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xb4e3f9fffeaf50cd"},"type":"device_connected"}'
Zigbee2MQTT:info  2022-02-24 10:15:49: Starting interview of '0xb4e3f9fffeaf50cd'
Zigbee2MQTT:info  2022-02-24 10:15:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeaf50cd","ieee_address":"0xb4e3f9fffeaf50cd","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-02-24 10:15:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xb4e3f9fffeaf50cd"},"type":"pairing"}'
Zigbee2MQTT:debug 2022-02-24 10:15:50: Device '0xb4e3f9fffeaf50cd' announced itself
Zigbee2MQTT:info  2022-02-24 10:15:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffeaf50cd","ieee_address":"0xb4e3f9fffeaf50cd"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-02-24 10:15:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xb4e3f9fffeaf50cd"},"type":"device_announced"}'
Zigbee2MQTT:debug 2022-02-24 10:15:50: Received Zigbee message from '0xb4e3f9fffeaf50cd', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_gubdgai2","modelId":"TS0601"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-24 10:15:50: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-02-24 10:15:50: Received Zigbee message from '0xb4e3f9fffeaf50cd', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-24 10:15:50: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-02-24 10:15:50: Received Zigbee message from '0xb4e3f9fffeaf50cd', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-24 10:15:50: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-02-24 10:15:51: Received Zigbee message from '0xb4e3f9fffeaf50cd', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-24 10:15:51: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info  2022-02-24 10:15:51: Successfully interviewed '0xb4e3f9fffeaf50cd', device has successfully been paired
Zigbee2MQTT:warn  2022-02-24 10:15:51: Device '0xb4e3f9fffeaf50cd' with Zigbee model 'TS0601' and manufacturer name '_TZE200_gubdgai2' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:info  2022-02-24 10:15:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xb4e3f9fffeaf50cd","ieee_address":"0xb4e3f9fffeaf50cd","status":"successful","supported":false},"type":"device_interview"}'
Zigbee2MQTT:info  2022-02-24 10:15:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xb4e3f9fffeaf50cd","supported":false},"type":"pairing"}'
Zigbee2MQTT:debug 2022-02-24 10:15:55: Received Zigbee message from '0xb4e3f9fffeaf50cd', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2022-02-24 10:15:55: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_gubdgai2'
Zigbee2MQTT:warn  2022-02-24 10:15:55: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2022-02-24 10:15:58: Received Zigbee message from '0xb4e3f9fffeaf50cd', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1}],"seq":790}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2022-02-24 10:15:58: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_gubdgai2'
Zigbee2MQTT:warn  2022-02-24 10:15:58: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2022-02-24 10:15:58: Received Zigbee message from '0xb4e3f9fffeaf50cd', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":7}],"seq":1046}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2022-02-24 10:15:58: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_gubdgai2'
Zigbee2MQTT:warn  2022-02-24 10:15:58: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2022-02-24 10:15:59: Received Zigbee message from '0xb4e3f9fffeaf50cd', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2}],"seq":1302}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2022-02-24 10:15:59: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_gubdgai2'
Zigbee2MQTT:warn  2022-02-24 10:15:59: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2022-02-24 10:15:59: Received Zigbee message from '0xb4e3f9fffeaf50cd', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":5}],"seq":1558}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2022-02-24 10:15:59: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_gubdgai2'
Zigbee2MQTT:warn  2022-02-24 10:15:59: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2022-02-24 10:16:00: Received Zigbee message from '0xb4e3f9fffeaf50cd', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":45334}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2022-02-24 10:16:00: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_gubdgai2'
Zigbee2MQTT:warn  2022-02-24 10:16:00: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2022-02-24 10:18:17: Saving state to file /config/zigbee2mqtt/state.json

### External converter

```shell
const fz = require( 'zigbee-herdsman-converters/converters/fromZigbee' );
const tz = require( 'zigbee-herdsman-converters/converters/toZigbee' );
const exposes = require( 'zigbee-herdsman-converters/lib/exposes' );
const reporting = require( 'zigbee-herdsman-converters/lib/reporting' );
const extend = require( 'zigbee-herdsman-converters/lib/extend' );
const tuya = require( 'zigbee-herdsman-converters/lib/tuya' );
const e = exposes.presets;
const ea = exposes.access;
[... 5 lines cut ...]
            ],
        fingerprint: [
[... cut ...]
        description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Tuya', model: 'M515EGZT'},
            ],
        fromZigbee: [
            fz.tuya_cover,
            fz.ignore_basic_report,
            ],
        toZigbee: [
            tz.tuya_cover_control,
            tz.tuya_cover_options,
            ],
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            ],
    };

module.exports = definition;
```

Burki24, 2022-02-25:
@tenkaibr 
Will you try the following external converter?
Maybe it will run, otherwise you need to sniff the new datapoints

```js
const fz = require( 'zigbee-herdsman-converters/converters/fromZigbee' );
const tz = require( 'zigbee-herdsman-converters/converters/toZigbee' );
const exposes = require( 'zigbee-herdsman-converters/lib/exposes' );
const reporting = require( 'zigbee-herdsman-converters/lib/reporting' );
const extend = require( 'zigbee-herdsman-converters/lib/extend' );
const tuya = require( 'zigbee-herdsman-converters/lib/tuya' );
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: [
            'owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja', // Curtain motors
            'mcdj3aq', 'mcdj3aq\u0000', // Tubular motors
            ],
        fingerprint: [
            { modelID: 'TS0601', manufacturerName: '_TZE200_gubdgai2' },
            ],
        model: 'TS0601_cover',
        vendor: 'TuYa',
        description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Tuya', model: 'M515EGZT'},
            ],
        fromZigbee: [
            fz.tuya_cover,
            fz.ignore_basic_report,
            fz.ZMAM02_cover,  // datapoints of Zemismart RF-Converter
            fz.battery,
            ],
        toZigbee: [
            tz.tuya_cover_control,
            tz.tuya_cover_options,
            tz.ZMAM02_cover, // datapoints of Zemismart RF-Converter
            ],
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.composite('options', 'options')
            .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(255)
                .withDescription('Motor speed')),
            exposes.enum('motor_working_mode', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02MotorWorkingMode)), // Continuos/steps
            exposes.numeric('percent_state', ea.STATE).withValueMin(0).withValueMax(100).withValueStep(1).withUnit('%'),
            exposes.enum('motor_direction', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Direction)), // forward/backward
            exposes.enum('border', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Border)), // Top/Bottom Endstops
            // ---------------------------------------------------------------------------------
            // DP exists, but not used at the moment
            // exposes.numeric('percent_control', ea.STATE_SET).withValueMin(0).withValueMax(100).withValueStep(1).withUnit('%'),
            // exposes.enum('work_state', ea.STATE, Object.values(tuya.ZMAM02.AM02WorkState)),
            // exposes.numeric('countdown_left', ea.STATE).withUnit('s'),
            // exposes.numeric('time_total', ea.STATE).withUnit('ms'),
            // exposes.enum('situation_set', ea.STATE, Object.values(tuya.ZMAM02.AM02Situation)),
            ],
    };

module.exports = definition;
```

tenkaibr, 2022-02-25:
> @tenkaibr Will you try the following external converter? Maybe it will run, otherwise you need to sniff the new datapoints

Thanks for the quick response!

I tested the code above but it seems there is something wrong that I couldn't identify.

The server just doesn't start. Here's the error:

> zigbee2mqtt@1.23.0 start
node index.js
(node:361) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'AM02MotorWorkingMode' of undefined
(Use `node --trace-warnings ...` to show where the warning was created)
(node:361) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:361) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.`

Burki24, 2022-02-26:
> > @tenkaibr Will you try the following external converter? Maybe it will run, otherwise you need to sniff the new datapoints
> 
> Thanks for the quick response!
> 
> I tested the code above but it seems there is something wrong that I couldn't identify.
> 
> The server just doesn't start. Here's the error:
> 
> > zigbee2mqtt@1.23.0 start
> > node index.js
> > (node:361) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'AM02MotorWorkingMode' of undefined
> > (Use `node --trace-warnings ...` to show where the warning was created)
> > (node:361) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
> > (node:361) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.`
> 
> The strange thing is that I'm not getting any mqtt messages when using the physical buttons.

Hmm, at the moment, I'm not sure. I tried the external converter and it runs without any errors. Are u sure, that you've copied all things right? 
It looks like, there is a mistake getting the "AM02MotorWorkingMode" inside the /lib/tuya.js. Not sure, why.

Can you please take a look inside /lib/tuya.js searching the string "AM02MotorWorkingMode" there have to be two entries:

```js
AM02MotorWorkingMode: 20,
...
const ZMLookups:
...
AM02MotorWorkingMode: {
        0: 'continuous',
        1: 'intermittently',
    },
```

Burki24, 2022-02-27:
Okay, motor_working_mode is used for some covers to change from continue running to intermittend run. So it is not for setting up motor_speed. 

AS you use some parts of TuYa-cover, there is no DP20 or DP16 known. For that, you can delete the following parts:

```js
fz.tuya_cover_control
fz.tuya_cover_options
```
and:
```js
tz.tuya_cover_control
tz.tuya_cover_options. 
```
these functions are integrated into:

```js
fz.ZMAM02_cover
tz.ZMAM02_cover
```
So, here is the update (with battery, hoefully, it will run):

```js
const fz = require( 'zigbee-herdsman-converters/converters/fromZigbee' );
const tz = require( 'zigbee-herdsman-converters/converters/toZigbee' );
const exposes = require( 'zigbee-herdsman-converters/lib/exposes' );
const reporting = require( 'zigbee-herdsman-converters/lib/reporting' );
const extend = require( 'zigbee-herdsman-converters/lib/extend' );
const tuya = require( 'zigbee-herdsman-converters/lib/tuya' );
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: [
            'owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja', // Curtain motors
            'mcdj3aq', 'mcdj3aq\u0000', // Tubular motors
            ],
        fingerprint: [
            { modelID: 'TS0601', manufacturerName: '_TZE200_iossyxra' },
            ],
        model: 'TS0601_cover',
        vendor: 'TuYa',
        description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Tuya', model: 'M515EGZT'},
            ],
        fromZigbee: [
            fz.ZMAM02_cover
            ],
        toZigbee: [
            tz.ZMAM02_cover,
            ],
        exposes: [e.battery(),
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.composite('options', 'options')
            .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(255)
                .withDescription('Motor speed')),
            exposes.enum('motor_working_mode', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02MotorWorkingMode)),
            exposes.numeric('percent_state', ea.STATE).withValueMin(0).withValueMax(100).withValueStep(1).withUnit('%'),
            exposes.enum('mode', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Mode)),
            exposes.enum('motor_direction', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Direction)),
            exposes.enum('border', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Border)),
            // ---------------------------------------------------------------------------------
            // DP exists, but not used at the moment
            // exposes.numeric('percent_control', ea.STATE_SET).withValueMin(0).withValueMax(100).withValueStep(1).withUnit('%'),
            // exposes.enum('work_state', ea.STATE, Object.values(tuya.ZMAM02.AM02WorkState)),
            // exposes.numeric('countdown_left', ea.STATE).withUnit('s'),
            // exposes.numeric('time_total', ea.STATE).withUnit('ms'),
            // exposes.enum('situation_set', ea.STATE, Object.values(tuya.ZMAM02.AM02Situation)),
            ],
    };

module.exports = definition;

```

tenkaibr, 2022-02-28:
Unfortunately the battery status is not working. So i removed. Below is the final code.

Thank you very much!

What should I do to include this device in the next updates?

```
const fz = require( 'zigbee-herdsman-converters/converters/fromZigbee' );
const tz = require( 'zigbee-herdsman-converters/converters/toZigbee' );
const exposes = require( 'zigbee-herdsman-converters/lib/exposes' );
const reporting = require( 'zigbee-herdsman-converters/lib/reporting' );
const extend = require( 'zigbee-herdsman-converters/lib/extend' );
const tuya = require( 'zigbee-herdsman-converters/lib/tuya' );
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: [
            'owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja',
            'mcdj3aq', 'mcdj3aq\u0000',
            ],
        fingerprint: [
            { modelID: 'TS0601', manufacturerName: '_TZE200_gubdgai2' },
            ],           
        model: 'TS0601_cover',
        vendor: 'Zemismart',
        description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Tuya', model: 'M515EGZT'},
            ],
        fromZigbee: [
            fz.ZMAM02_cover
            ],
        toZigbee: [
            tz.ZMAM02_cover,
            ],
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.enum('motor_direction', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Direction)),
            exposes.enum('border', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Border)),
            ],
    };

module.exports = definition;
```

Burki24, 2022-03-01:
Hi @tenkaibr,

you only need to put the following lines into zemsmart.js under: /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices:

```js
{
        zigbeeModel: ['owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja', 'mcdj3aq', 'mcdj3aq\u0000'],
        fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_gubdgai2' }],           
        model: 'M515EGBZTN',
        vendor: 'Zemismart',
        description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
        fromZigbee: [fz.ZMAM02_cover],
        toZigbee: [tz.ZMAM02_cover,],
        exposes: [e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.enum('motor_direction', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Direction)),
            exposes.enum('border', ea.STATE_SET, Object.values(tuya.ZMLookups.AM02Border)),
        ],
    };
```

After that, please make a PR to koenkk.

closed 2022-04-09 by github-actions[bot]

leroyloren, 2022-04-19:
Would it be possible to invert the state, please? This device is currently open as closed.

https://github.com/Koenkk/zigbee-herdsman-converters/issues/4040#issuecomment-1100924941