Koenkk/zigbee2mqtt #11595: LEDVANCE AC25697 Flickering irregularly

cr4yfish, 2022-02-24:
### What happened?

### Description
The light flickers irregularly. Sometimes it continuously flickers for a couple of minutes, then briefly stops and starts to flicker again.
I let 2 seperate LEDVANCE AC25697's running for about 15 minutes. Both did not stop flickering and had the exact same issues.

As far as it seems, the light does take commands as "change state", "brightness", "color_xy". OTA checks give timeouts.

I cross checked the lamp and power (voltage). Other lights work fine in the sam conditions, so I deduced the lights themselves being the problem.

### LEDVANCE Support
I've written the LEDVANCE support about this, but they don't seem to care as their answer indicates: 

> SMART+ Zigbee products can in general be used with compatible smart home systems that are based on the Zigbee standard. However, we recommend using the SMART+ Zigbee products in combination with one of our official partner systems. The SMART+ Zigbee products were not tested with MQTT and the compatibility and behavior of connected products is unknown to us.
If the lamps behave normally when these are not connected with the system and there is also no flickering visible it must be a problem of the compatibility as the SMART+ Zigbee lamps themselves  have no such hidden function.

By the way, the lights continue to flicker once removed from zigbee2mqtt. Only after resetting they stop (and start flickering once paired again).

### Finding a solution
What I've tried so far:
- Search the web for similar problems (found no solution)
- Reset light twice
- Re-Pairing
- Re-configuration
- Closing distance to Adapter (Conbee2)
- Updating zigbee2mqtt (newest release still has the issue)
- Updating Adapter (already newest version)
- Turning light off and on again
- Turning off my WIFI (Both 2.4GHz and 5GHz bands)

I already ordered IKEA Lights to replace the LEDVANCE ones and send those back to Amazon, altough I'd still like to solve this issue.

### What did you expect to happen?

The light not flickering.

### How to reproduce it (minimal and precise)

1. Turn on the light
2. Pair
3. Starts flickering
6. Reset Light
7. Stops flickering
8. Pair
9. Starts flickering

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Debug log

Here is the log:
(notice that around line 100 I started to try out different effects to see if they're the problem)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8132796/log.txt)

hellcry37, 2022-03-01:
I have this type of bulb in to my hole home, over 10, works ok. You might have a faulty device or as usual is because of the conbee stick

cr4yfish, 2022-03-02:
Thanks for the insight. The IKEA Lamps actually flicker as well.
I fixed this now by switching the zigbee channel.

closed 2022-03-02 by cr4yfish

hellcry37, 2022-03-02:
just as curiosity, what channel did you used (clearly some interference) and what channel did you switch to?

cr4yfish, 2022-03-15:
@hellcry37 I was at 11, now at 20.