Koenkk/zigbee2mqtt #11599: Xiaomi Sensors wrong battery percentage

Smandurlo, 2022-02-24:
### What happened?

I realized that all my xiaomi sensors (windows&door MCCGQ11LM and humidity&temperature WSDCGQ11LM) report 100% battery but they are not. If I pair them with my old xiaomi zigbee gateway, the % is correct according with the mV of the battery (around 40-50%).

Latest version of zigbee2mqtt running on docker and latest firmare for the cc2531 coordinator.

### What did you expect to happen?

To report the correct % of battery

### How to reproduce it (minimal and precise)

Just add the sensor and check the % of the battery

### Zigbee2MQTT version

    1.23.0 commit: 6ea61ba

### Adapter firmware version

20211115

### Adapter

cc2531

antoinelibert, 2022-02-24:
Probably related to this latest change: https://github.com/Koenkk/zigbee2mqtt/issues/8499 ?

McGiverGim, 2022-02-24:
Yes, I think the same. I understand to use a logarithmic scale and not lineal, but make 3V equal to 100% battery does not seem a good approach to me. The old 3.2V seems more correct to me.

Smandurlo, 2022-02-24:
> Probably related to this latest change: #8499 ?

indeed. I don't know why I didn't find the topic, I searched for similar topics before opening it.

closed 2022-04-03 by github-actions[bot]