Koenkk/zigbee2mqtt #11602: New version LIDL USB power strip HG06338 - Firmware 1.0.6 - 09-2021

francoisa43000, 2022-02-24:
### Link

https://www.zigbee2mqtt.io/devices/HG06338.html#lidl-hg06338

### Database entry

{"id":27,"type":"Router","ieeeAddr":"0x0c4314fffef631d1","nwkAddr":34379,"manufId":4098,"manufName":"_TZ3000_vzopcetz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_vzopcetz","stackVersion":0,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdc598c","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdc598c","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdc598c","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1645743458110,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,
I bought 3 LIDL HG06338 power strips, this one does not work correctly. I have the same model but older dating from 2020, this one works correctly. On the Lidl HOME application, the Firmware version is different 1.0.5 for the one that works and 1.0.6 for the one that does not work. Z2M recognizes the multi-socket well but when I select "State1" all the sockets light up, the multi-socket makes a "toggle" on each state. Attached is the line present in the database of the power strip that works.

Can you see if an integration is possible?

Cordially

---------------------------------------------------------------------------------------------------------------------------------
{"id":2,"type":"Router","ieeeAddr":"0xbc33acfffe452086","nwkAddr":6053,"manufId":4098,"manufName":"_TZ3000_vzopcetz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdc598c","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[3,4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdc598c","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":266,"inClusterList":[3,4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdc598c","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1645736444516,"defaultSendRequestWhen":"immediate"}

big-raf, 2022-02-26:
Hello,

I have exactly the same behaviour. I added below the debug logs when the power strip joins the zigbee network:

Zigbee2MQTT:info  2022-02-26 17:08:23: Starting interview of '0xXXXXXXXXXXXXXXXXX'
Zigbee2MQTT:info  2022-02-26 17:08:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xXXXXXXXXXXXXXXXXX","ieee_address":"0xXXXXXXXXXXXXXXXXX","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-02-26 17:08:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xXXXXXXXXXXXXXXXXX"},"type":"pairing"}'
Zigbee2MQTT:debug 2022-02-26 17:08:23: Device '0xXXXXXXXXXXXXXXXXX' announced itself
Zigbee2MQTT:info  2022-02-26 17:08:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xXXXXXXXXXXXXXXXXX","ieee_address":"0xXXXXXXXXXXXXXXXXX"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-02-26 17:08:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xXXXXXXXXXXXXXXXXX"},"type":"device_announced"}'
Zigbee2MQTT:debug 2022-02-26 17:08:24: Received Zigbee message from '0xXXXXXXXXXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_vzopcetz","modelId":"TS011F"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-26 17:08:24: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-02-26 17:08:24: Received Zigbee message from '0xXXXXXXXXXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-26 17:08:24: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-02-26 17:08:24: Received Zigbee message from '0xXXXXXXXXXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-26 17:08:24: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-02-26 17:08:24: Received Zigbee message from '0xXXXXXXXXXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"appVersion":70}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-26 17:08:24: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-02-26 17:08:24: Received Zigbee message from '0xXXXXXXXXXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-26 17:08:24: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-02-26 17:08:24: Received Zigbee message from '0xXXXXXXXXXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-26 17:08:24: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-02-26 17:08:24: Received Zigbee message from '0xXXXXXXXXXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-26 17:08:24: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-02-26 17:08:25: Received Zigbee message from '0xXXXXXXXXXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-26 17:08:25: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info  2022-02-26 17:08:26: Successfully interviewed '0xXXXXXXXXXXXXXXXXX', device has successfully been paired
Zigbee2MQTT:info  2022-02-26 17:08:26: Device '0xXXXXXXXXXXXXXXXXX' is supported, identified as: Lidl Silvercrest 3 gang switch, with 4 USB (EU, FR, CZ, BS) (HG06338)
Zigbee2MQTT:info  2022-02-26 17:08:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Silvercrest 3 gang switch, with 4 USB (EU, FR, CZ, BS)","exposes":[{"endpoint":"l1","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l1","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l2","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l2","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l3","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l3","name":"state","property":"state_l3","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"HG06338","options":[],"supports_ota":false,"vendor":"Lidl"},"friendly_name":"0xXXXXXXXXXXXXXXXXX","ieee_address":"0xXXXXXXXXXXXXXXXXX","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2022-02-26 17:08:26: Configuring '0xXXXXXXXXXXXXXXXXX'
Zigbee2MQTT:info  2022-02-26 17:08:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Silvercrest 3 gang switch, with 4 USB (EU, FR, CZ, BS)","friendly_name":"0xXXXXXXXXXXXXXXXXX","model":"HG06338","supported":true,"vendor":"Lidl"},"type":"pairing"}'
[... cut ...] easurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:error 2022-02-26 17:08:26: Failed to configure '0xXXXXXXXXXXXXXXXXX', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at Object.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:24)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/lidl.js:450:33)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:115:13))

djoris, 2022-03-05:
FYI, this seems duplicate to #5493 and #9564

francoisa43000, 2022-03-08:
solution provided in this thread : https://github.com/Koekk/zigbee2mqtt/issues/9564

closed 2022-03-08 by francoisa43000