Koenkk/zigbee2mqtt #11608: TS0601 - _TZE200_chyvmhay - only reports local temperature XX.9 or XX.0 not XX.[1-8]

gmarchart, 2022-02-25:
### What happened?

Home Assistant 2022.2.9 
Zigbee2mqtt
Current version: 1.23.0-1 
Home Assistant OS 7.4 on rp4
conbee 2 - with lates fw

Zigbee Modell
TS0601
Zigbee Hersteller
_TZE200_chyvmhay
Beschreibung
Silvercrest radiator valve with thermostat

Hello Support, 

since ever i use z2m - Valve only reports local temperature XX.9 or XX.0 - never XX.[1-8] in HA.

if i send a temperature calibration to the valve the local temperature is updated with the current temperature - like  20.2 or something other then XX.9 or XX.0 - but normal behavior is only to see XX.9 or XX.0 in Homeassistant. 

After temperatur calibration sent to the valve and the temperature grow or go down, the behavor is back to only show XX.9 and XX.0 local temperature on valve.

### What did you expect to happen?

i want to see in normal behavior that HA will show the correct temperatur from valve - like something other then XX.9 and XX.0 

### How to reproduce it (minimal and precise)

let HA with Valve run and the only report is XX.9 and XX.0 (or XX) after a while. 

### Zigbee2MQTT version

 1.23.0-1

### Adapter firmware version

 0x26660700

### Adapter

deconz/conbee 2

gmarchart, 2022-03-10:
is this problem only on my setup or is this behavior also on other systems shown as by mine ?!

Thanks for a feedback !

Ik4ru5, 2022-03-26:
No you are not alone. I thought the localtemp was broken, but i can confirm that in the last day I only observed temperatures with values XX.0 and XX.9

closed 2022-05-04 by github-actions[bot]