Koenkk/zigbee2mqtt #11610: Xiaomi/Aqara sensors report 100% battery

andrejgorin, 2022-02-25:
### What happened?

Hello! Today I migrated all my Xiaomi/Aqara sensors (11 pcs) from native Xiaomi zigbee hub to Sonoff stick + zigbee2mqtt. It looks like batteries voltage is reported incorrectly. Before migration all my batteries (around 2 years old) were at 40-50% level. This value is highly believable. However, I got value 100% in Zigbee2mqtt now. All batteries have voltage above 3000 mV in Zigbee2mqtt. It's obviously not possible. I attached log entry for reference below.
I'm wondering, is it possible to do something to get correct results from my sensors? Thank you in advance. 

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

Latest from github 

### Adapter

Sonoff zigbee 3.0 usb dongle plus 

### Debug log

Zigbee2MQTT:info  2022-02-25 14:19:59: MQTT publish: topic 'zigbee2mqtt/Cube', payload '{"action":"wakeup","action_angle":null,"action_from_side":null,"action_side":null,"action_to_side":null,"battery":100,"linkquality":127,"side":0,"voltage":3065}'

andrejgorin, 2022-02-25:
Just a few words more... Looks like Zigbee2mqtt assumes 3V as 100%, what is, obviously, wrong. Also I would like to know, does Zigbee2mqtt takes temperature of sensor in consideration? As it's necessary to correctly calculate battery percentage. Info regarding cr2032 battery:
[https://uncannier.com/cr2032-lithium-battery-charge-level-measurement/](url)

Koenkk, 2022-02-25:
Let's keep this in https://github.com/Koenkk/zigbee2mqtt/issues/8499

closed 2022-02-25 by Koenkk