Koenkk/zigbee2mqtt #11616: Aqara human body movement and illuminance sensor RTCGQ11LM not reporting battery level

dannykellett, 2022-02-25:
### What happened?

Aqara human body movement and illuminance sensor doesn't show battery when using z2mqtt.

mqqt explorere shows:
{
  "battery": null,
  "illuminance": 20,
  "illuminance_lux": 20,
  "linkquality": 145,
  "occupancy": false,
  "temperature": null,
  "voltage": null
}

Home assistant dev tools show:
battery: null
illuminance: 20
illuminance_lux: 20
linkquality: 145
occupancy: false
temperature: null
voltage: null
device_class: motion
friendly_name: motion_sensor_office_occupancy

I disabled and enabled zha and it showed, assuming this means the unit isn't faulty?

Thanks

### What did you expect to happen?

To see the battery % as documented here: https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html

### How to reproduce it (minimal and precise)

Installed versions and this is the only zigbee device.
View values in mqtt explorer and homeassistant

### Zigbee2MQTT version

1.23.0 commit: unknown

### Adapter firmware version

20211217

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

2022-02-25T18:42:26.946Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,4,0,0,0,0,2,38]
2022-02-25T18:42:26.958Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,4,0,96,254,16,69,132,0,0,0,0,0,10,2,94,192,32,8,0,1,0,0,0,110,254,23,69,255,0,0,0,4,128,0,58,0,0,0,0,0,10,2,94,192,32,8,0,1,0,0,0,172]
2022-02-25T18:42:26.958Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,4,0,96,254,16,69,132,0,0,0,0,0,10,2,94,192,32,8,0,1,0,0,0,110,254,23,69,255,0,0,0,4,128,0,58,0,0,0,0,0,10,2,94,192,32,8,0,1,0,0,0,172]
2022-02-25T18:42:26.958Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 4 - [0] - 96
2022-02-25T18:42:26.959Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - simpleDescReq - {"status":0}
2022-02-25T18:42:26.959Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,16,69,132,0,0,0,0,0,10,2,94,192,32,8,0,1,0,0,0,110,254,23,69,255,0,0,0,4,128,0,58,0,0,0,0,0,10,2,94,192,32,8,0,1,0,0,0,172]
2022-02-25T18:42:26.959Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 16 - 2 - 5 - 132 - [0,0,0,0,0,10,2,94,192,32,8,0,1,0,0,0] - 110
2022-02-25T18:42:26.959Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":0,"status":0,"nwkaddr":0,"len":10,"endpoint":2,"profileid":49246,"deviceid":2080,"deviceversion":0,"numinclusters":1,"inclusterlist":[0],"numoutclusters":0,"outclusterlist":[]}
2022-02-25T18:42:26.960Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,23,69,255,0,0,0,4,128,0,58,0,0,0,0,0,10,2,94,192,32,8,0,1,0,0,0,172]
2022-02-25T18:42:26.960Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 23 - 2 - 5 - 255 - [0,0,0,4,128,0,58,0,0,0,0,0,10,2,94,192,32,8,0,1,0,0,0] - 172
2022-02-25T18:42:26.960Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - msgCbIncoming - {"srcaddr":0,"wasbroadcast":0,"clusterid":32772,"securityuse":0,"seqnum":58,"macdstaddr":0,"msgdata":{"type":"Buffer","data":[]}}
2022-02-25T18:42:26.960Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-02-25T18:42:26.961Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":0,"nwkaddrofinterest":0,"endpoint":1}
2022-02-25T18:42:26.961Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,4,0,0,0,0,1,37]
2022-02-25T18:42:26.971Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,4,0,96,254,22,69,132,0,0,0,0,0,16,1,4,1,0,4,0,2,0,0,25,0,2,0,5,2,5,220,254,29,69,255,0,0,0,4,128,0,59,0,0,0,0,0,16,1,4,1,0,4,0,2,0,0,25,0,2,0,5]
2022-02-25T18:42:26.972Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,4,0,96,254,22,69,132,0,0,0,0,0,16,1,4,1,0,4,0,2,0,0,25,0,2,0,5,2,5,220,254,29,69,255,0,0,0,4,128,0,59,0,0,0,0,0,16,1,4,1,0,4,0,2,0,0,25,0,2,0,5]
2022-02-25T18:42:26.972Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 4 - [0] - 96
[... 57 lines cut ...]
Zigbee2MQTT:debug 2022-02-25 18:42:27: Received MQTT message on 'zigbee2mqtt/bridge/config' with data '{"commit":"unknown","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20211217,"transportrev":2},"type":"zStack3x0"},"log_level":"warn","network":{"channel":11,"extendedPanID":"0x00124b0024cae636","panID":6754},"permit_join":false,"version":"1.23.0"}'
Zigbee2MQTT:info  2022-02-25 18:42:27: Started frontend on port 0.0.0.0:8099
Zigbee2MQTT:info  2022-02-25 18:42:27: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"unknown","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20211217,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b0024cae636","panID":6754},"permit_join":false,"version":"1.23.0"}'
Zigbee2MQTT:info  2022-02-25 18:42:27: MQTT publish: topic 'homeassistant/sensor/0x00158d0007755b21/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007755b21"],"manufacturer":"Xiaomi","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","name":"motion_sensor_office"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/motion_sensor_office","name":"motion_sensor_office_battery","state_class":"measurement","state_topic":"zigbee2mqtt/motion_sensor_office","unique_id":"0x00158d0007755b21_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2022-02-25 18:42:27: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0007755b21/occupancy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007755b21"],"manufacturer":"Xiaomi","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","name":"motion_sensor_office"},"device_class":"motion","json_attributes_topic":"zigbee2mqtt/motion_sensor_office","name":"motion_sensor_office_occupancy","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/motion_sensor_office","unique_id":"0x00158d0007755b21_occupancy_zigbee2mqtt","value_template":"{{ value_json.occupancy }}"}'
Zigbee2MQTT:info  2022-02-25 18:42:27: MQTT publish: topic 'homeassistant/sensor/0x00158d0007755b21/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007755b21"],"manufacturer":"Xiaomi","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","name":"motion_sensor_office"},"device_class":"temperature","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/motion_sensor_office","name":"motion_sensor_office_temperature","state_class":"measurement","state_topic":"zigbee2mqtt/motion_sensor_office","unique_id":"0x00158d0007755b21_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Zigbee2MQTT:info  2022-02-25 18:42:27: MQTT publish: topic 'homeassistant/sensor/0x00158d0007755b21/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007755b21"],"manufacturer":"Xiaomi","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","name":"motion_sensor_office"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/motion_sensor_office","name":"motion_sensor_office_voltage","state_class":"measurement","state_topic":"zigbee2mqtt/motion_sensor_office","unique_id":"0x00158d0007755b21_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
Zigbee2MQTT:info  2022-02-25 18:42:27: MQTT publish: topic 'homeassistant/sensor/0x00158d0007755b21/illuminance_lux/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007755b21"],"manufacturer":"Xiaomi","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","name":"motion_sensor_office"},"device_class":"illuminance","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/motion_sensor_office","name":"motion_sensor_office_illuminance_lux","state_class":"measurement","state_topic":"zigbee2mqtt/motion_sensor_office","unique_id":"0x00158d0007755b21_illuminance_lux_zigbee2mqtt","unit_of_measurement":"lx","value_template":"{{ value_json.illuminance }}"}'
[... cut ...] Xiaomi","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","name":"motion_sensor_office"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/motion_sensor_office","name":"motion_sensor_office_linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/motion_sensor_office","unique_id":"0x00158d0007755b21_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'

McGiverGim, 2022-02-25:
Can you give some hours and check again? In the latest version of z2m it takes some hours to appear in xiaomi devices. I don't know why.

dannykellett, 2022-02-27:
Hi @McGiverGim thank you. You were right. I just checked and it's appeared now. I don't know who manages this page [RTCGQ11LM](https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html) but if this is a known issue, I think adding a comment may save your time with posts like mine. 
Thanks again

McGiverGim, 2022-02-27:
This has happened only with latest version, of z2m and with several different Xiaomi devices. This pages can be edited by anybody who wants collaborate. I think there is a link at each page to edit it.

closed 2022-04-07 by github-actions[bot]