Koenkk/zigbee2mqtt #11621: OSRAM 4058075816459 (4x Switch) - sending right top/button hold twice

UweLoyal, 2022-02-25:
### What happened?

I have the effect that "hold" action for "right_top" and "right_bottom" is sending twice with a view seconds distance

Rgds
UweLoyal

### Zigbee2MQTT version

2-1.2.7.1.

### Adapter firmware version

20210708

### Adapter

Zigbee Coordinator in ioBroker

Koenkk, 2022-02-26:
Can you provide the debug log when these events are send twice?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

UweLoyal, 2022-02-26:
I activate the Zigbee-herdsman debug info in ioBroker, but where I can find the Logfile?

Could you tell me the location?

Rgds

UweLoyal

UweLoyal, 2022-02-26:
I’ll now used the ioBrober Log, hope this will help you also.

 

See in Log:
0_userdata.0.Lampensteuerung.EG.Flur.Licht_Taster.right_top_hold

0_userdata.0.Lampensteuerung.EG.Flur.Licht_Taster.right_bottom_hold

Rgds
UweLoyal

 

2022-02-26 15:33:35.758 - info: zigbee.0 (2003) Loglevel changed from "info" to "debug"
2022-02-26 15:33:43.817 - debug: zigbee.0 (2003) handleMessage. {"type":"commandStatusChangeNotification","device":{"_events":{},"_eventsCount":0,"_maxListeners":100,"ID":9,"_type":"EndDevice","_ieeeAddr":"0xec1bbdfffea3e49b","_networkAddress":54222,"_manufacturerID":4619,"_endpoints":[{"_events":{},"_eventsCount":0,"_maxListeners":100,"ID":1,"profileID":260,"deviceID":1026,"inputClusters":[0,1,3,1280,2821],"outputClusters":[25],"deviceNetworkAddress":54222,"deviceIeeeAddress":"0xec1bbdfffea3e49b","clusters":{"genBasic":{"attributes":{"modelId":"TY0202","manufacturerName":"_TZ1800_fcdjzz3s","powerSource":3,"zclVersion":1,"appVersion":65,"stackVersion":2,"hwVersion":1,"dateCode":"2019.12.12"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002393402d","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"_binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002393402d","endpointID":1}],"_configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{},"pendingRequests":[]}],"_manufacturerName":"_TZ1800_fcdjzz3s","_powerSource":"Battery","_modelID":"TY0202","_applicationVersion":65,"_stackVersion":2,"_zclVersion":1,"_hardwareVersion":1,"_dateCode":"2019.12.12","_interviewCompleted":true,"_interviewing":false,"_skipDefaultResponse":false,"_skipTimeResponse":false,"meta":{"configured":1},"_lastSeen":1645886023817,"_defaultSendRequestWhen":"immediate","_linkquality":66},"endpoint":"[Circular]","data":{"zonestatus":33,"extendedstatus":0},"linkquality":66,"groupID":0,"cluster":"ssIasZone","meta":{"zclTransactionSequenceNumber":22,"manufacturerCode":null,"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0}}}
2022-02-26 15:33:43.820 - debug: zigbee.0 (2003) Received Zigbee message from '0xec1bbdfffea3e49b', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":33,"extendedstatus":0}' from endpoint 1 with groupID 0
2022-02-26 15:33:43.822 - debug: zigbee.0 (2003) Type commandStatusChangeNotification device {"type":"device","device":{"_events":{},"_eventsCount":0,"_maxListeners":100,"ID":9,"_type":"EndDevice","_ieeeAddr":"0xec1bbdfffea3e49b","_networkAddress":54222,"_manufacturerID":4619,"_endpoints":[{"_events":{},"_eventsCount":0,"_maxListeners":100,"ID":1,"profileID":260,"deviceID":1026,"inputClusters":[0,1,3,1280,2821],"outputClusters":[25],"deviceNetworkAddress":54222,"deviceIeeeAddress":"0xec1bbdfffea3e49b","clusters":{"genBasic":{"attributes":{"modelId":"TY0202","manufacturerName":"_TZ1800_fcdjzz3s","powerSource":3,"zclVersion":1,"appVersion":65,"stackVersion":2,"hwVersion":1,"dateCode":"2019.12.12"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002393402d","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"_binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002393402d","endpointID":1}],"_configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{},"pendingRequests":[]}],"_manufacturerName":"_TZ1800_fcdjzz3s","_powerSource":"Battery","_modelID":"TY0202","_applicationVersion":65,"_stackVersion":2,"_zclVersion":1,"_hardwareVersion":1,"_dateCode":"2019.12.12","_interviewCompleted":true,"_interviewing":false,"_skipDefaultResponse":false,"_skipTimeResponse":false,"meta":{"configured":1},"_lastSeen":1645886023817,"_defaultSendRequestWhen":"immediate","_linkquality":66},"mapped":{"fingerprint":[{"modelID":"TY0202","manufacturerName":"_TZ1800_fcdjzz3s"}],"model":"HG06335/HG07310","vendor":"Lidl","description":"Silvercrest smart motion sensor","fromZigbee":[{"cluster":"ssIasZone","type":"commandStatusChangeNotification"},{"cluster":"genPowerCfg","type":["attributeReport","readResponse"]}],"toZigbee":[{"key":["scene_store"]},{"key":["scene_recall"]},{"key":["scene_add"]},{"key":["scene_remove"]},{"key":["scene_remove_all"]},{"key":["read"]},{"key":["write"]},{"key":["command"]}],"exposes":[{"type":"binary","name":"occupancy","property":"occupancy","access":1,"value_on":true,"value_off":false,"description":"Indicates whether the device detected occupancy"},{"type":"binary","name":"battery_low","property":"battery_low","access":1,"value_on":true,"value_off":false,"description":"Indicates if the battery of this device is almost empty"},{"type":"binary","name":"tamper","property":"tamper","access":1,"value_on":true,"value_off":false,"description":"Indicates whether the device is tampered"},{"type":"numeric","name":"battery","property":"battery","access":1,"unit":"%","description":"Remaining battery in %","value_min":0,"value_max":100},{"type":"numeric","name":"linkquality","property":"linkquality","access":1,"unit":"lqi","description":"Link quality (signal strength)","value_min":0,"value_max":255}],"options":[]},"name":"0xec1bbdfffea3e49b"} incoming event: {"type":"commandStatusChangeNotification","device":{"_eve [... cut ...] ]}
2022-02-26 15:33:44.045 - debug: zigbee.0 (2003) Change state 'colortemp' at device 0x842e14fffe9c1638 type 'HG06106A'
2022-02-26 15:33:44.049 - debug: zigbee.0 (2003) entity: {"type":"device","device":{"_events":{},"_eventsCount":0,"_maxListeners":100,"ID":5,"_type":"Router","_ieeeAddr":"0x842e14fffe9c1638","_networkAddress":1826,"_manufacturerID":4098,"_endpoints":[{"_events":{},"_eventsCount":0,"_maxListeners":100,"ID":1,"profileID":260,"deviceID":269,"inputClusters":[0,3,4,5,6,8,768,4096]

UweLoyal, 2022-02-26:
I’ll now used the ioBrober Log, hope this will help you also.

See in Log:
0_userdata.0.Lampensteuerung.EG.Flur.Licht_Taster.right_top_hold
0_userdata.0.Lampensteuerung.EG.Flur.Licht_Taster.right_bottom_hold

Rgds
UweLoyal
[Debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/8146871/Debug.txt)

UweLoyal, 2022-02-26:
> You are using the legacy integration, this should work fine with legacy disabled (https://www.zigbee2mqtt.io/devices/4058075816459.html#options) (I don't know how to set this option in iobroker)

If left works fine, why right not?
For that that’s not logical with legacy.

I also find nothing with legacy in ioBroker, using standard installation.

Rgds
UweLoyal

Jogobo, 2022-03-04:
With legacy disabled two buttons (don't remember which) do not work and a different kind of messages is sent. So this isn't a solution for the double sent messages.

closed 2022-04-11 by github-actions[bot]