Koenkk/zigbee2mqtt #11625: [New device support]: Hue white and ambiance runner spot 3 (black)

boust3r, 2022-02-26:
### Link

https://www.philips-hue.com/de-de/p/hue-white-ambiance-runner-dreier-spot/8719514338401

### Database entry

      modelZigbee:929003046301_03     type:Router     nwk:27825     manuf id:4107     manufacturer:Philips     power:Mains (single phase)     app version:2     hard version:3     zcl version:2     stack version:1     date code:20200327     build:1.65.11_hB798F2B     interviewed:true     configured:false      endpoint:11     profile:260     input clusters:genBasic (0)     genIdentify (3)     genGroups (4)     genScenes (5)     genOnOff (6)     genLevelCtrl (8)     touchlink (4096)     manuSpecificSamsungAccelerometer (64514)     lightingColorCtrl (768)     output clusters:genOta (25)      endpoint:242     profile:41440     output clusters:greenPower (33)

### Comments

I am new to this kind of topic. But I think this must be easy to implement, since the Runner spot 3 (white) is already implemented. Isn't it possible to add the Zigbee model number (929003046301_03) to the entry of the philips.js file in the same element as the white version and it should work right away?

boust3r, 2022-02-26:
This is the code from the philips.js

  {
        zigbeeModel: ['5309331P6', '5309330P6'],
        model: '5309331P6',
        vendor: 'Philips',
        description: 'Hue White ambiance Runner triple spotlight',
        meta: {turnsOffAtBrightness1: true},
        extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}),
        ota: ota.zigbeeOTA,
    },

Koenkk, 2022-02-27:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-27 by Koenkk

boust3r, 2022-02-27:
Thank you for the fast implementation. I hope to also be able to implement if fast into the iobroker zigbee.adapter :)