Koenkk/zigbee2mqtt #11627: [Feature request]: Webhook for device availability

newhinton, 2022-02-26:
### Is your feature request related to a problem? Please describe

Sometimes devices are going offline. This is more or less expected, and the frontend already does a great job displaying the state!

However, usually i dont look too often into the webui, so it may take a while for me to notice a device has gone offline. A notification would be great!

### Describe the solution you'd like

A webhook call that can be defined in the advanced config which allows to make a request to an external service, eg. gotify or home assistant that sends a json with some information (deviceid/name, last seen etc.)

### Describe alternatives you've considered

An Dummy-Device that acts as a status entity which contains that information published to homeassistant or other services

### Additional context

No further context

Koenkk, 2022-02-27:
You can use the MQTT availability topic and send a notification based on that (`zigbee2mqtt/[FRIENDLY_NAME]/availability`)

newhinton, 2022-03-30:
Closed because above comment

closed 2022-03-30 by newhinton