Koenkk/zigbee2mqtt #11632: Can not change click mode on Aqara H1 Wireless Switch (No network route, 205)

Auxx, 2022-02-27:
### What happened?

I have Zigbee2MQTT running on Raspberry Pi 4 with Sonoff Zigbee 3.0 Dongle (CC2652P) and three Aqara devices connected directly to the coordinator, no routers. One of the devices is H1 switch which has two click modes: fast and multi. When I try to change click mode through Zigbee2MQTT UI I get the following error:

```
Publish 'set' 'click_mode' to 'Study switch' failed: 'Error: Write 0x54ef4410002a1367/1 aqaraOpple({"293":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
```

Everything else works just fine - I'm getting click events from the switch and all the data from other devices (Aqara sensors). The switch was the first device I've paired and it exhibited such behaviour since the start.

Sonoff Zigbee 3.0 Dongle is flashed with the most recent firmware (CC1352P2_CC2652P_launchpad_coordinator_20211217.zip). Zigbee2MQTT is running inside docker as per instructions.

### What did you expect to happen?

I expect no errors and click mode to change.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 Dongle (CC2652P)

dannykellett, 2022-02-28:
+1 Exactly the same setup and problem

PaulMcMillan, 2022-03-16:
I have a similar problem, although mine involves a timeout rather than "No network route".

```
Publish 'set' 'click_mode' to '2ButtonH1' failed: 'Error: Write 0x54ef4410000684ef/1 aqaraOpple({"293":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 35568 - 1 - 14 - 64704 - 4 after 10000ms)'
```

I also am using the Sonoff 3.0 dongle and `CC1352P2_CC2652P_launchpad_coordinator_20211217.zip` firmware. Everything else seems functional.

jdeut, 2022-04-09:
I want to chime in with a similar error
```
Publish 'set' 'click_mode' to 'küche_wandschalter' failed: 'Error: Write 0x54ef44100012324c/1 aqaraOpple({"293":{"value":2,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
```

Zigbee2MQTT Version:  [1.25.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.25.0) commit: [6f1460e4](https://github.com/Koenkk/zigbee2mqtt/commit/6f1460e4)
Coordinator Typ: ConBee2/RaspBee2
Coordinator Version: 0x26720700

PaulMcMillan, 2022-04-10:
I can confirm that this is still an issue in the April 2022 release of firmware and Z2M.

sanasol, 2022-04-11:
@Auxx @jdeut @PaulMcMillan @dannykellett 

Click any switch button 5 times(switch leds will blink) and then fastly change mode.
Try few times if not work and there is some delay before applying also.

Auxx, 2022-04-11:
@sanasol tried pressing 5 times multiple ways (change mode, press 5 times, then press 5 times and change mode) - nothing works.

PaulMcMillan, 2022-04-15:
@sanasol huh! That totally did the trick, and swapped the mode. I just kinda clicked all the things a bunch of times very fast at the same time and it worked.

Auxx, 2022-04-15:
I tried it a few more times and noticed a new error:

```
Publish 'set' 'click_mode' to 'Study switch' failed: 'Error: Write 0x54ef4410002a1367/1 aqaraOpple({"293":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
```

sanasol, 2022-04-15:
@Auxx click switch first and try change settings, works for me, just got 6 new switches - each time works)
If you get no result wait until all errors gone(10s timeout errors) and try again.

@PaulMcMillan yeah lightning speed indeed, using it after Tuya Wireless Switches(TS0041-TS0043), that going to trash now.
Also clicking is very smooth and pleasurable.

PaulMcMillan, 2022-04-19:
My one compliant (and I'll admit it's minor) is that I really wish they offered a way to allow both fast response and press-and hold.

One option would be to send the press event on key-up rather than key-down. Key-down is very slightly faster, but press-on-key-up would allow press-and-hold without any additional artificial latency.

Another option would be to send separate key-down and key-up messages so my automation could figure out what I wanted on the backend, maybe with a separate "hold" event after a very short delay.

My approach has been to design my automation such that the first fast thing I do is a subset of the next scene if multiple inputs come after. So the first click turns on my headroom instantly with no delay, if I detect a second click shortly after the first click, it sends "turn on the area around the bedroom" and if I click a third time it sends "turn on the whole house". This combines the best of both instant response and multi-click behaviors, but I'm sad I can't also add press-and-hold into the mix with this device.

closed 2022-06-29 by github-actions[bot]