Koenkk/zigbee2mqtt #11641: [New device support]: Woox R7067

benas1989, 2022-02-27:
### Link

https://wooxhome.com/products-c10/security-c6/woox-r7067-single-radiator-valve-unit-p69

### Database entry

{"id":7,"type":"EndDevice","ieeeAddr":"0x5c0272fffef142ad","nwkAddr":54126,"manufId":4098,"manufName":"_TZE200_wnvhlcgl","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0007_ )\u0012\u0007_ )\u0012","65506":27,"modelId":"TS0601","manufacturerName":"_TZE200_wnvhlcgl","appVersion":64,"stackVersion":0,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x60a423fffe20423b","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1645978327060,"defaultSendRequestWhen":"immediate"}

### Comments

Holiday schedule - only raw data. Cannot set.
Weekly schedule - only raw data. Cannot set.
The temperature does not update when the mode is changed. 
Cannot activate presets eco, comfort. I dont know how to read data from manuSpecificTuya cluster.

Everything in the attached files below: 
[R7067.zip](https://github.com/Koenkk/zigbee2mqtt/files/8149019/R7067.zip)

benas1989, 2022-03-30:
:D

closed 2022-05-09 by github-actions[bot]

viprotechlv, 2022-08-10:
Hey! How I can use your solution?

jeanaaaa, 2022-09-11:
Thanks @benas1989 for your sharing.
To be able to use it I updated the JS line 62 and 63 as follows
`         .withPreset(['Comfort', 'Eco']),`
`        e.away_mode(),`
Did someone was able to handle 
- the "off" state for the HVAC mode 
- the "Holiday" mode

BenisDracke, 2023-01-11:
How to install / use the files?

elninosi, 2023-12-09:
@Koenkk  can we include this device in next release?

Koenkk, 2023-12-10:
Could someone make a pull request for this?

MrRasmus, 2024-03-02:
Nobody able to add this to home assistant?

elninosi, 2024-03-02:
No :(

Been busy lately, do you want to give it a go?

MrRasmus, 2024-03-02:
I believe it can be fixed with using the Tuya model: 'TS0601_thermostat'
`   model: 'TS0601_thermostat',
        vendor: 'TuYa',
        description: 'Radiator valve with thermostat',
        whiteLabel: [
            {vendor: 'Moes', model: 'HY368'},
            {vendor: 'Moes', model: 'HY369RT'},
            {vendor: 'SHOJZJ', model: '378RT'},
            {vendor: 'Silvercrest', model: 'TVR01'},
            {vendor: 'Immax', model: '07732B'},
            tuya.whitelabel('Immax', '07732L', 'Radiator valve with thermostat', ['_TZE200_rufdtfyv']),
            {vendor: 'Evolveo', model: 'Heat M30'},`
            
Zigbee Model:
TS0601
Zigbee Producent:
_TZE200_wnvhlcgl

https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html

But somebody smarther at coding that me, would have to do it. I attempted to add it with the support a new device without luck. It can't find the external converter, and I don't know how to add it to my configuration as I dont have anything regarding my mqtt in my configuration.

MrRasmus, 2024-03-09:
These files do not work proberly for me - I keep getting an error with: 
One error is this Error on my zigbee2mqtt Addon in home assistant:
Exception while calling fromZigbee converter: tuya.getDataValue is not a function}
The other error is I don't see the temperature the devices was supposed to send.
or any other setting. But im able to set the heating setpoint after a couple of attempts. This is the only thing I tried, but the termostat does not keep the temp.

jeanaaaa, 2025-05-19:
After some research I make it working by copying some codes from the z2m libraries.

You'll find hereafter the zip containing the files to be uploaded in the same folders for a homeassistant install.

If someone knows how to integrate it in the normal z2m repository it will be better.

[woox_R7067.zip](https://github.com/user-attachments/files/20301123/woox_R7067.zip)

elninosi, 2025-05-19:
> After some research I make it working by copying some codes from the z2m libraries.
>
> You'll find hereafter the zip containing the files to be uploaded in the same folders for a homeassistant [...]

New device support ticket opened: https://github.com/Koenkk/zigbee2mqtt/issues/27450