Koenkk/zigbee2mqtt #11647: Conbee II response failed - no response received

Jtef, 2022-02-28:
### What happened?

Tested 4 Tuya module switches on a HA Blue (core-2022.2.6, supervisor-2022.01.1). Each module has 2 gangs and they are configured to work 2 way:
Front 1 1 <-> Front 2 1
Front 1 2 <-> Front 2 2
Back 1 1 <-> Back 2 1
Back 1 2 <-> Back 2 2
In HA when I turn on Front 1 1, Front 1 2, Back 1 1, and Back 1 2 the corresponding switch should turn on but in 80% of cases I get errors with "no response received" from one or more switches.
I got this error on the main HA device and I moved to a HA installed in a virtual machine with only the switches paired to zigbee network. I used an extension USB cable and I was in the same room with the switches (2 meters between Conbee 2 and the switches).
I also noticed the same error, from time to time, to the relays (Zonoff zbmini) that I use for heating, and also for Ikea Tradfri sockets.
I easily reproduce the error on multiple actions on/off. Also tested Conbee II with deconz and the error couldn't be reproduced but I noticed a delay instead of an error (max 1 sec delay for some switches but no error).

Additional info:
Tested also with CC2531 as coordinator and the problem doesn't reproduce.
Tested with Sonoff 3.0 USB dongle plus and the same problem.

### What did you expect to happen?

I expect the switches/relays to turn on/off instead receiving an error or at least a small delay instead of an error.

### How to reproduce it (minimal and precise)

Pair 4 relays (Zonoff zbmini) or tradfri sockets using Conbee 2 and turn on/off all of them from HA. At some point you'll get this error:
`Publish 'set' 'state' to 'Back 1 Bedroom 1' failed: 'Error: Command 0xa4c1386d10db1302/2 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'`

I assume it can be reproduced with other devices too, it doesn't seem brand related.

### Zigbee2MQTT version

1.23.0-1

### Adapter firmware version

deCONZ_ConBeeII_0x26720700.bin.GCF    19-Aug-2021 22:20

### Adapter

Conbee II

### Debug log

`Debug: Received MQTT message on 'zigbee2mqtt/Front 1 Bedroom 1/l1/set' with data 'ON'
Debug: Publishing 'set' 'state' to 'Front 1 Bedroom 1'
Debug: Received MQTT message on 'zigbee2mqtt/Front 1 Bedroom 1/l2/set' with data 'ON'
Debug: Publishing 'set' 'state' to 'Front 1 Bedroom 1'
Debug: Received MQTT message on 'zigbee2mqtt/Back 1 Bedroom 1/l1/set' with data 'ON'
Debug: Received MQTT message on 'zigbee2mqtt/Back 1 Bedroom 1/l2/set' with data 'ON'
Debug: Publishing 'set' 'state' to 'Back 1 Bedroom 1'
Debug: Publishing 'set' 'state' to 'Back 1 Bedroom 1'
Info: MQTT publish: topic 'zigbee2mqtt/Front 1 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":null}'
Debug: Received Zigbee message from 'Front 1 Bedroom 1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info: MQTT publish: topic 'zigbee2mqtt/Front 1 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":null}'
Debug: Received MQTT message on 'zigbee2mqtt/Front 2 Bedroom 1/l1/set' with data 'ON'
Debug: Publishing 'set' 'state' to 'Front 2 Bedroom 1'
Debug: Received Zigbee message from 'Front 1 Bedroom 1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID null
Info: MQTT publish: topic 'zigbee2mqtt/Front 1 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":"ON"}'
Info: MQTT publish: topic 'zigbee2mqtt/Front 1 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":"ON"}'
Info: MQTT publish: topic 'zigbee2mqtt/Back 1 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":"OFF"}'
Info: MQTT publish: topic 'zigbee2mqtt/Front 2 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":"OFF"}'
Debug: Received Zigbee message from 'Front 2 Bedroom 1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info: MQTT publish: topic 'zigbee2mqtt/Front 2 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":"OFF"}'
Debug: Received Zigbee message from 'Back 1 Bedroom 1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info: MQTT publish: topic 'zigbee2mqtt/Back 1 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":"OFF"}'
Debug: Received MQTT message on 'zigbee2mqtt/Front 2 Bedroom 1/l2/set' with data 'ON'
Debug: Publishing 'set' 'state' to 'Front 2 Bedroom 1'
Debug: Received MQTT message on 'zigbee2mqtt/Back 2 Bedroom 1/l1/set' with data 'ON'
Debug: Publishing 'set' 'state' to 'Back 2 Bedroom 1'
Info: MQTT publish: topic 'zigbee2mqtt/Front 2 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":"ON"}'
Debug: Received Zigbee message from 'Front 2 Bedroom 1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID null
Info: MQTT publish: topic 'zigbee2mqtt/Front 2 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":"ON"}'
Info: MQTT publish: topic 'zigbee2mqtt/Back 2 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":"OFF"}'
Debug: Received Zigbee message from 'Back 2 Bedroom 1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info: MQTT publish: topic 'zigbee2mqtt/Back 2 Bedroom 1', payload '{"linkquality":255,"state_l1":"ON","state_l2":"OFF"}'
**Error: Publish 'set' 'state' to 'Back 1 Bedroom 1' failed: 'Error: Command 0xa4c1386d10db1302/2 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Debug Error: Command 0xa4c1386d10db1302/2 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received) at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:608:23) at Generator.throw (<anonymous>) at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:25:65)**
Info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Back 1 Bedroom 1' failed: 'Error: Command 0xa4c1386d10db1302/2 genOnOff.on({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"Back 1 Bedroom 1"},"type":"zigbee_publish_error"}'
Debug: Saving state to file /config/zigbee2mqtt/state.json`

bednim, 2022-03-09:
have got same issue with ZBMINI relay - on latest z2m and conbee II with latest FW.

Don't even know where to start debugging?

Jtef, 2022-03-09:
I've performed more tests using 4 relays with different versions of z2m and firmwares for Conbee II but no luck, same issue. Also tried different channels 11, 15, 20 (even I don't have wifi AP around me).
I have a short video on how to reproduce the problem. If needed I'll post it.

bednim, 2022-03-09:
I can replicate the issue very easily - just quickly toggle power switch, error quickly appears. Changed WiFi channels, don't really want to mess with zigbee channel as I've got 60 zigbee devices...

SeanHama, 2022-03-25:
Im following this issue, having the same problem with devices falling off LQI = nothing. Struggling to get some devices working again, weird things is there is zigbee devices in the same gang switch working fine. Using latest z2m and Conbee 2 and same firmware as OG.

Publish 'set' 'state' to 'Balcony Small' failed: 'Error: Command 0x086bd7fffe4b0293/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

levran, 2022-04-06:
I join, there are also a lot of problems with the "no response received", the wifi and zigbee channels are very spaced apart. 
And the feeling that it got worse after version 1.23 and aqara relay firmware update

KennethSeiero, 2022-05-02:
Same here, following. Latest versions, 40+ devices, started recently.

KPWhiver, 2022-05-03:
Same here, fixed it by adding a delay between sending the individual MQTT messages to zigbee2mqtt. Looking through the deconz code it seems that deconz simply tries again if a device does not respond to a state change in time: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/9f26cd6b3cd07a375527e0bafabae3b07342cc7e/state_change.cpp#L50. Perhaps to solve for this very issue. Apparently zigbee2mqtt does not do that.

bednim, 2022-05-03:
So in theory changing adapter would help? As in moving from conbee to
cc2652rb for example? I'm asking as I've got new adapter sitting in the
drawer waiting to be plugged in.

Thanks!

SeanHama, 2022-05-03:
Can you move adaptors without having to repair every device again?

KPWhiver, 2022-05-03:
> 

If it is indeed the case that the conbee II has some stability issues that need the mentioned software solution to fix then I suspect switching to another adapter would help. However, reading [this](https://www.zigbee2mqtt.io/guide/faq/#what-does-and-does-not-require-repairing-of-all-devices) I expect you will have to repair every device. The true solution would probably be for zigbee2mqtt to also try again when a device refuses to change state (in combination with a conbee II).

bednim, 2022-05-03:
that delay didn't work for me at all unfortunately, hence the change of the
coordinator.

its worth to remember that conbee was in experimental stage not so long ago.

cameronjust, 2022-05-16:
My entire Conbee II network has started playing up with these errors too :(

Is there a solution/fix for this yet?

I thought the Conbee II was one of the better USB devices. Does anyone have a recommendation of a more stable coordinator that works well with Zigbee2mqtt?

bednim, 2022-05-17:
I've swapped for cc2652rb (slae.sh) and these errors are finally gone.

RazoredRaven, 2022-05-18:
Any updates for a fix on this for those that don't have a spare adaptor lying around?

SeanHama, 2022-05-18:
Did you have to repair all your devices again?

bednim, 2022-05-18:
Yes, I had to, it was bit more difficult than conbee as all my external
nodes had to be brought inside in order to pair. Once done - move them back
outside.

cameronjust, 2022-05-18:
> Yes, I had to, it was bit more difficult than conbee as all my external nodes had to be brought inside in order to pair. Once done - move them back outside.

Am I right in assuming it was just as simple as holding in the re-pair button for each device while near the controller?

Zigbee2MQTT still has all the IEEE Addresses in the config file so it just remaps to the same friendlyname once you re-pair. Moderately painful but at least you don't have to rebuild all automations which are based on friendlyname.

Only thing I can think that would maybe not work is the act of re-pairing might reset some custom setting you might have applied to the device. Just tested with a "Aqara vibration sensor" with a custom sensitivity applied (via a MQTT set command) and it didn't seem to get reset when I re-paired.

bednim, 2022-05-18:
That's right, all friendly names are reapplied, it's not 'hard reset' -
just pairing and popping back in z2m.

As to custom settings - no idea.

In case anyone wants to go this way - remove conbee from configuration.yaml
and change panid one number up.

anderswernersson, 2022-05-27:
Same issue here. adding some 350ms "adapter delay" seems to make the situation better though.

vandalon, 2022-06-21:
Even with 350ms delay I see this error now and then in the logs (using conbee II, latest firmware).

bednim, 2022-07-04:
Guys, so the getonoff errors are back, my whole zigbee network became a mess all of sudden. That's different stick (cc2652rb) but issues are terrible - aqara devices dropping out, Ikea dropping out, sonoff dropping out.

Getonoff flooding my log.

MattDahEpic, 2022-08-09:
I am also experiencing this issue. ConBee II, zm 1.27.0, HA 2022.8.2

adgenet, 2022-08-10:
I have also been experiencing this issue. No devices ever drop from my network though, it just seems like they occasionally do not respond unless you send them a command multiple times.
I set adapter delay to 350ms as suggested earlier, the problem becomes very infrequent.
My adapter is a ConbeeII running 0x26780700.

MunkeyBalls, 2022-08-28:
Just switched from deconz (phoscon) to zigbee2mqtt and was experiencing the same issue. Downgraded to version 1.25.2 and everything is working fine.

scholdan, 2022-08-29:
Just installed 32 ikea gu10 bulbs, and got huge issues with unresponsive, they are pretty close to the stick and signal strength is 255 on most of them.
Tried setting adapter delay to 350ms and now all works but are damm slow and not in sync (light a flashing christmas light) so that also sucks. should i just go a buy another zigbee adapter?

bednim, 2022-08-29:
if these bulbs are in same room put them all in z2m group - this will make
them respond better.

I had conbee II and now got rid of it in favor of cc2652RB (slae.sh)

zansko, 2022-08-29:
I'm experiencing the same issue, 350 delay solves the problem but introduces latency.

Conbee II running 0x26780700. My network is quite small, ~8 hue lights, hue movement sensors, some Aqara switches and sensors.

MunkeyBalls, 2022-08-29:
If you revert to version 1.25.2, are the problems still there? For me it resolved all my issues, might be worth a try before buying different hardware.

zansko, 2022-09-11:
Finally managed to find some time.

I'm using managed hass OS so AFAIK there is no easy way to downgrade, unfortunately my backups are all on latest version as it's a quite fresh install.

Instead I moved z2m files (database, config, state) to a new machine and tested it out there where I could easily switch version.

---
I can reproduce issue on `1.25.2` and `1.27.2`.

What I did is created a simple script that would flood z2m with messages to turn off devices. I used `mosquitto_pub` and tried different number of devices - from 1 to 5.

Findings:
  * Issue is not reproducible with only one device.
  * The more device you add, the sooner the issue appears.
  * Setting `concurrency` to 1 does not resolve the issue

The issue seems to originate from https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/deconz/adapter/deconzAdapter.ts#L658. @KPWhiver mentioned that deconz has some kind of retry logic.

When I have more time I will dig deeper and see if I can resolve this. Any help appreciated.

chris400, 2022-09-28:
just arrived here.
Any news?
Facing the very same problem.

bednim, 2022-09-28:
I've moved from conbee2 to cc2652rb and much better but bit perfect (you
won't get perfect i don't think).

SeanHama, 2022-09-28:
Did you have to repair all your devices?

bednim, 2022-09-29:
Yep, it was worth it though. Works better than conbee2.

SeanHama, 2022-09-29:
The last firmware release for Conbee2 has been stable for me thank god.

mateusz-dziurdziak, 2022-09-29:
I had same issue with two Xiaomi ZNJLBL01LM (Aqara roller shade). Devices were located at the end of my apartment, farthest from my coordinater. Almost every command issues to those devices ended with error. There were 3 routers located in 5m range from xiaomi devices but it wasn't helpful.

What helped was buying long usb extension cable and moving coordinater nearer. It immediately worked. 

I know this is not a solution for most people but for some it may be a temporary fix

bednim, 2022-10-11:
Apparently not but last time when I was repairing all devices it took me
less than 30 mins. Not that bad.

Last time was changing channels, this time adaptor.

SeanHama, 2022-10-11:
Thanks - super annoying so Ill stick with conbee

KennethSeiero, 2022-10-11:
Hello all, I've previously had this issue, and none of the above worked.

However, i recently moved (to a new apartment), and in that process I disconnected all my devices. I noticed, that after disconnecting probably all my devices (apart from some lights), the issue were gone. 

I have no issues in my new apartment, haven't installed all the devices/plugs yet either. So I'm considering this issue to be related to some sort of bad zigbee route/binding?

Anyway, just wanted to let you know. Cheers.

wifiuk, 2022-11-05:
I am getting loads of these errors since the last update

Also using Conbee II 
[image]

Rustymage, 2022-11-10:
I'm facing the same issue.

```
2022-11-10 08:04:25Publish 'set' 'color' to 'upstairsbathroom' failed: 'Error: Command 0x00178801038e3b84/11 lightingColorCtrl.moveToColor({"transtime":0,"colorx":11272,"colory":48955}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
2022-11-10 08:04:36Publish 'set' 'brightness' to 'upstairsbathroom' failed: 'Error: Command 0x00178801038e3b84/11 genLevelCtrl.moveToLevelWithOnOff({"level":50,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
2022-11-10 08:04:47Publish 'set' 'color' to 'upstairsbathroom' failed: 'Error: Command 0x00178801038e3b84/11 lightingColorCtrl.moveToColor({"transtime":0,"colorx":45940,"colory":19595}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
```

```
Zigbee2MQTT version
    [1.28.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.2) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Coordinator type
    ConBee2/RaspBee2

Coordinator revision
    0x26580700

Coordinator IEEE Address
    0x00212effff07a4d8

Frontend version
    0.6.114

Stats

        Total 58
        By device type
        End devices: 39
        Router: 19
        By power source
        Battery: 37
        Mains (single phase): 19
        undefined: 2
        By vendor
        LUMI: 35
        Signify Netherlands B.V.: 9
        IKEA of Sweden: 7
        eWeLink: 3
        GLEDOPTO: 2
        Philips: 2
        By model
        lumi.weather: 11
        lumi.remote.b1acn01: 10
        lumi.sensor_magnet.aq2: 8
        LCT015: 7
        TRADFRI control outlet: 7
        lumi.sensor_wleak.aq1: 3
        lumi.sensor_motion.aq2: 2
        TH01: 2
        GL-C-008: 2
        LCT012: 1
        LST002: 1
        RWL021: 1
        LWA001: 1
        DS01: 1
        lumi.motion.ac02: 1
```

clickbg, 2022-11-14:
+1 from me, running ConBee 2 on a Proxmox VM:

```
Zigbee2MQTT version
    [1.28.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.2) commit: [360a777](https://github.com/Koenkk/zigbee2mqtt/commit/360a777)

Coordinator type
    ConBee2/RaspBee2

Coordinator revision
    0x26780700

Coordinator IEEE Address
    0x00212effff095771

Frontend version
    0.6.114

Stats

        Total 60
        By device type
        Router: 40
        End devices: 20
        By power source
        Mains (single phase): 40
        Battery: 20
        By vendor
        IKEA of Sweden: 55
        LUMI: 5
        By model
        TRADFRIbulbGU10WS345lm: 12
        TRADFRIbulbE14WSglobeopal470lm: 8
        TRADFRI SHORTCUT Button: 8
        TRADFRI control outlet: 4
        Remote Control N2: 4
        TRADFRI bulb E27 WW 806lm: 4
        TRADFRI bulb E27 WS clear 806lm: 4
        lumi.weather: 3
        TRADFRI Signal Repeater: 3
        TRADFRI motion sensor: 3
        TRADFRI bulb E27 CWS 806lm: 2
        ASKVADER on/off switch: 1
        TRADFRIbulbE14WScandleopal470lm: 1
        lumi.vibration.aq1: 1
        TRADFRIbulbE27WSglobeopal1055lm: 1
        lumi.sensor_wleak.aq1: 1
```

`2022-11-14 15:21:29Publish 'set' 'brightness' to 'Kitchen - Sphere Lamp' failed: 'Error: Command 0x8cf681fffe32d5c7/1 genLevelCtrl.moveToLevelWithOnOff({"level":216,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'`

`error 2022-11-14 15:34:25Publish 'set' 'brightness' to 'Kitchen - Sphere Lamp' failed: 'Error: Command 0x8cf681fffe32d5c7/1 genLevelCtrl.moveToLevelWithOnOff({"level":216,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'`

`error 2022-11-14 15:34:25Publish 'set' 'state' to 'Bedroom - Nightstand Lamp' failed: 'Error: Command 0x2c1165fffeb39046/1 genLevelCtrl.moveToLevelWithOnOff({"level":178,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'`

philyholiday, 2022-11-20:
Same issues for me with different device types (lights, switches, covers). Conbee II running on k3s on Raspberry Pi.

`Error 2022-11-19 18:20:22Publish 'set' 'state' to '0x0017880100c1b79e' failed: 'Error: Command 0x0017880100c1b79e/11 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":5}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-19 18:53:38Publish 'set' 'brightness' to '0x00178801089f85ed' failed: 'Error: Command 0x00178801089f85ed/11 genLevelCtrl.moveToLevelWithOnOff({"level":202,"transtime":5}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-19 22:03:09Publish 'set' 'state' to '0x00178801089f85ed' failed: 'Error: Command 0x00178801089f85ed/11 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":5}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-19 23:42:20Publish 'set' 'brightness' to '0x0017880100c1d2bb' failed: 'Error: Command 0x0017880100c1d2bb/11 genLevelCtrl.moveToLevelWithOnOff({"level":202,"transtime":5}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-20 00:04:09Publish 'set' 'position' to '0x54ef4410004eeb2a' failed: 'Error: Command 0x54ef4410004eeb2a/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":100}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-20 00:04:10Publish 'set' 'state' to '0x0017880101132fa8' failed: 'Error: Command 0x0017880101132fa8/11 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":5}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-11-20 00:27:44Publish 'set' 'position' to '0x847127fffea9c8b4' failed: 'Error: Command 0x847127fffea9c8b4/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":60}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'`

wifiuk, 2022-11-30:
I have also noticed a huge battery drain from battery devices since this started happening.

Nostras, 2022-11-30:
Sometimes I don't get an error either.

https://user-images.githubusercontent.com/4203238/204758045-49a30da4-5715-421c-b13b-f07fbcd9317e.mp4

I'm stumped honestly.

clickbg, 2022-11-30:
@Nostras , I just switched to SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P - actually bought 3 of them, two are flashed as routers 1 is flashed as coordinator. Works great, just don't set the antenna gain to 20 as it crashes the unit. Something like 9 works great. Also make sure to get the -P which has a CC2652P chip rather than the -E device which has EFR32MG21 chip. The -E might work well but I have no experience with it and is marked as experimental for Z2M.

philyholiday, 2022-12-01:
I downgraded Conbee II to revision 0x26580700.
Same error messages ("writeUndiv":false}) failed (no response received)) and behaviour (sometimes devices don't react).

Strange thing is, it just started a few weeks ago without changing the environment other than upgrading zigbee2mqtt to latest releases (can't tell with which version it started).

Nostras, 2022-12-02:
Thanks @clickbg, I just ordered one. It should arrive on the fifth.

Noticeably it seems that my philips hue light (Hue Filament Globe XL Ambiance E27), GU (TRADFRIbulbGU10WS345lm) and tradfri light downstairs (TRADFRIbulbE27WSglobeopal1055lm) behave as expected (I also have some inputs which I won't consider here, but these seem to work OK as well). It seems that the lights that are in the living room are exclusively giving me a headache. After some more experimentation it seems there's a really weird interaction with Adaptive Lighting that really breaks some things.
https://github.com/basnijholt/adaptive-lighting

After disabling adaptive lighting control the devices seem to behave again.

Weirdly this only applies to the tradfri, the hue light is also included in this control but seems to don't care at all (as in, keeps working).

To designate the actual real root cause here is really tough, but I can consistently break and "fix" it.
@Koenkk maybe you can chime in real fast and tell me if it's something zigbee2mqtt could have a hand in.

Barsonax, 2022-12-06:
I had massive issues with the conbee2 with devices working for a short while after they were paired but then after some minutes nothing would happen anymore.

Then I downgraded the firmware to 0x26580700 as I read in https://www.zigbee2mqtt.io/guide/adapters/#other that newer firmwares are unstable. After that I reset the (hue) lamps and repaired them. This fixed all problems. This also caused the devices to be properly shown on the Map with a link.

Going to test with a few lamps for now to see if its really stable, if not I will let you guys now.

wifiuk, 2022-12-08:
so moving to a Sonof 3.0 USB, not had a single error since.

Svisstack, 2023-01-07:
same issue here; did anyone tried downgrading conbee2 below the 0x26580700 ?

The problem looks unrelated to the zigbee2mqtt apart from the fact that the software could have optional retry logic setting like 1s timeout 3 times retry on timeout that would help resolve possible other issues @Koenkk - anything like this is on the roadmap? It's complex to add?

Basically it looks like conbee2 have a bug in all or starting from currently unknown version of the firmware when they drop or unable to deliver messages, usually when the multiple messages are sent in short time span but not limited to. They do not have an issue within their own ecosystem because they have 1s timeout with retry and this I think solved all their issues on the hardware/protocol level.

I do not want to repair, so I'm looking how to make this work.

Set the concurrency to 1 and delay to 500ms - will see if this make it acceptable but the issue is still there as I able to reproduce.

@Barsonax let us know if this fixed your problem 

```

Zigbee2MQTT version
1.29.1 commit: unknown
Coordinator type
ConBee2/RaspBee2
Coordinator revision
0x26720700
Coordinator IEEE Address
0x00212effff07b9f3
Frontend version
0.6.121
Stats
Total 33
By device type
Router: 33
By power source
Mains (single phase): 33
By vendor
_TZ3000_tqlv4ug4: 13
_TZ3000_ltt60asa: 9
_TZ3000_4uuaja4a: 8
_TZ3000_ajv2vfow: 2
_TZ3210_wdexaypg: 1
By model
TS0001: 15
TS0004: 9
TS130F: 8
TS110E: 1
```

Barsonax, 2023-01-07:
> same issue here; did anyone tried downgrading conbee2 below the 0x26580700 ?
>
> The problem looks unrelated to the zigbee2mqtt apart from the fact that the software could have optional retry logic [...]

Since using 0x26580700 I have not had issues anymore.

Doesn't feel good though that iam basically stuck at a older firmware though. Thinking of moving to a sonoff.

fthiery, 2023-01-07:
Hi there; also getting these issues while trying to understand why my HA group of 8 lights would not consistently toggle. Digging further i found out that while the lamps DO turn on in reality, the state in zigbee2mqtt (and in HA) is wrong for some of the devices (still OFF). Because i use HA's generic toggle service, and because the state of the devices contained within the group is not consistent, it appears to the user that the button triggering the toggle needs to be woken up (because you need to repeatedly press it until it finally does something). I'm considering to switch to a 2 button action (one button for ON, one for OFF) to avoid this toggle problem -- obviously this is out of scope of z2m but i thought that may help others.

The errors visible in the z2m web UI pointed me to this issue.

To summarize, some of the fix/workaround suggestions are
A) define concurrency to 1 and/or set a 350-500 ms delay (at /#/settings/settings)
B) upgrade or downgrade the Conbee II firmware
C) switch to Sonoff 3.0 USB dongle plus

Tried A, did not seem to do anything for me. Am i required to restart z2m after changing such values using the web UI or is that supposed to be active on the fly ?

Tried B, without success. I even reached a point where z2m would not start anymore. Plus, it's completely opaque what the Conbee 2 firmware updates contain/do (or does anyone have any pointers/details ?). I also don't like the idea to stick to a past/unmaintained firmware.

Regarding C, the opening post contains "Tested with Sonoff 3.0 USB dongle plus and the same problem.". So why would switching to Sonoff fix the problem (plus, it requires you to repair everything, right ?). Could it be that the actual "solution" is to re-pair everything ? @Jtef can you tell us if you tried the [P or E variant](https://www.zigbee2mqtt.io/guide/adapters/) of the Sonoff ? Did you solve your issue at all ?

fthiery, 2023-01-07:
Suspecting that settings in the web UI of z2m were not applied on-the-fly, i restarted z2m and it seems to fix my problem ! However, i agree that if deconz have a retry mecanism, maybe we should have one too.

Svisstack, 2023-01-07:
> @ github. com>

Yes and that will be in the scope; basically the software is router of the messages and retry option make sense; just store attempt number in the message and retry on failure while incrementing the number but I do not know if this is really that simple

Artur Pietrzyk
tel.: (+48) 512 085 564

Jtef, 2023-01-07:
> Hi there; also getting these issues while trying to understand why my HA group of 8 lights would not consistently toggle. Digging further i found out that while the lamps DO turn on in reality, the state in zigbee2mqtt (and in HA) is wrong for some of the devices (still OFF). Because i use HA's generic toggle service, and because the state of the devices contained within the group is not consistent, it appears to the user that the button triggering the toggle needs to be woken up (because you need to repeatedly press it until it finally does something). I'm considering to switch to a 2 button action (one button for ON, one for OFF) to avoid this toggle problem -- obviously this is out of scope of z2m but i thought that may help others.
> 
> The errors visible in the z2m web UI pointed me to this issue.
> 
> To summarize, some of the fix/workaround suggestions are A) define concurrency to 1 and/or set a 350-500 ms delay (at /#/settings/settings) B) upgrade or downgrade the Conbee II firmware C) switch to Sonoff 3.0 USB dongle plus
> 
> Tried A, did not seem to do anything for me. Am i required to restart z2m after changing such values using the web UI or is that supposed to be active on the fly ?
> 
> Tried B, without success. I even reached a point where z2m would not start anymore. Plus, it's completely opaque what the Conbee 2 firmware updates contain/do (or does anyone have any pointers/details ?). I also don't like the idea to stick to a past/unmaintained firmware.
> 
> Regarding C, the opening post contains "Tested with Sonoff 3.0 USB dongle plus and the same problem.". So why would switching to Sonoff fix the problem (plus, it requires you to repair everything, right ?). Could it be that the actual "solution" is to re-pair everything ? @Jtef can you tell us if you tried the [P or E variant](https://www.zigbee2mqtt.io/guide/adapters/) of the Sonoff ? Did you solve your issue at all ? Also, i make heavy use of Ikea Tradfri scene launchers / buttons (battery-powered), and there are [reports](https://community.home-assistant.io/t/sonoff-zigbee-usb-dongle-not-working-with-ikea-tradfri-buttons/385878) of incompatibility and/or battery drain... Which makes me even more reluctant to switch over.

@fthiery  I tested with Sonoff ZBDongle-P. The E version is new, haven't tried it. I haven't found a solution for the issue but I'm using Conbee with Deconz and the Sonoff P with Z2m. The heating relays and the light switches (basically all the devices that are turned on/off in the same time) are in Deconz. The garden lights (3 bulbs), all the sensors, and individual sockets are still on z2m and rarely I get a timeout error (5-6 months). I've also implemented a retry mechanism for the garden bulbs (an automation in Home Assistant, it runs every 5 minutes) and if the sun is up and any garden bulb is on it turns it off. 
Looking forward to a fix so I can rely on Z2m and remove Deconz.

Drealine, 2023-02-03:
Same issue here.
Add 350 delay solve the issue at the time.

jhbruhn, 2023-03-06:
This is still happening very often for me, including lights turning on by themselves when the error appears.

fthiery, 2023-03-06:
Anyone knows which alternative zigbee stick would be a drop-in replacement (i.e. without repairing everyting ?)

bednim, 2023-03-06:
there isnt any unfortunately.

I personally went with sale.sh CC2652RB stick.

still - its not perfect - but I dont think any network will be perfect. it
all matters like end devices, routers etc etc.

sonoff zbmini for example - crap antenna - disconnects all the time!

good luck

Svisstack, 2023-03-06:
Setting high delay 500ms or 1s resolved issue for me but significantly reduced bandwidth; however i do not need high bandwidth from this network - i must admit that i have in almost all automations last action repeated 3 times with some delay to mitigate the risk of lost packets on the zigbee layer; I should probably remove that to ensure that issue is trully resolved but this setup works for me for few months without issues

Artur Pietrzyk
tel.: (+48) 512 085 564

[32 more replies]

closed 2023-06-20 by github-actions[bot]