Koenkk/zigbee2mqtt #11648: Tuya TS011F  (TZ3000_typdpbpg) powers off

phoxy84, 2022-02-28:
### What happened?

When I use the device they shut down suddenly. This happens maybe once every one or two days. Don't know if it is a Z2M issue or a problem with the plug.

I have four of them, all have the same problem.

I made an automation to power the device on automatically but that cannot be the solution :)
[image]

### What did you expect to happen?

The plug just stays on.

### How to reproduce it (minimal and precise)

I don't know. Just power on the device and wait.

### Zigbee2MQTT version

1.23.0-dev commit

### Adapter firmware version

20211217

### Adapter

zzh

### Debug log

`debug 2022-02-28 10:58:30: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-02-28 10:58:30: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0.36,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":79,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":238}'
debug 2022-02-28 10:58:30: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-02-28 10:58:30: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0.36,"energy":2.51,"indicator_mode":null,"linkquality":65,"power":79,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":238}'
debug 2022-02-28 10:58:30: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":4}' from endpoint 1 with groupID 0
debug 2022-02-28 10:58:30: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":4}'
debug 2022-02-28 10:58:30: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":0}' from endpoint 1 with groupID 0
info  2022-02-28 10:58:30: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0.36,"energy":2.51,"indicator_mode":null,"linkquality":69,"power":79,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":0}'
debug 2022-02-28 10:58:30: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
debug 2022-02-28 10:58:30: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
info  2022-02-28 10:58:30: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0.36,"energy":2.51,"indicator_mode":null,"linkquality":69,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":0}'
info  2022-02-28 10:58:30: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0,"energy":2.51,"indicator_mode":null,"linkquality":69,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":0}'
debug 2022-02-28 10:58:35: Received MQTT message on 'zigbee2mqtt/iMac/set' with data 'ON'
debug 2022-02-28 10:58:35: Publishing 'set' 'state' to 'iMac'
info  2022-02-28 10:58:35: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":0}'
debug 2022-02-28 10:58:35: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-02-28 10:58:35: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":0}'
debug 2022-02-28 10:58:36: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-02-28 10:58:36: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":0}'
debug 2022-02-28 10:58:37: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":0}' from endpoint 1 with groupID 0
debug 2022-02-28 10:58:37: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":0}'
debug 2022-02-28 10:58:37: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":241}' from endpoint 1 with groupID 0
info  2022-02-28 10:58:37: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":241}'
debug 2022-02-28 10:58:41: Received MQTT message on 'zigbee2mqtt/iMac/set' with data 'ON'
debug 2022-02-28 10:58:41: Publishing 'set' 'state' to 'iMac'
info  2022-02-28 10:58:41: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
debug 2022-02-28 10:58:42: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-02-28 10:58:42: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
debug 2022-02-28 10:58:56: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":39}' from endpoint 1 with groupID 0
info  2022-02-28 10:58:56: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0,"energy":2.51,"indicator_mode":null,"linkquality":51,"power":39,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
debug 2022-02-28 10:58:59: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":171}' from endpoint 1 with groupID 0
info  2022-02-28 10:58:59: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0.17,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":39,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
debug 2022-02-28 10:59:01: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":95}' from endpoint 1 with groupID 0
info  2022-02-28 10:59:01: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0.17,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":95,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
debug 2022-02-28 10:59:03: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":392}' from endpoint 1 with groupID 0
info  2022-02-28 10:59:03: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0.39,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":95,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
debug 2022-02-28 10:59:10: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":107}' from endpoint 1 with groupID 0
info  2022-02-28 10:59:10: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0.39,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":107,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
debug 2022-02-28 10:59:12: Received Zigbee message from 'iMac', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":458}' from endpoint 1 with groupID 0
info  2022-02-28 10:59:12: MQTT publish: topic 'zigbee2mqtt/iMac', payload '{"child_lock":null,"current":0.46,"energy":2.51,"indicator_mode":null,"linkquality":47,"power":107,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":241}'
`

Koenkk, 2022-02-28:
Can you provide the debug log when the switch turns off by itself?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

phoxy84, 2022-03-01:
I thought I did, here for an other device, off at 15:29:31, turned on by automation 15 seconds later, 15:29:46

`debug 2022-03-01 15:29:31: Received Zigbee message from 'waterkoker', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-03-01 15:29:31: MQTT publish: topic 'zigbee2mqtt/waterkoker', payload '{"child_lock":"UNLOCK","current":0,"energy":3.11,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":237}'
debug 2022-03-01 15:29:31: Received Zigbee message from 'waterkoker', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":0}' from endpoint 1 with groupID 0
info  2022-03-01 15:29:31: MQTT publish: topic 'zigbee2mqtt/waterkoker', payload '{"child_lock":"UNLOCK","current":0,"energy":3.11,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":0}'
debug 2022-03-01 15:29:31: Received Zigbee message from 'waterkoker', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":0}' from endpoint 1 with groupID 0
info  2022-03-01 15:29:31: MQTT publish: topic 'zigbee2mqtt/waterkoker', payload '{"child_lock":"UNLOCK","current":0,"energy":3.11,"indicator_mode":"off/on","linkquality":80,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":0}'
debug 2022-03-01 15:29:31: Received Zigbee message from 'waterkoker', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":4}' from endpoint 1 with groupID 0
debug 2022-03-01 15:29:31: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":4}'`

`debug 2022-03-01 15:29:38: Received Zigbee message from 'waterkoker', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":0}' from endpoint 1 with groupID 0
debug 2022-03-01 15:29:38: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":0}'
debug 2022-03-01 15:29:38: Received Zigbee message from 'waterkoker', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":236}' from endpoint 1 with groupID 0
info  2022-03-01 15:29:38: MQTT publish: topic 'zigbee2mqtt/waterkoker', payload '{"child_lock":"UNLOCK","current":0,"energy":3.11,"indicator_mode":"off/on","linkquality":72,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":236}'`

`debug 2022-03-01 15:29:46: Received MQTT message on 'zigbee2mqtt/waterkoker/set' with data 'ON'
debug 2022-03-01 15:29:46: Publishing 'set' 'state' to 'waterkoker'
info  2022-03-01 15:29:46: MQTT publish: topic 'zigbee2mqtt/waterkoker', payload '{"child_lock":"UNLOCK","current":0,"energy":3.11,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":236}'
debug 2022-03-01 15:29:46: Received Zigbee message from 'waterkoker', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-03-01 15:29:46: MQTT publish: topic 'zigbee2mqtt/waterkoker', payload '{"child_lock":"UNLOCK","current":0,"energy":3.11,"indicator_mode":"off/on","linkquality":76,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":236}'`

Koenkk, 2022-03-01:
I don't see any request send from z2m that turns of the device; so there are 2 possibilities left:
- the plug turns off by itself
- another device in your network turns off the plug, this can be checked via sniffing: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

durkimat, 2022-03-02:
I have a very similar scenario. I have 5 TS011F, all slightly different (different LED light colours, labelling etc) identifying as either TS_011F_plug_1 or TS_011F_plug_3. At first I thought everything was fine, but then one started randomly turning off. Now another two (including the one hosting my internet and zigbee2mqtt and HA services) has started doing the same, so I now have 3 of the 5 doing it. another has a different problem, so am not using, and the 5th reports as 'plug_1' and is the only one that appears to be completely stable. Sorry I have no answer to give, but am definitely interested and thought I would share. I've seen a few other threads with similar issues elsewhere.

sadrov, 2022-03-02:
Interestingly enough, I have 4 TS011F_plug_3 plugs and as of yesterday one of them started to show the same behaviour. I already tried resetting it but the same happens after disconnecting it from the zigbee network and switching on the plug using the built in hardware button. It looks like it stays on as long as nothing is plugged in, but of course that's not very useful. 
When it first happened the plug was providing 2500w for a certain time so I thought maybe it overheated and switched itself as a result of a built in protection mechanism or something but now it shuts off also at more modest wattages so I'm not sure what it could be. It's definitely not being controlled by another zigbee device because it also switches off when not connected to any network. It's also irregular; sometimes it's after 15 minutes, sometime it stays one for an hour...  Hopefully the other plugs won't be next .

andreamoretto, 2022-03-03:
I have reported a very similar issue, see #11500. I have some TS011F_plug_1 plugs that behaves exactly the same as yours, and I have some other TS011F_plug_3 that does not show this issue at all.

As far as I understand, the affected variants may have an issue reading the voltage: when the measured voltage drops to 0, the plug disconnect the load. Remember that "voltage" is the input voltage, not the output voltage (which depends on the load switch being open or close). I monitor several appliances in my home, all in the same circuit branch, and only TS011F_plug_1 plugs report temporary drop to 0V in voltage.

My guess is that's a firmware issue, do any of you guys know if the firmware can be upgraded somehow?

phoxy84, 2022-03-03:
Thanks for the response. I have a spare one that I just plugged in the socket, without load. After a couple of days, that one is still on. I will try again with load, but not connected to zigbee.

I don't have a tuya bridge and don't know if there is any update available.

sadrov, 2022-03-03:
Yes I did the same test, mine stayed on all night and morning without a load and connected to a z2m network. At 9 this morning I plugged in a load and it switched off around noon, every since then I switch it on manually when I notice it went off but it always goes off again after 15-60 minutes, same behaviour when not connected to a zigbee network. I tried with this setting but to no avail
[image]
I'm not sure how to check the firmware version of these, I don't have a tuya bridge either.

giejay, 2022-03-04:
Having the exact same issue. I have 3 TS011F_plug_1 and 3 TS011F_plug_3. The plug_3 devices work fine, never go off, the plug_1 randomly turn off.

Still ON:
debug 2022-03-03 22:02:22: Received Zigbee message from 'camera_stijn_switch', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,493]}' from endpoint 1 with groupID 0
info  2022-03-03 22:02:22: MQTT publish: topic 'zigbee2mqtt-meter-pi/camera_stijn_switch', payload '{"current":0.02,"energy":4.93,"linkquality":98,"power":4,"power_outage_memory":"on","state":"ON","voltage":235}'

Going Off:
debug 2022-03-03 22:03:15: Received Zigbee message from 'camera_stijn_switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-03-03 22:03:15: MQTT publish: topic 'zigbee2mqtt-meter-pi/camera_stijn_switch', payload '{"current":0.02,"energy":4.93,"linkquality":98,"power":4,"power_outage_memory":"on","state":"OFF","voltage":235}'

navycrow, 2022-03-05:
Same problem here with a fresh TS011F_plug_1. It works for 3/4 days then goes off. Last time, in the night (according history) while there was no activity on it.

nickles-lee, 2022-03-08:
Also running into this problem. I'm running the latest firmware, so perhaps it's a bug in that?

MaxKorlaar, 2022-03-08:
Same here with a [TS011F_plug_3](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3). No logs unfortunately as it happened to happen with the exact plug that's powering my server 🤦🏻 . Also one of the plugs was manufactured with a defect so it immediately tripped the ground breakers, might be worth warning for on the docs for these cheap plugs..

phoxy84, 2022-03-08:
> Also running into this problem. I'm running the latest firmware, so perhaps it's a bug in that?

Did you update them yourself? With a (tuya) bridge? Maybe we can download that firmware with the manual provided on the z2m website ([https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html](url)

nickles-lee, 2022-03-09:
> > Also running into this problem. I'm running the latest firmware, so perhaps it's a bug in that?
> 
> Did you update them yourself? With a (tuya) bridge? Maybe we can download that firmware with the manual provided on the z2m website [here](https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html)

I installed the updates that were showing up in `zigbee2mqtt`, I've never had them tied into a Tuya bridge.

MattWestb, 2022-03-09:
If the device is supporting Over Current Alarm it can being implanted by attribute `0x8003` and its possible see if its the protection that  is shutting it down.

andreamoretto, 2022-03-09:
> If the device is supporting Over Current Alarm it can being implanted by tuya attribute `0x8003` and its possible see if its the protection that is shutting it down.

How to do this?

nickles-lee, 2022-03-10:
@MattWestb just to provide a bit of skepticism to that: I've been able to run my oven's cleaning cycle without any random shutoffs (~2.9kW sustained), but I'll get random shutoffs in the middle of the night when there's no electrical load to speak of (<5W)

I have a watchdog automation that keeps things running, but the pattern to the shutoffs makes me think that it's something other than overcurrent protection being tripped.

MattWestb, 2022-03-10:
@nickles-lee I dont have the device but i knowing tuya have implanted the attribute is some device and **if** the device is supporting it i think its good to knowing if its shutting down for protection. Also some user looks only having problem then they is having load on the device but it can being hardware problems and also firmware that is making the bad thing but in the end its can helping implanting the attribute and returning the device to the seller **if** its being triggered and the max load is not being used.

navycrow, 2022-03-11:
I don't know if it can help. I noticed while looking at the history that the socket does not only turn off, it also reports when this happens a voltage equal to 0 (A normal switch off doesn't change voltage)

MattWestb, 2022-03-11:
@navycrow Its sounds like over current protection is disabling the output then its zero V.
Only way to knowing it is implanting the tuya attribute `0x8003` and you can getting it reported if its happens.
I think its not the same problems for all users but i think some is have the problems with the over current protection.

navycrow, 2022-03-12:
Maybe but personally, I don't really believe in this hypothesis. In my case, the socket always cuts when it does not work.

RHAD1969, 2022-03-12:
I have the same problem with the ts011f_plug_1. Cut off power randomly.

phoxy84, 2022-03-15:
> I installed the updates that were showing up in `zigbee2mqtt`, I've never had them tied into a Tuya bridge.

For both my devices, there are no z2m ota updates available. I have _TZ3000_u5u4cakc (TS011F_plug_3) and _TZ3000_typdpbpg (TS011F_plug_1), running z2m 1.24.0-dev commit: [42f229b]

meceo, 2022-03-15:
I have the same issue. See video [on dropbox](https://www.dropbox.com/s/9wyl3kklj4f4ssl/zigbee%20sockets%20issue.mov?dl=0) showing socket turning off by itself 2 times.

giejay, 2022-03-16:
So, as a first step I removed 2 of my 3 Tuyas from the socket to see if they where somehow interfering with each other. That was not the case, the last one still turned off. Then I removed the last one from my ZigBee network to see if that does anything. Still turning off after a day or so. 

The only way to use them now is to make an automation in HA and not having them connected to a critical system

navycrow, 2022-03-16:
I complained to the Aliexpress seller. He consulted Tuya who told him that a Tuya hub was needed for it to work properly. Of course, I don't believe it but to remove the doubt, I bought one and paired the sockets on it (4). For the moment none, after 1 day no socket has cut. On the other hand, I notice in the Tuya Smart Life application that the socket has Error parameters (OV_CR, Over voltage, Over power, Under current, Under voltage, under power). Can this be integrated into Z2MQTT??

giejay, 2022-03-16:
Can you remove one of the sockets from the Tuya bridge to see if that introduces the issue again?

jar87, 2022-03-16:
Hey! I think i found solution for that issue. You need Zigbee/Tuya gateway. Disconnect that smart plug from z2m and connect it via gateway and Your Tuya/SmartLife app. Then start updating firmware. After succesfully update re-pair with z2m again and that`s all. It should work (at least in my case is working perfectly!).

phoxy84, 2022-03-16:
> Hey! I think i found solution for that issue. You need Zigbee/Tuya gateway. Disconnect that smart plug from z2m and connect it via gateway and Your Tuya/SmartLife app. Then start updating firmware. [...]

Sounds great! I don't have Tuya bridge. Is it possible to get the ota update and implement in z2m?

jar87, 2022-03-16:
> Sounds great! I don't have Tuya bridge. Is it possible to get the ota update and implement in z2m?

To be honest i`ve checked OTA and it`s not showing update/new firmware... That`s why i decided update it via gateway and it worked!

meceo, 2022-03-16:
I was able to trigger update of tuya devices via zigbee2mqtt
[image]

atchoum31, 2022-03-16:
> I was able to trigger update of tuya devices via zigbee2mqtt
> [image]

Already done few days ago. Didn’t fix.

jar87, 2022-03-16:
@atchoum31 and @meceo few minutes ago i`ve tried update via z2m and it didn`t work...
[image]

i`m updating rest of sockets by gateway, i will write u results soon ;)

@navycrow idk wha`t wrong but new version should be 1.0.5
[image]

phoxy84, 2022-03-16:
No updates here. 

[image]

jar87, 2022-03-16:
Like i wrote i had to re-pair my sockets to zigbee/tuya bridge/gateawy, and it`s updating firmawe. After succesfully updated i`m re-pairing socket with my cc2652p with z2m and it works perfectly!:)

durkimat, 2022-03-17:
I've ordered a hub to try and get my 5 sockets working. Will see if it reports firmware version as one of my 5 is working fine. Would be better if ota could work in z2m obvs. Will post any findings here...

jar87, 2022-03-17:
@durkimat Since yesterday they`re working. I`ve updated all sockets. Only issue which i notices and also info is on z2m site is about reporting. Plug_1 isn`t reporting because of some issue. Hopefully they will fix it asap..

durkimat, 2022-03-17:
I had one randomly stop reporting as well. I removed and re-added, which didn't help, in the end I rebooted following ha upgrades and it started reporting again. It was less priority for me than them switching off considering my fridge was connected to one of them 🙄

jar87, 2022-03-17:
I`ve restarted HA, re-paired sockets - didn`t work at all :(

navycrow, 2022-03-17:
After 2 days with tuya hub. I can confirm that it did not solve anything. Sockets continue to cut off. I NOT recommand [haozee Official Store](https://fr.aliexpress.com/store/203725) for this model.

phoxy84, 2022-03-17:
I ordered a Tuya bridge and will keep you all posted

bolidesoft, 2022-03-18:
Same problem with two my TS011F plugs. I am sure it is either firmware or plug's hardware issue. It is not related to zigbee2mqtt. A left two my plugs on without any load and without zigbee network for the night and found them both turned off in the morning.
And one more bad thing about these plugs. When I unpacked them, something was inside of one plug when I shake it. Finally, a piece of tinned wire about 8 mm long fell out of the socket! I've got mine from "Aubess Smart Scene Store" on Aliexpress

phoxy84, 2022-03-18:
I have some new logs. The plug is connected to a monitor, maximum power 50w. I turned on the monitor at 08:01:44 (so shut down was before that). Don't know if it is helpful in any way.

Plug shuts down at 07:55:17, voltage is 0. 
`debug 2022-03-18 07:55:17: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-03-18 07:55:17: MQTT publish: topic 'zigbee2mqtt/2e beeldscherm', payload '{"child_lock":"UNLOCK","current":0,"energy":2.51,"indicator_mode":"off/on","linkquality":54,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":234}'
debug 2022-03-18 07:55:17: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":0}' from endpoint 1 with groupID 0
info  2022-03-18 07:55:17: MQTT publish: topic 'zigbee2mqtt/2e beeldscherm', payload '{"child_lock":"UNLOCK","current":0,"energy":2.51,"indicator_mode":"off/on","linkquality":58,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":0}'
debug 2022-03-18 07:55:18: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-03-18 07:55:18: MQTT publish: topic 'zigbee2mqtt/2e beeldscherm', payload '{"child_lock":"UNLOCK","current":0,"energy":2.51,"indicator_mode":"off/on","linkquality":54,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":0}'
debug 2022-03-18 07:55:20: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":0}' from endpoint 1 with groupID 0
info  2022-03-18 07:55:20: MQTT publish: topic 'zigbee2mqtt/2e beeldscherm', payload '{"child_lock":"UNLOCK","current":0,"energy":2.51,"indicator_mode":"off/on","linkquality":43,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":0}'`

I also see some errors:
`debug 2022-03-18 07:55:18: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":4}' from endpoint 1 with groupID 0
debug 2022-03-18 07:55:18: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":4}'
debug 2022-03-18 07:55:18: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":4}' from endpoint 1 with groupID 0
debug 2022-03-18 07:55:18: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":4}'
debug 2022-03-18 07:55:18: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":4}' from endpoint 1 with groupID 0
debug 2022-03-18 07:55:18: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":4}'
debug 2022-03-18 07:55:18: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":4}' from endpoint 1 with groupID 0
debug 2022-03-18 07:55:18: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":4}'
debug 2022-03-18 07:55:18: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":4}' from endpoint 1 with groupID 0
debug 2022-03-18 07:55:18: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":4}'
debug 2022-03-18 07:55:19: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":4}' from endpoint 1 with groupID 0
debug 2022-03-18 07:55:19: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":4}'
debug 2022-03-18 07:55:19: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":4}' from endpoint 1 with groupID 0
debug 2022-03-18 07:55:19: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":4}'
debug 2022-03-18 07:55:19: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":4}' from endpoint 1 with groupID 0
debug 2022-03-18 07:55:19: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":4}'
debug 2022-03-18 07:55:19: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":4}' from endpoint 1 with groupID 0
debug 2022-03-18 07:55:19: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53249":4}'`

Complete log, filtered by device:
```
debug 2022-03-18 07:26:02: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
[... cut ...] date_available":null,"voltage":235}'
debug 2022-03-18 07:55:35: Received Zigbee message from '2e beeldscherm', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-03-18 07:55:35: MQTT publish: topic 'zigbee2mqtt/2e beeldscherm', payload '{"child_lock":"UNLOCK","current":0,"energy":2.51,"indicator_mode":"off/on","linkquality":43,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":235}'
```

navycrow, 2022-03-18:
> When I unpacked them, something was inside of one plug when I shake it. Finally, a piece of tinned wire about 8 mm long fell out of the socket! I've got mine from "Aubess Smart Scene Store" on Aliexpress
Same problem with of mine from [haozee Official Store](https://fr.aliexpress.com/store/203725). Very disapointed.

MaxKorlaar, 2022-03-18:
> > When I unpacked them, something was inside of one plug when I shake it. Finally, a piece of tinned wire about 8 mm long fell out of the socket! I've got mine from "Aubess Smart Scene Store" on Aliexpress
> 
> Same problem with one of mine from [haozee Official Store](https://fr.aliexpress.com/store/203725). Very disapointed.

+1, one out of four that I ordered from that same store triggered my earth-leakage circuit breaker the moment I plugged something in, quite dangerous.

RHAD1969, 2022-03-18:
> Hey! I think i found solution for that issue. You need Zigbee/Tuya gateway. Disconnect that smart plug from z2m and connect it via gateway and Your Tuya/SmartLife app. Then start updating firmware. [...]

What is the most recent firmware?

jar87, 2022-03-18:
@RHAD1969  all sockets were updated to 1.0.5 firmware but i noticed it`s not reporting power consumption at all:(

RHAD1969, 2022-03-18:
@jar87 I've added the TS011F_plug_1 to a Smart-bridge10 and it is reporting version V1.0.10, but I still have the sudden cutoffs problem

phoxy84, 2022-03-18:
> @RHAD1969 all sockets were updated to 1.0.5 firmware but i noticed it`s not reporting power consumption at all:(

I think they work with polling, is it a plug_3, or (still) plug_1?

RHAD1969, 2022-03-18:
@phoxy84 It reports as a JLT1 on my smart-bridge10 and it is reporting itself as a TS011F_plug_1 on Zigbee2MQTT

phoxy84, 2022-03-18:
> @phoxy84 It reports as a JLT1 on my smart-bridge10 and it is reporting itself as a TS011F_plug_1 on Zigbee2MQTT

Sorry, don't know then. Did you open an issue already on that problem?

jar87, 2022-03-18:
@phoxy84 in my caase plug_1 is still plug_1;)

jar87, 2022-03-18:
Hopefully future update of z2m will solve problem.

RHAD1969, 2022-03-20:
@phoxy84  
> Sorry, don't know then. Did you open an issue already on that problem?

I've opened an issue. The responded with obvious answers. They want the ID of my device to push an update. But my device is not connected to a Tuya bridge and therefore not visible on their platform.

I just connected one plug using my smart-bridge10 and connected my 'always on' laserprinter to the plug. I'm curious if I get power cut offs using this config. I hope to eliminate Home Assistant / Zigbee2Mqt as the cause of this issue.

durkimat, 2022-03-20:
Ok, got a couple hooked up to a tuya ZigBee hub. They're on v1.04 and there's no update. They're still turning off. In the logs for one or turned off twice in a row, it wasn't off the second time, it still triggered the log event.

RHAD1969, 2022-03-21:
Tuya just got back at me via Whatsapp, so they seems to take the issue seriously. I'll keep you posted.

jar87, 2022-03-21:
Hopefully they will fix it asap :)

navycrow, 2022-03-21:
Strange thing.
I paired my 4 sockets on tuya bridge 8 days ago. The firmware was 1.0.10.
In despite, they continue to cut off once or twice a day.
On wednesday, i made a new request to the seller and sent the virtual ids
of the sockets (found in Tuya Smart Life app).
Since wednesday, no sockets had cut off ???? !!!!! However the firmware is
still 1.0.10.
Weird !!

phoxy84, 2022-03-21:
> Tuya just got back at me via Whatsapp, so they seems to take the issue seriously. I'll keep you posted.

That sounds great, hope they fix it.

> Strange thing. I paired my 4 sockets on tuya bridge 8 days ago. The firmware was 1.0.10. In despite, they continue to cut off once or twice a day. On wednesday, i made a new request to the seller [...]

Well, I use four of them and they don't shot down every day, since last week:
1 turned of twice (cordless vacuum cleaner)
1 turned of four times (pc monitor)
1 turned of five times (kettle)
1 tuned of twice (floor heating pump)

The last is I think the most used one. The kettle has the biggest spike in watts.

navycrow, 2022-03-21:
Well, less then one hour after my latest post, one socket cut off again. 
So finally : not solved at all. I believed it :(

[510 more replies]