Koenkk/zigbee2mqtt #11649: IKEA Starkvind not reporting PM25 or Air Quality

stefanschoemaker, 2022-02-28:
### What happened?

I setup the IKEA Starkvind but the PM25 reports -1 and the Air Quality as unknown.

Status:

{
    "air_quality": "unknown",
    "child_lock": "UNLOCK",
    "fan_mode": "3",
    "fan_speed": 3,
    "fan_state": "ON",
    "led_enable": true,
    "linkquality": 123,
    "pm25": -1,
    "replace_filter": false
}

### Zigbee2MQTT version

1.23.0-dev commit: 987205c

### Adapter firmware version

20201026

### Adapter

slae.sh_cc2652rb

sjorge, 2022-02-28:
How long has the device been on? Mine usually only starts reporting after about 5-10 minutes.
Also can you share a screenshot of the devices reporting tab.

Koenkk, 2022-02-28:
@sjorge could it be due to the outdated coordinator firmware? 20201026

stefanschoemaker, 2022-03-02:
Here is the reporting screenshot:
[image]

I will try to update the coordinator later

stefanschoemaker, 2022-03-02:
I've just updated the Coordinator firmware to 20211217 and let it run for an hour, but still isn't reporting

sjorge, 2022-03-02:
OK at least the reporting matches what I have/is in the device definition
[image]

Koenkk, 2022-03-02:
@stefanschoemaker after updating to the latest fw, did you also try to 
- force remove the device via the frontend
- stop z2m
- start z2m
- repair the device.

stefanschoemaker, 2022-03-06:
@Koenkk yeah did that also, still the same result

stefanschoemaker, 2022-03-08:
okay I think it's a problem with the Starkvind itself, I had a old Ikea Tradfri gateway lying around and connected to that. In the tradfri app it's saying "Collecting information"

stefanschoemaker, 2022-03-11:
It was indeed a problem with the unit, got it replaced and worked instantly. Thanks for helping!

sjorge, 2022-03-11:
Glad to hear! Enjoy the STARKVIND!