Koenkk/zigbee2mqtt #11663: Tuya TS0202 TZ3000_mmtwjmaq don't show Battery %

jocamane, 2022-03-01:
### What happened?

Tuya TS0202 TZ3000_mmtwjmaq don't show Battery %

[image]

### How to reproduce it (minimal and precise)

When join don't show battery %

### Zigbee2MQTT version

1.24.0-1

### Adapter firmware version

0x26720700

### Adapter

ConBee2 

### Debug log

{"id":35,"type":"EndDevice","ieeeAddr":"0xa4c138981d3988ff","nwkAddr":51615,"manufId":4417,"manufName":"_TZ3000_mmtwjmaq","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"powerSource":3,"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"currentZoneSensitivityLevel":1,"iasCieAddr":"0x00212effff07afaa","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1643543150792,"useImplicitCheckin":true}

Koenkk, 2022-03-23:
Could you provide the data points again?

jocamane, 2022-03-23:
Sure,

PIR state = Code 1
Battery level = Code 4

Thanks

jocamane, 2022-03-23:
In z2m:
[image]

[image]

jocamane, 2022-03-23:
and another mov sensor i have, **TS0202 _TYZB01_vwqnz1sn** have the same problem

Koenkk, 2022-03-24:
Can you try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    TS0202: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case 1:
                    result.occupancy = value === 1;
                    break;
                case 4:
                    result.battery = value;
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:TS0202 Unrecognized DP #${dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    },
};

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_ef5xlc9q'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_vwqnz1sn'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_2b8f6cio'},
        {modelID: 'TS0202', manufacturerName: '_TZE200_bq5c8xfe'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_qjqgmqxr'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_mmtwjmaq'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_kmh5qpmb'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_zwvaj5wy'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_bsvqrxru'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_tv3wxhcz'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_hqbdru35'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_otvn3lne'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_tiwq83wk'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_ykwcwxmz'},
        {modelID: 'WHD02', manufacturerName: '_TZ3000_hktqahrq'}],
    model: 'TS0202',
    vendor: 'TuYa',
    description: 'Motion sensor',
    whiteLabel: [{vendor: 'Mercator Ikuü', model: 'SMA02P'}, {vendor: 'TuYa', model: 'TY-ZPR06'}],
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report, fzLocal.TS0202],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery(), e.battery_voltage()],
    configure: async (device, coordinatorEndpoint, logger) => {
        /* uncomment this if it doesn't work */
        // const endpoint = device.getEndpoint(1);
        // await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    },
};

module.exports = definition;

```

If it doens't work, remove the two `//` from the configure method and try again

jocamane, 2022-03-24:
Hi, thanks

Don't work :(

**Results with your original external:**
[image]
[image]

**Results with two // removed:**
[image]
[image]

(I suppose this)
.....
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery(), e.battery_voltage()],
    configure: async (device, coordinatorEndpoint, logger) => {
        /* uncomment this if it doesn't work */
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
...

Right ?

jocamane, 2022-03-24:
I stop the z2m, change the external and after start z2m again

Koenkk, 2022-03-26:
> (I suppose this)

Yes, did you repair the device after this. If that doesn't help, can you force a configure via the z2m frontend; click on the device and press the yellow reconfigure button (right before doing that make sure to wakeup the device by pressing the button on it)

jocamane, 2022-03-26:
yes Koenkk, I still did better.
I deleted the device, changed the external, restarted the z2m and then went back to trying to put the device on the z2m.
all this via z2m gui.

Koenkk, 2022-03-26:
Can you try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    TS0202: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case 1:
                    result.occupancy = value === 1;
                    break;
                case 4:
                    result.battery = value;
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:TS0202 Unrecognized DP #${dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    },
};

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_ef5xlc9q'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_vwqnz1sn'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_2b8f6cio'},
        {modelID: 'TS0202', manufacturerName: '_TZE200_bq5c8xfe'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_qjqgmqxr'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_mmtwjmaq'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_kmh5qpmb'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_zwvaj5wy'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_bsvqrxru'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_tv3wxhcz'},
        {modelID: 'TS0202', manufacturerName: '_TYZB01_hqbdru35'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_otvn3lne'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_tiwq83wk'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_ykwcwxmz'},
        {modelID: 'WHD02', manufacturerName: '_TZ3000_hktqahrq'}],
    model: 'TS0202',
    vendor: 'TuYa',
    description: 'Motion sensor',
    whiteLabel: [{vendor: 'Mercator Ikuü', model: 'SMA02P'}, {vendor: 'TuYa', model: 'TY-ZPR06'}],
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report, fzLocal.TS0202],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery(), e.battery_voltage()],
    configure: async (device, coordinatorEndpoint, logger) => {
        /* uncomment this if it doesn't work */
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
        // await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    },
};

module.exports = definition;

```

(make sure to repair the device after this)

jocamane, 2022-03-27:
Hi,

it's ok with _TZ3000_mmtwjmaq, but no ok with _TYZB01_vwqnz1sn.

The data points of  _TYZB01_vwqnz1sn:
PIR state = Code 1
Battery level = Code 4
temper alarma  = Code 5
Current Luminance = Code 6
PIR Sensitivity = Code 9
Keep time = Code 10

States:

[image]

[image]

Thanks

Koenkk, 2022-03-28:
Does the `_TYZB01_vwqnz1sn` start working when you uncomment `        // await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);` and repair it aftewards?

jocamane, 2022-03-29:
> Does the `_TYZB01_vwqnz1sn` start working when you uncomment ` // await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);` and repair it aftewards?

no koenkk, it's the same

jocamane, 2022-03-29:
can we (you can), create an external just for this sensor and leave the other one as it was working?

Koenkk, 2022-03-29:
Could you maybe sniff the traffic when pairing the `_TYZB01_vwqnz1sn` to the tuya gateway? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

jocamane, 2022-03-29:
it is possible ? I have a conbee II and rpi4.
who i do ?

Koenkk, 2022-03-29:
For that you need to buy a CC2531 sniffer + flasher, up to you if you want to invest on this (no guarantees this will fix the issue)

gitariana, 2022-03-31:
Hello Koenk
i have the same problem with my humi/temp sensor TuYa TS0201.
Humi, temp and linkq come, battery% doesn't come.
Does the same fix work here?
Thanks

gitariana, 2022-04-13:
Hello Koenkk
i have the same problem with my humi/temp sensor TuYa TS0201.
Humi, temp and linkq come, battery% doesn't come.
Does the same fix work here?
Thanks

gitariana, 2022-04-18:
I have make the update from zigbee2mqtt 1.24.0 to 1.25.0 and then the battery value is coming.
Thank´s

closed 2022-05-26 by github-actions[bot]