Koenkk/zigbee2mqtt #11673: TS0504A

angustPL, 2022-03-02:
### What happened?

TS0504A recognized as TS0504B

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

CC2652P

Koenkk, 2022-03-02:
can you provide the data/database.db entry of this?

angustPL, 2022-03-03:
I hope that's what you mean

`{"id":7,"type":"Router","ieeeAddr":"0x60a423fffec20029","nwkAddr":5978,"manufId":4098,"manufName":"_TZ3000_ukuvyhaa","powerSource":"Mains (single phase)","modelId":"TS0504B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0504B","manufacturerName":"_TZ3000_ukuvyhaa","stackVersion":0,"dateCode":"","zclVersion":3}},"genOnOff":{"attributes":{"61440":0,"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":191}},"lightingColorCtrl":{"attributes":{"tuyaRgbMode":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1646307078505,"defaultSendRequestWhen":"immediate"}
{"id":8,"type":"Router","ieeeAddr":"0x60a423fffec20002","nwkAddr":3554,"manufId":4098,"manufName":"_TZ3000_ukuvyhaa","powerSource":"Mains (single phase)","modelId":"TS0504B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0504B","manufacturerName":"_TZ3000_ukuvyhaa","stackVersion":0,"dateCode":"","zclVersion":3}},"genOnOff":{"attributes":{"61440":0,"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":3}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1646307024575,"defaultSendRequestWhen":"immediate"}`

angustPL, 2022-03-03:
This is the problem because the device is actually TS0504A, unfortunately it is recognized as TS0504B. This is the problem as I am reporting on this issue.
I have 2 devices exactly like in this photo https://www.zigbee2mqtt.io/devices/TS0504A.html

Koenkk, 2022-03-05:
Then the device report incorrect information (something we cannot solve from z2m). As a workaround, stop z2m, and in the database.db change `"manufName":"_TZ3000_ukuvyhaa"` to `"manufName":"_TZ3000_nzbm4ad4"` and `"modelId":"TS0504B"` to `"modelId":"TS0504A"`

angustPL, 2022-03-05:
I understood. Problem has been solved. Thank you!
As for me, it can be closed.

angustPL, 2022-03-05:
Maybe one more ...
Is it possible to change the green and red colors in the configuration?
Theoretically, you can change the cables, but this option would make things easier.

Create a new issue for this

Koenkk, 2022-03-05:
@angustPL no that is not possible, an X/Y color is send to the device, the device itself is responsible for correctly controlling the RGB wires.

closed 2022-03-05 by angustPL