Koenkk/zigbee2mqtt #11675: Error after updating to 1.23.0-dev commit #1eb32870

SargonofAssyria, 2022-03-02:
### What happened?

Error is frequently present.
`error 2022-03-02 15:21:27: Exception while calling fromZigbee converter: fz.battery is not a function}`
the error is still present when I updated today to 1.24.0-dev commit #3535b2c6

### What did you expect to happen?

No error as before.

### Zigbee2MQTT version

1.24.0-dev

### Adapter firmware version

20220103

### Adapter

Texas Instruments LAUNCHXL-CC1352P-2

Koenkk, 2022-03-02:
should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

SargonofAssyria, 2022-03-04:
Tnx, That works, no more errors....

closed 2022-03-04 by SargonofAssyria