Koenkk/zigbee2mqtt #11677: Tuya TS004F delay on buttons 2, 3 and 4

dielessen, 2022-03-02:
### What happened?

Regarding https://github.com/Koenkk/zigbee2mqtt/issues/8149

I also have the same issue with the TS004F. With the delay on buttons 2,3,4.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

zzh! stick

dielessen, 2022-03-29:
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"4_single","battery":100,"linkquality":61,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"","battery":100,"linkquality":61,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4/action', payload '4_single'
Debug Received Zigbee message from 'Zolder schakelaar 4', type 'raw', cluster 'genOnOff', data '{"data":[1,72,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
Debug Received Zigbee message from 'Zolder schakelaar 4', type 'raw', cluster 'genOnOff', data '{"data":[1,72,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
Debug Received Zigbee message from 'Zolder schakelaar 4', type 'raw', cluster 'genOnOff', data '{"data":[1,73,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"4_single","battery":100,"linkquality":61,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"","battery":100,"linkquality":61,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4/action', payload '4_single'
Debug Received Zigbee message from 'Zolder schakelaar 4', type 'raw', cluster 'genOnOff', data '{"data":[1,74,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"4_single","battery":100,"linkquality":61,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"","battery":100,"linkquality":61,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4/action', payload '4_single'
Debug Received Zigbee message from 'Zolder schakelaar 4', type 'raw', cluster 'genOnOff', data '{"data":[1,74,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
Debug Received Zigbee message from 'Zolder schakelaar 4', type 'raw', cluster 'genOnOff', data '{"data":[1,75,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"4_single","battery":100,"linkquality":61,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"","battery":100,"linkquality":61,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4/action', payload '4_single'
Debug Received Zigbee message from 'Zolder schakelaar 4', type 'raw', cluster 'genOnOff', data '{"data":[1,75,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
Debug Received Zigbee message from 'Zolder schakelaar 4', type 'raw', cluster 'genOnOff', data '{"data":[1,76,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"4_single","battery":100,"linkquality":58,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"","battery":100,"linkquality":58,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4/action', payload '4_single'
Debug Received Zigbee message from 'Zolder schakelaar 4', type 'raw', cluster 'genOnOff', data '{"data":[1,77,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"4_single","battery":100,"linkquality":58,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4', payload '{"action":"","battery":100,"linkquality":58,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Zolder schakelaar 4/action', payload '4_single'
Debug Received Zigbee message from 'Zolder schakelaar 4', type 'raw', cluster 'genOnOff', data '{"data":[1,77,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0
Debug Received Zigbee message from 'Zolder schakelaar 4', type 'raw', cluster 'genOnOff', data '{"data":[1,77,253,0],"type":"Buffer"}' from endpoint 4 with groupID 0

tomhall12345, 2022-03-31:
Just wanted to say i'm having the exact same issue with a ZZH! Adapter.

mogim0gi, 2022-05-06:
I have same issue !

etique57, 2022-05-09:
Same issue for me with Slaesh's adapter.

I've no clue if that makes sense, but in the commit Koenkk/zigbee-herdsman-converters/pull/2848 shouldn't we have TS004F there as well?
[image]

Koenkk, 2022-05-10:
Could someone make a sniff when pressing the button on this device with both z2m 1.25.1 and the tuya bridge? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531 . For both sniffs also provide the network key.

etique57, 2022-05-11:
Thanks @Koenkk I've asked the TS004F discussion if they can help!

tomhall12345, 2022-05-28:
I don't have the tuya bridge otherwise I would do. Will a sniff with just the z2m help?

Koenkk, 2022-05-29:
@tomhall12345 > Will a sniff with just the z2m help?

I can give it a look

closed 2022-07-06 by github-actions[bot]

null-unknown, 2022-08-10:
I have the exact same issue but I am using ZHA. Could it be the device itself?

null-unknown, 2022-08-10:
@Koenkk I am also having this issue with z2m, could this be fixed?

Koenkk, 2022-08-10:
@null-unknown see https://github.com/Koenkk/zigbee2mqtt/issues/11677#issuecomment-1122670476

null-unknown, 2022-08-10:
Thanks for your reply @Koenkk. When I migrate from Tuya to home assistant, I also had this issue with the Tuya bridge. At this moment I am using a Sonoff 3.0 plus dongle, and I am not able to switch back. What about comment https://github.com/Koenkk/zigbee2mqtt/issues/11677#issuecomment-1121503732, could that be the fix?

Koenkk, 2022-08-11:
If the issue also appears with the original TuYa bridge I expect this is an issue with the device itself so we cannot fix it from z2m.

etique57, 2022-08-11:
Hi,

I have to say that I don't experience the problem anymore, and when I checked the file mentionned in #11677, it has now been updated to include TS004F.

[image]

null-unknown, 2022-08-11:
I see, did not notice that it is already been included. Seems indeed not to work as expected. Oh well, time to throw the TS004F away and get a TS0044.

konistehrad, 2023-10-21:
I'd like to reopen this if possible as part of my investigations into expanding support for TS004F's as I'm also experiencing this.

konistehrad, 2023-10-22:
Update: I think I've run the ball on this as far as I can. Unfortunately it does look like an implementation issue with the device itself. It presents four endpoints, but endpoints `[2,3,4]` have no associated clusters, meaning I think the device itself is incapable of receiving the default response that's sent in `fromZigbee:tuya_on_off_action` on any endpoint expect `1`, which does present the `genOnOff` cluster. The correct way to receive events from this device with no latency is as follows:

1) Pair device (natch)
2) Create an isolation group for this device and add it
3) Set the device to command mode (depends on #6303)
4) Consume the light link events (`on`, `off`, `brightness_step_up` and `brightness_step_down`) instead of `1_single`, etc