Koenkk/zigbee2mqtt #11678: Xiaomi LLKZMK11LM is not updating energy, temperature and voltage

DJTerentjev, 2022-03-02:
### What happened?

I try both last stable (1.24.0-1) and last edge version of Z2M. Relay is updated to  9-22-2021. I tried with 2 sticks cc2538 version 20211222 with support Aqara devices and SONOFF 20220103. Both same result: LLKZMK11LM is not updating energy, temperature and voltage:
[image]
I tried to connect same module to Xiaomi gateway and got energy:
[image]
Also I know at least two users wth same behavior  
I sniff the traffic between xiaomi gateway and relay module while pairing and while energy was growing.  

Could you please check this out and fix it if possible.
Please let me know if i can provide extra information

### Zigbee2MQTT version

1.24.0-1

### Adapter firmware version

20220103

### Adapter

SONOFF

### Debug log

[zigbee(pairing).pcapng.txt](https://github.com/Koenkk/zigbee2mqtt/files/8170682/zigbee.pairing.pcapng.txt)
[zigbee(energy).pcapng.txt](https://github.com/Koenkk/zigbee2mqtt/files/8170692/zigbee.energy.pcapng.txt)

McGiverGim, 2022-03-02:
Look here 
https://github.com/Koenkk/zigbee2mqtt/issues/7112

DJTerentjev, 2022-03-02:
> Look here #7112

I saw it, but that issue is closed and no solution.

Koenkk, 2022-03-02:
@McGiverGim if I understand correctly, the code from https://github.com/Koenkk/zigbee2mqtt/issues/7112#issuecomment-1057111602 should fix this. @DJTerentjev can you confirm this?

Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/7112

closed 2022-03-02 by Koenkk