Koenkk/zigbee2mqtt #11681: Connection to Linkind ZS130000078 lost after some hours/days

vguna, 2022-03-02:
### What happened?

The keypad is successfully paired and known to z2m. It can be used to enter codes, switch arm modes etc. without issue. But after a couple of days (for others one or a couple of hours), out of the sudden, it is not able to communicate to z2m anymore. The red "network" LED is blinking that it usually uses to signal that the keypad has to be initialized (e.g. via send "disarm" payload) first. But sending the payload is not able to init it again. Only after resetting/repairing it with z2m makes it work again.
While the keypad is not able to reconnect, z2m can still receive its health status messages though.

At the time, the keypad disconnected, the follow log was printed in z2m:

```
error 2022-03-02 20:14:00: Error: CommandResponse 0x588e81fffe21eede/1 ssIasAce.getPanelStatusRsp({"panelstatus":0,"secondsremain":0,"audiblenotif":0,"alarmstatus":0}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
```
Followed by those, when trying to send the usual disarm payload:
```
error 2022-03-02 20:21:13: Error: CommandResponse 0x588e81fffe21eede/1 ssIasAce.getPanelStatusRsp({"panelstatus":0,"secondsremain":0,"audiblenotif":0,"alarmstatus":0}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
```

There is already a [discussion](https://github.com/Koenkk/zigbee2mqtt/discussions/11453) going on, where 2 other guys having similar issues and also reporting that they can see similar errors in the logs.

We're running out of ideas what might cause this.

### What did you expect to happen?

Connection between the keypad and z2m is not lost.

### How to reproduce it (minimal and precise)

Init the keypad and use it for a couple of hours/days. Maybe also with long idle pauses in between. At some point it should disconnect.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20220103

### Adapter

SONOFF 3.0 USB Dongle Plus

writepatrick, 2022-03-05:
I have a similar or identical problem. Sometimes it works, sometimes after activating it by proximity it takes really long to (I assume) get the current state, then fail with a red blinking connection error on the keypad.
I use a Conbee2 stick with newest firmware.
For me it is not necessary to repair it. Most of the times another attempt with proximity activation is successful.
I will try to isolate the error message if the keypad is not acting like it should.

vguna, 2022-03-05:
Thanks for sharing! After it shows the red led, do you have to send the disarm payload to make it work again? Or really only retry with proximity? Do you have any automation in the background like HA or so that sends the init payload maybe?

writepatrick, 2022-03-05:
Hi,
no up until now I didn’t have the repairing issue. It works kind of, but is unresponsive and laggy (like timing out). I thought this could maybe be an issue of a low energy state of the keypad in which it is going to.
The only automation I am using is this blueprint [Here](https://community.home-assistant.io/t/zigbee2mqtt-sync-keypad-and-alarm-control-panel-states/345311) 

I also fiddles around this the occupancy clearing times from the default 90s up until 18h. I doesn’t improve this behaviour.
I was thinking about trying to prevent the keypad to go into „sleep mode“ (not even knowing if something like this exists) by initialising it every 30min.

vguna, 2022-03-05:
Mine was again dropping out from a previous working state. Again "no network route (205)". Moved it next to the sender, but the getpanelstatusresp error 205 continues. No auto recover or so. I'm wondering what it actually means. Does it nean it gets a panelStatusRequest

callacomp, 2022-03-05:
I have the same issue as vguna but using the same blueprint as writepatrick. Once it drops off mine doesn't ever connect again but continues to send stats to HA without an issue. I've tried sending a message to it but 205 error. Have to take battery out or reset to get it working again. LQI never drops below 85.

vguna, 2022-03-06:
So, one night keeping it next to the coordinator does not bring it back to live. I can confirm  that. In the log at the bottom you can see that I tried to send the disarm payload twice, but this did not have any effect. If I'm near proximity, it logs those line though:

```
Received Zigbee message from 'LinkindKeypad', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0`
```

So for me it looks like it is still connected, but somehow the disarm does not get through and the keypad does not get the panel status response as requested.

Maybe let's loop in @robertalexa, @GSzabados and @stefangries since it seems they're the experts with this keypad.

[linkind.log](https://github.com/Koenkk/zigbee2mqtt/files/8192355/linkind.log)

GSzabados, 2022-03-06:
Just out of curiosity, what router devices do you have on your network?

writepatrick, 2022-03-06:
Hi, 
just for debugging I am currently on zigbee2mqtt version 1.23.0-dev commit: [afe94a7]

My LQI is at 255 since it is just 2 meters away from the coordinator.

Did somebody use the included hub? How is the zigbee traffic there (is it even possible to sniff the traffic?), maybe we are missing some commands to keep the connection alive? Since I am just a hobbyist, this is my best guess. 
I‘d love to have the time to debug this, but with a family of four this is not possible for me.😞

robertalexa, 2022-03-06:
Hi guys, don't have any keypada anymore so my following thoughts are just guesses.

I have seen very similar behaviour with my thermostat remote from Hive.

And in my case my conclusion was a poor connection issue. The device wants to connect to the device it was originally adopted by, even though that device has a poor connection if moved around.

My suggestion would be to try to pair it using the closest router to the keypad.

In the backened, next ro the permit join button there is a dropdown that allows you to choose which router should do the pairing. Try to use one that is close to the keypad and monitor if it stilla falls off the network.

At least this will eliminate some variables...

writepatrick, 2022-03-06:
> Just out of curiosity, what router devices do you have on your network?

Just one IKEA TRÅDFRI lightbulb

vguna, 2022-03-06:
@GSzabados : I have no routers on my network. Also this network only contains of the PIR sensor and the keypad that came with the Linkind bundle. And the PIR is only for playing around with it. I'm using the SONOFF 3.0 USB Dongle Plus on a rpi 4, connected via USB and 1,5m long cable. Firmware is 20220103 that came with the stick. Z2M version is 1.23.0.
I also tried already to shift channels, but this does not make a huge difference in LQI as it seems.

@writepatrick : not used the hub at all yet. Maybe I will try one day to see if there is a new firmware available and flash it. I saw some guys also "hacked" it. It seems it's OpenWRT on it or so - but didn't follow closely. I also thought about it - that we might miss some commands it needs to stay alive.

vguna, 2022-03-06:
@writepatrick : what firmware does your keypad have? Mine is 3.13.

> And in my case my conclusion was a poor connection issue. The device wants to connect to the device it was originally adopted by, even though that device has a poor connection if moved around.

Hm since I only have one coordinator in the network, it should always be the same it should try to connect to. Also once it lost connection, moving it next to the coordinator does not bring it back. But it still talks to the coordinator and I can see received messages in the logs. So it has not really abandoned it, no? Sadly I don't know much about the zigbee protocols and handshakes needed :/.

GSzabados, 2022-03-06:
> I saw some guys also "hacked" it.

It was me as well...The MAC ack should be a communication issue on the Zigbee mesh. Either the device itself or interference corrupts the packages.

vguna, 2022-03-06:
> It was me as well...The MAC ack should be a communication issue on the Zigbee mesh. Either the device itself or interference corrupts the packages.

Huh, small world it is ;). IIRC the mac ack error only happened not that much, but 205 did. To understand better: how can it be, that the coordinator can receive messages from the keypad without an issue, but sending a response (panelState response as it seems?) results in 205 to the keypad?

But even if there is one or two corrupt messages, shouldn't it auto-recover if placed next to the coordinator (maybe 1-2 meters away) eventually? With a LQI of 176 or so.

GSzabados, 2022-03-06:
> Huh, small world it is ;). IIRC the mac ack error only happened not that much, but 205 did. To understand better: how can it be, that the coordinator can receive messages from the keypad without an issue, but sending a response (panelState response as it seems?) results in 205 to the keypad?

That is a good question, but it looks like a routing issue. 

With battery powered devices routing is different.

You can try the following and check what is happening.

1. Wake up the keypad
2. Hit reconfigure in Z2M
3. Important, make sure that the device will not go to sleep meanwhile doing the reconfigure.

Or to add a twist, before doing any of these, just try to restart Z2M first and then do the above.

@writepatrick, is your IKEA bulb running the latest firmware? If not, try to update it, and then pair the keypad through the bulb.

vguna, 2022-03-06:
@GSzabados : thanks, I will try that! Should I do it as a pre-setup or only if it already went south to revive it?

GSzabados, 2022-03-06:
Try to revive it that way. But I usually do a reconfigure with my battery powered devices when I pair them.

callacomp, 2022-03-07:
I'll try this also.
I have connected the keypad back to its original hub and it seems we're all on the same firmware, no updates. 3.13 is the latest.
I'm using latest firmware on sonoff 3+, but also had another CC2652 BLE Simplelink 2.4G device that wasn't working hence the switch to sonoff(everything else worked mind). I do have a few routers like plugs, switches   etc but have put it 2 feet from Coordinator now, still stops working

writepatrick, 2022-03-09:
@GSzabados 
„is your IKEA bulb running the latest firmware? If not, try to update it, and then pair the keypad through the bulb.“
I have created a network map in z2m and it is a mess, since most devices are bound to the coordinator even if they are far away. I tried to bind some sensors to the lightbulb because they are 1-2meters away, but I was not able to bind them to the tradfri bulb.
Is there a special trick to bind to an router, other then just to enable binding to a specific router?
I am on the latest ikea firmware.

GSzabados, 2022-03-09:
> I am on the latest ikea firmware.

The latest firmware from the non-test url (test url thingy unchecked in the OTA update settings)? IKEA just recently released some updates for most of their devices which solves a lot of networking issues, IMHO.

There is a drop down when you enable pairing and you can select which device you want to pair through. That is the easiest way to set that, but it only works if you are using the latest firmware on the CC2652 coordinators and if you migrated from and old one then you had to re-pair all devices. There were some backup issues which was not allowing the pairing through repeaters. It was a long time ago, but I don't remember when exactly. Of course it depends when you built your network first...
If you use ConBee, then I don't know how is that works there. I have only CC2531 and CC2652 coordinators. 

I believe there are some limitations on the coordinators for end devices (not repeaters), I mean how many of them can connect directly to the coordinator, but I cannot remember what is that number.

writepatrick, 2022-03-09:
> The latest firmware from the non-test url

yes. I have read that connecting to the lightbulb is quite tricky because you have to be quite close.
I have another router in the system a Heimann Sirene, which is much closer to the keypad. I will try to bind the keypad to the sirene, may this helps.

My network doesn't look promising at all.
[image]

vguna, 2022-03-12:
Also some feedback from my side. I moved the location of the keypad from the garage to my office room upstairs where it has even less LQI strength since there is a armor concrete ceiling inbetween. Triggering it each day a couple of times via proximity and there is no issue - for 4.5 days now. This is really strange since also the distance is more far away. It takes 2-4 led blinks now to connect though whereas in the garage it took 1-2. Could be concidence of course. So the "only" difference in the garage is that the other zigbee motion detector is nearby (4 meters away, also battery driven) and that a wallbox is next to the keypad. So I will now move it back to garage but will disable one of the two to see if it has an impact. Will report back.

Here's the LQI strength of they keypad monitored over the last days (moved on 03/07):

[image]

writepatrick, 2022-03-16:
> yes. I have read that connecting to the lightbulb is quite tricky because you have to be quite close.
> I have another router in the system a Heimann Sirene, which is much closer to the keypad. I will try to bind the keypad to the sirene, maybe this helps.

I managed to rebind the keypad to a nearby router (sirene_2) instead of the coordinator. Maybe this helps. Nonetheless I didn't experience further connection issues. It is only sometimes laggy but didn't fail (red link symbol) one.

I'll keep you posted if the new route is better.

GSzabados, 2022-03-16:
I tried to reproduce this issue with my setup but I couldn't managed it. The keypad responds every time correctly, even when I managed to crash most of the repeaters on the network. Once the repeaters recovered the keypad works as expected.

vguna, 2022-03-19:
Hey guys,
also an update from my side. After doing ok a couple of days in the garage, with wallbox completely powered down, it disconnected again.

What I checked:
- No sign of USB resetting the Sonoff stick via dmesg
- No sign of any related errors in syslog
- Uptime 12 days
- No sign of restart of z2m

Also the linkquality was ok I would say during that time window. On the 12th I moved it to the garage - after it was doing ok in my office - with smaller LQI:

[image]

Here's the whole debug log, from keypad working, to non-working - detected around 2022-03-19 11:35.

[linkind.log](https://github.com/Koenkk/zigbee2mqtt/files/8309880/linkind.log)

Also timeline extraction from the log what might be interesting:

[timeline.txt](https://github.com/Koenkk/zigbee2mqtt/files/8309893/timeline.txt)

You can also find the suggested fix with reconfiguring the Linkind in the logs. For me it also had no effect - also with restarting z2m before doing it.

Does somebody know if the `tamper` flag has some influence in the whole pairing/bringing it back handshake or is it just informational?

I have not removed batteries yet so I can try a couple of other things the next days with bringing it back.

vguna, 2022-03-19:
@GSzabados: how does your env look like? You also have the same Linkind keypad like us? Means: same firmware and also the Sonoff stick? You're running it standalone or together with the sirence hub that came with it (if that's even possible)? How do you actually integrate the keypad: are you using this blueprint thingy together with HA? Where is the Zigbee stick connected to? Rpi?

vguna, 2022-03-19:
Ok, now I had a weird situation. I upgraded to z2m 1.24 now in the hope of magical fixes.
Started z2m. Then did a reconfigure and in between also send the disarm payload. Then suddenly the keys looked like when they usually are "active" - and the keypad went off. But after activating it again via proximity, it shows the usual red led behavior.

GSzabados, 2022-03-20:
@vguna, can you do a favour and place a repeater in the garage? Make sure to have an LQI 20-50. A plug would be great.

Then pair your Keypad and Motion sensor through that.

I mean delete, force remove, restart z2m, pair through that repeater. And try this exercise again?

vguna, 2022-03-20:
That would mean I have to buy a plug that costs at least half of the keypad just for testing this 🤔 ...

GSzabados, 2022-03-20:
Which country are you from?

An IKEA plug cost 10 EUR.

I've bought last week a pair of Nous plugs which has power measurements as well.

https://nous.technology/product/a1z-1.html

They cost 50 PLN each, 100 PLN as a pair. Roughly about 10.5 EUR for a plug.

Of course the repeater/router can be a light bulb or any powered device, like a Sonoff Zigbee Mini for a switch. The plug was just a suggestion. (But make sure do not use old Osram bulbs... They are terrible repaters/routers.)

callacomp, 2022-03-20:
I've got the same issues and can try this, I've got numerous repeaters throughout the house and garage. Is there a way to force it to a specific repeater, or do I just need distance away from the coordinator?

GSzabados, 2022-03-20:
There is drop down where you start the pairing, the Permit to Join has a list of repeaters. Choose the repeater you want to pair the device through.

callacomp, 2022-03-20:
OMG thanks, never looked there before!

writepatrick, 2022-03-21:
> I managed to rebind the keypad to a nearby router (sirene_2) instead of the coordinator. Maybe this helps. Nonetheless I didn't experience further connection issues. It is only sometimes laggy but [...]

Hi since I paired the keypad to the nearby "Heiman Sirene" the response time after proximity activation got significantly better. 
Connected to the Coordinator (Conbee2 Stick): 2-3sec
Connected to the router (Heiman sirene): <2sec

vguna, 2022-03-21:
> Which country are you from?

Germany :).

You're right, the IKEA plug costs 12,99€ - that's fair. Although quite bulky. I think I give a plug a shot - I can use it for other things then if it's not helping with the keypad.

GSzabados, 2022-03-21:
> Germany :).

A bit cheaper:

https://www.idealo.de/preisvergleich/OffersOfProduct/200966190_-100306990-silvercrest.html

This is the one what I previously mentioned, Nous A1Z, with energy monitoring, but unfortunately out of stock on Amazon:

https://www.amazon.de/dp/B0054PSH0G/

Disregarding the fact, that they must come from the same Polish warehouse, where I ordered it from. :laughing: 

The IKEA Plug 10 EUR price, I got it from IKEA Netherlands. Damn, I have never thought that there would be 30 percent price difference in Germany.

vguna, 2022-03-21:
Hm for me it shows it's in stock - but it looks like it's still warm :)? https://github.com/Koenkk/zigbee2mqtt/issues/11863#issuecomment-1074127941
Not sure if I should bet on those newcomers. They are doing ok?

vguna, 2022-03-22:
Ok I bought the Nous one and paired the keypad with this one. Let's see how it goes 🤞.

[image]

GSzabados, 2022-03-22:
> Not sure if I should bet on those newcomers.

Actually, I am the one who opened that issue. 😉 

And they are working well.

vguna, 2022-03-22:
> Actually, I am the one who opened that issue. 

Yeah I saw - hence the question If they are good :). Good to hear they do the job! I admit I was too lazy for now to pair the motion sensor as well.

vguna, 2022-03-22:
Hmm it seems the keypad meanwhile connected to the coordinator directly. Why is that? Is it possible to force it to use only one dedicated router?

callacomp, 2022-03-24:
This is interesting, I've moved it to the back of my garage and connected it through a plug acting as a repeater. For the first time ever it's been working for a day. I will continue to monitor it. Also its much faster at responding. As soon as I wake it up the right lights come on, also arming and disarming is almost instant. I'll give it another day then bring it back in. Any idea what this means? As others have asked can you permanently fix it to a router?

GSzabados, 2022-03-29:
Eventually, I have ran into this issue. Today the keypad was not responding as expected. I've got the `Mac no Ack` error.

I've had a few other devices which stopped working as well. And all are using IAS cluster for their basic communication. (Linkind Door Sensor, and a SmartThings/SmartSense Multi Sensor (Door sensor))

So battery removal and replacement helped, but the SmartThings one had to be deleted and re-paired to make it work.

callacomp, 2022-03-29:
Well mine worked in the garage through a repeater for nearly 4 days without an issue and was really responsive. Ive now brought it back in, and within 4 hours it's stopped. So connected directly to the coordinator is a problem. So is it an issue with the coordinator?

vguna, 2022-04-03:
@GSzabados : What does IAS cluster mean? What coordinator do you have? Is it also a SONOFF 3.0 Plus USB Dongle?

Until now, after adding the just newly bought zigbee plug right next to the Keypad, it did not lose connection and it's quicker in connecting as reported by @callacomp. Meanwhile the motion sensor also auto-switched to it without doing anything. I think it's almost a week now. I will continue monitor and report back.

What does this mean in the end? That the SONOFF dongle is not 100% compatible with the keypad or vice-versa? That maybe in edge case error situations the one of the two didn't implement the spec right?

[image]

callacomp, 2022-04-11:
Ok, so after brining it back in the house for a few days and it stopped, I put it back out of range of my coordinator and within a day,  without taking batteries out or reconnecting it, it started working again through a repeater. Back to reacting fast? Got to be coordinator issue.

callacomp, 2022-04-16:
@Koenkk does this warrant looking at from the coordinator firmware?

Koenkk, 2022-04-17:
@callacomp could you provide the herdsman debug log starting from the point where it works until it starts failing?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

callacomp, 2022-04-22:
I'm no Linux guru and this took some doing. I may leave some hints to what I did to get them off for others. Then someone will show me a simple way. The file is over three days, and 100Mb in size. Where can I drop this, or should I start again. I can add it to a OneDrive, dropbox, etc which is best for you?

Koenkk, 2022-04-22:
>  I can add it to a OneDrive, dropbox, etc which is best for you?

Doesn't matter for me.

callacomp, 2022-05-09:
sorry @Koenkk for the delay, here are the logs
[https://www.dropbox.com/s/x7we0cypzs8k8su/log%20%281%29.txt?dl=0](url)
Let me know if these are the wrong ones.

What I can say is that I have now followed someone else's comment. I reconnected it via the coordinator and then made a wall plug available only and reconnected it to that and no the coordinator, so far it's been working for over a day and relatively quick.

Koenkk, 2022-05-09:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZB-KeypadGeneric-D0002'],
    model: 'ZS130000078',
    vendor: 'Linkind',
    description: 'Security keypad battery CUSTOM',
    meta: {battery: {voltageToPercentage: '3V_2100'}},
    fromZigbee: [fz.command_arm_with_transaction, fz.battery, fz.ias_ace_occupancy_with_timeout,
        fz.ias_smoke_alarm_1, fz.command_panic],
    exposes: [e.battery(), e.battery_voltage(), e.battery_low(), e.occupancy(), e.tamper(),
        exposes.numeric('action_code', ea.STATE).withDescription('Pin code introduced.'),
        exposes.numeric('action_transaction', ea.STATE).withDescription('Last action transaction number.'),
        exposes.text('action_zone', ea.STATE).withDescription('Alarm zone. Default value 23'),
        e.action([
            'panic', 'disarm', 'arm_day_zones', 'arm_all_zones', 'exit_delay', 'entry_delay'])],
    toZigbee: [tz.arm_mode],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const clusters = ['genPowerCfg', 'ssIasZone', 'ssIasAce', 'genBasic', 'genIdentify'];
        await reporting.bind(endpoint, coordinatorEndpoint, clusters);
        await reporting.batteryVoltage(endpoint);
        await device.getEndpoint(1).unbind('genPollCtrl', coordinatorEndpoint);
        device.defaultSendRequestWhen = 'immediate';
        device.save();
    },
    onEvent: async (type, data, device) => {
        if (type === 'message' && data.type === 'commandGetPanelStatus' && data.cluster === 'ssIasAce' &&
            globalStore.hasValue(device.getEndpoint(1), 'panelStatus')) {
            const payload = {
                panelstatus: globalStore.getValue(device.getEndpoint(1), 'panelStatus'),
                secondsremain: 0x00, audiblenotif: 0x00, alarmstatus: 0x00,
            };
            await device.getEndpoint(1).commandResponse(
                'ssIasAce', 'getPanelStatusRsp', payload, {}, data.meta.zclTransactionSequenceNumber,
            );
        }
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m
- open dashboard, see if the device description now ends with CUSTOM (this ensures the converter has been loaded correctly).
- repair the device
- press the yellow configure button, make sure it succeeds (if it fails try to press a button on the device right before pressing it)

callacomp, 2022-05-09:
Thanks @Koenkk. All good so far. Now seen as "Security keypad battery CUSTOM", then re-joined, and then reconfigured. Time will tell now, I'll keep trying it over the next few days and  let you know.

callacomp, 2022-05-09:
One step at a time, but its's noticeable that battery has is oddly reported. Straight to coordinator and through plug.
[image]

callacomp, 2022-05-10:
already stopped, but didn't have herdsman debug on. enabled and restarted but going to bed now. report tomorrow.

callacomp, 2022-05-11:
@Koenkk logs after your recent update. It worked for a while but now doesn't work.
[https://www.dropbox.com/s/lryqmmmdr9j16zh/log%20%283%29.txt?dl=0](url)

Koenkk, 2022-05-11:
@callacomp will need a sniff to dive further into this; any chance you can provide one starting at the point where it works until it starts failing? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

vguna, 2022-05-15:
Hey guys,
just to report back. After 6-7 weeks doing ok with the power plug as a router, it failed yesterday when I restarted OpenHab and so z2m. It could not reconnect after I removed batteries. I could see from the map that it connected to the coordinator instead of the power plug. So I had to remove and repair to the plug to make it work again.

closed 2022-06-22 by github-actions[bot]