Koenkk/zigbee2mqtt #11683: ZLinky_TIC difference between camelcase/snake case state

Phyks, 2022-03-02:
### What happened?

There is a difference between camelCase and snakeCase values in state of ZLinky_TIC. See for instance:

```
    "apparentPower": 350,
    "apparentPowerPhB": 65535,
    "apparentPowerPhC": 65535,
    "apparent_power": 3870
```

and

```
    "rmsCurrent": 1,
    "rms_current": 7
```

camelCase value seem to be stuck and the source of truth seems to be snake case.

### What did you expect to happen?

Without any further details and with this naming, I would have expected the snake case to be the same values as camel case.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

antoinelibert, 2022-03-02:
It looks like remainders of a previous dev version. Maybe try to reset the state db?

Phyks, 2022-03-03:
Indeed, this fixed it. Thanks!

closed 2022-03-03 by Phyks